v4l2loopback/v4l2loopback #383: error "buffer 0 was not queued, this indicate a driver bug"

step-, 2020-11-29:
A gstreame pipeline involving a video test source and two loopback devices prints error message "buffer 0 was not queued, this indicate a driver bug" to my tmux terminal.

Versions:
- GNU/Linux 5.4.60 # 1 SMP PREEMPT  x86_64
- v4l2loopback-0.12.5_00198_g9c965bd
- gst2-libav-091418, gst2-plugins-base-091418, gst2-plugins-good-091418, gstreamer2-091418

```
v4l2-ctl --list-devices
Dummy video device (0x0000) (platform:v4l2loopback-000):
        /dev/video2

Dummy video device (0x0001) (platform:v4l2loopback-001):
        /dev/video3

Dualpix HD720p for Notebooks: D (usb-0000:00:1d.0-1.5.4):
        /dev/video0
        /dev/video1
```
```
~ # date +%s.%N; gst-launch-1.0 videotestsrc \! tee name=t \! queue \! v4l2sink device=/dev/video2 t. \! queue \! v4l2sink device=/dev/video2; date +%s.%N
1606653019.590470227
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
```
Nothing prints to this tmux terminal until I switch focus to another window; when I do this error message starts printing apparently endlessly...
```
0:00:45.631844629 31964      0x167e190 ERROR          v4l2allocator gstv4l2allocator.c:1304:gst_v
4l2_allocator_dqbuf:<v4l2sink1:pool:sink:allocator> buffer 0 was not queued, this indicate a driver bug.
0:00:45.664912792 31964      0x167e190 ERROR          v4l2allocator gstv4l2allocator.c:1304:gst_v
```
It keeps repeating until I switch focus back and hit ^C
```
4l2_allocator_dqbuf:<v4l2sink1:pool:sink:allocator> buffer 0 was not queued, this indicate a driver bug.
^Chandling interrupt.

Interrupt: Stopping pipeline ...
Execution ended after 0:00:47.215840918
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
0:00:47.247666197 31964      0x167e1e0 ERROR          v4l2allocator gstv4l2allocator.c:1304:gst_v
4l2_allocator_dqbuf:<v4l2sink0:pool:sink:allocator> buffer 0 was not queued, this indicate a driver bug.
Setting pipeline to NULL ...
Freeing pipeline ...
~ #
```
Thank you.

cout, 2021-09-30:
I am writing my first gstreamer application and I was getting this same error.  The pipeline consists of appsrc, videoconvert, and v42lsink.

The error went away when I set is-live to true for the appsrc.

step-, 2021-09-30:
Thank you. I can't confirm either way because I no longer have that video hardware at my disposal.

DennisZhangOiler, 2024-10-17:
Hi @step- and @cout ,
this could be a hardware issue, check this [link](https://forums.developer.nvidia.com/t/gstreamer-pipeline-hangs-after-hours-due-to-errors/174219).

stephematician, 2025-01-28:
> ```
> gst-launch-1.0 videotestsrc \! tee name=t \! queue \! v4l2sink device=/dev/video2 t. \! queue \! v4l2sink device=/dev/video2
> ```

@step-  - that string doesn't look right to me anyway, were you trying to sink to the same device (`/dev/video2`) twice? That would almost certainly fail.

step-, 2025-01-29:
@stephematician, thanks for your note. After all this time, I don't know what I was trying to do.

closed 2025-01-29 by step-