v4l2loopback/v4l2loopback #385: image in virtual cam horizontaly inverted

tristansca, 2020-12-01:
Is it possible to invert horizontaly the image because it is not right:

[image]

please help

closed 2020-12-01 by tristansca

umlaeute, 2020-12-01:
the answer is *no*, but i think you figured this out already.

robertvxt, 2021-01-11:
Because the v4l2loopback module does not support this so update the solution for my case as workaround.

I'm using FFMPEG to transcode the input stream/video into the virtual camera (generated by v4l2loopback) and set AVFilter flag to `hflip`.
i.e. `ffmpeg -i $stream_url -fflags nobuffer -vf hflip -pix_fmt yuv420p -f v4l2 /dev/video0`

Hope this's useful to someone.