v4l2loopback/v4l2loopback #388: Can't build since kernel 5.10.x

Creteil, 2020-12-18:
Trying to build v4l2loopback 0.12.5 (current git) with linux 5.10.1, I get this error:

```
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/bigbob/tmp/Building/v4l2loopback/v4l2loopback modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.1-051001-generic'
make[3]: *** No rule to make target 'scripts/module.lds', needed by '/home/bigbob/tmp/Building/v4l2loopback/v4l2loopback/v4l2loopback.ko'.  Stop.
make[2]: *** [scripts/Makefile.modpost:117: __modpost] Error 2
make[1]: *** [Makefile:1709: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.1-051001-generic'
make: *** [Makefile:44: v4l2loopback.ko] Error 2
Command exited with non-zero status 2
0.66user 0.37system 0:01.18elapsed 88%CPU (0avgtext+0avgdata 19760maxresident)k
19928inputs+688outputs (0major+126508minor)pagefaults 0swaps
bigbob@bigbob-UX31A:~/tmp/Building/v4l2loopback/v4l2loopback$
```

umlaeute, 2020-12-18:
can you build other out-of-tree modules, or is this simply an issue with the kernel's build-system for such modules.

see also: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131

the launchpad bug seems to indicate that this is indeed a bug in the packaging of the kernel-sources.

umlaeute, 2020-12-18:
i'm closing this for now.
if you still think the issue is with `v4l2loopback`, just respond, so I'll answer it again.

closed 2020-12-18 by umlaeute

KrakenTyio, 2021-02-16:
based on rest of the world its new feature from kernel which block compilation on 5.10.14+

`This is NOT_A_BUG , just a new feature starting from the kernels 5.10.0 onwards.`

https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131 navigate to bottom

MartinX3, 2021-02-17:
Yes, it would be nice, if it would work with kernel 5.10 :)

umlaeute, 2021-02-18:
iIt's working nicely on my Debian/sid system with `linux-5.10.13`.

so there are two possibilities:
1. it's indeed a bug in the packaging and [Debian fixed it](https://bugs.debian.org/975571#24). in this case please report bugs to your distribution, and not to this repository
2. it's indeed intended behaviour and the Debian fix just hides it. since I'm on Debian, I cannot reproduce the issue, and *will not* attempt to fix it. In this case, please submit a PR.

in *any* case: please do stop whining.

KrakenTyio, 2021-02-18:
On 5.10.13 works ok, but issue is showing 5.10.14-5.10.16, not tested on 5.11.x
Many kernels modules broken on this versions.

umlaeute, 2021-02-18:
i was mainly commenting on @MartinX3 's remark that "it would be nice, if it would work with kernel 5.10".

in my book `5.10.13` is definitely part of `5.10`.

pgaertig, 2021-03-15:
Here is the workaround which helps compile the module with Kernel 5.10.14, Ubuntu 20.10, v4l2loopback 0.12.5-1:
```bash
bash
sudo cp /usr/src/linux-headers-`uname -r`/scripts/module.lds.S /usr/src/linux-headers-`uname -r`/scripts/module.lds
sudo sed -i '/^[@#]/d' /usr/src/linux-headers-`uname -r`/scripts/module.lds
sudo dpkg-reconfigure v4l2loopback-dkms
```