v4l2loopback/v4l2loopback #390: compiling for custom kernel 5.9.12-050912-generic fails due to missing build folder

nielek2, 2020-12-29:
looking for advice on how to compile v4l2loopback...

`uname -r`
5.9.12-050912-generic

git clone --> `make`

>Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/****/Tools/v4l2loopback modules
make[1]: *** /lib/modules/5.9.12-050912-generic/build: No such file or directory.  Stop.
make: *** [Makefile:44: v4l2loopback.ko] Error 2

the `lib/modules/ folders` of older kernels on my machine contain a build subfolder, but `/lib/modules/5.9.12-050912-generic/build`does indeed not exist

similar problem descriptions lead to the suggestion to `sudo apt-get install linux-headers-$(uname -r)` which leads to

>Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package linux-headers-5.9.12-050912-generic is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'linux-headers-5.9.12-050912-generic' has no installation candidate

any tips & tricks or suggestions?

umlaeute, 2020-12-29:
file a bug against your distribution's kernel-package?

closed 2020-12-29 by umlaeute

nielek2, 2020-12-29:
ok, solved it, thanks @umlaeute (:

if anyone runs into the same issue:
after unsuccessfully trying to fix the missing /lib/modules/5.9.12-050912-generic/build symlink to the kernel source, i installed a newer ubuntu mainline kernel (5.9.12 --> 5.9.16). now compiling worked... (; 

so yes, probably a bug in the ubuntu mainline kernel package.. this time i used a tool called "ubuntu mainline kernel installer" after installing the kernel manually last time

osilkin98, 2021-06-18:
I encountered this issue because I had two kernels installed on Fedora so I solved it by going into my grub config and setting `GRUB_DEFAULT=0` which let grub select the more recent kernel on bootup.