v4l2loopback/v4l2loopback #391: Configuring v4l2loopback module to run at startup doesn't work, either that or I'm just not understanding this

jconnolly837, 2021-01-04:
I'm using Fedora 33 Workstation, I successfully compiled and installed v4l2loopback. I'm able to activate the module by running "sudo modprobe v4l2loopback video_nr=2" and it's active for /dev/video2. But adding "options v4l2loopback video_nr=2" to /etc/modules-load.d/v4l2loopback.conf and running "sudo dracut -f" before restarting acts like I didn't even do anything. I did have "options v4l2loopback card_label="device number 2"" under "options v4l2loopback video_nr=2" in there for a while, I thought removing it might do something, but it did nothing. I also tried just having "v4l2loopback video_nr=2" and still nothing. 

Hopefully I'm just misunderstanding the readme instructions so someone can correct me. I'm sure I did this right, but it's obviously not working. 

I suppose it's not such a big deal, but I'd really prefer not having to activate the module manually between reboots if I wanna use v4l2loopback.

umlaeute, 2021-01-04:
1. please don't use this bug-tracker for user-support.
2. please don't post the same issue twice (#392 is an exact duplicate of this)
3. please consult your distribution's manual on how to load modules persistently

having said that, the things you did are not really related to loading ("activating") the module at boot time
1. `dracut` is a tool for creating an initramfs (for making modules *available* to the kernel before it can read the root filesystem - which is important if you need those modules for accessing the root filesystem). since `v4l2loopback` is not essential for *booting* your system, there's no need adding it to the initramfs
2. putting stuff into `/etc/modules-load.d/v4l2loopback.conf` is great for providing default arguments wen the module is loaded. however, this files doesn't *initiate* the loading of the module itself.

anyhow.
on my Debian system, i can just list the modules i want to load at boot time in `/etc/modules`.
Fedora might do the same (or not: if so, you ought to consult a Fedora user forum)

closed 2021-01-04 by umlaeute

tovrstra, 2021-03-17:
@jconnolly837 You seem to make mistakes with the configuration files on Fedora. E.g. the following works for me:

- **/etc/modules-load.d/v4l2loopback.conf** to auto-load on boot:
  ```
  v4l2loopback
  ```
- **/etc/modprobe.d/v4l2loopback.conf** to set default options:
  ```
  options v4l2loopback exclusive_caps=1
  options v4l2loopback devices=2
  options v4l2loopback video_nr=7,8
  options v4l2loopback card_label="tablet,zoom"
  ```

AFAIK, this is correctly explained in the README

Dylan700Manes, 2022-11-05:
Pour charger correctement un modprobe en bullseye chez :: raspberry 
Voici la mérhode :: crontab -e
@reboot /usr/sbin/modprobe v4l2loopback video_nr=77,88,99;
le reste, :: 
:: #echo "v4l2loopback ">/etc/modules-load.d/v4l2loopback.conf;
:: #echo "v4l2loopback optioms video_nr=77,88,99">/etc/modprobe.d/v4l2loopback.conf;
Ou encore :: # echo "v4l2loopback video_nr=77,88,99">> "/etc/modules";
Est décoratif [✨⛲🗿] touristique [✈]