v4l2loopback/v4l2loopback #394: Key was rejected by service

sathed, 2021-01-14:
I installed the `v4l2loopback` module using the DKMS instructions without any error, but now I'm getting the following error when I try to load the module.

    # modprobe v4l2loopback
    modprobe: ERROR: could not insert 'v4l2loopback': Key was rejected by service

I've googled for an answer, but I can't seem to find anything that works. I did try uninstalling and unbuilding the module:

```
# dkms unbuild v4l2loopback -v ${version}

-------- Uninstall Beginning --------
Module:  v4l2loopback
Version: 0.12.5
Kernel:  5.10.6-200.fc33.x86_64 (x86_64)
-------------------------------------

Status: This module version was INACTIVE for this kernel.
depmod....

DKMS: uninstall completed.
```
But now I can't add or remove the module because of the following error(s).

```
# dkms remove v4l2loopback -v ${version}
Error! There is no instance of v4l2loopback 0.12.5
# dkms add v4l2loopback -v ${version}
Error! DKMS tree already contains: v4l2loopback-0.12.5
You cannot add the same module/version combo more than once.
```
Anyone have any thoughts?

Also, I'm using Fedora 33.

umlaeute, 2021-01-15:
module signing is an issue with your kernel/distribution. it's not something that can be solved in `v4l2loopback`.
so I'm closing this.

please contact a distribution specific forum.
if you find a solution, don't be afraid to post it here (so other people can benefit from your research)

closed 2021-01-15 by umlaeute

gotev, 2021-01-19:
@sathed I have the same problem on Fedora 32 and @umlaeute is right. Fedora enforces only signed modules to be loadable, so we have to compile the module and then sign it.

Found this guide which may help: https://docs.fedoraproject.org/en-US/fedora/rawhide/system-administrators-guide/kernel-module-driver-configuration/Working_with_Kernel_Modules/#sect-signing-kernel-modules-for-secure-boot

Underknowledge, 2021-04-05:
Finally got around to Install v4l2loopback and bashed it.
Chances are that I went a little overboard with this. It's not pretty, just thrown together.
https://gist.github.com/Underknowledge/78bdf079469f3f5eb4d1dfb9419cc149

When SecureBoot is enabled a generic x509 key will be generated and imported (you have to reboot the PC in this step) 
Afterward the script downloads and installs the latest tagged version of v4l2loopback and signs the module (or not).

hhlp, 2022-09-01:
You don't need to do all this stuff, just signed:

```
sudo cat /usr/share/doc/akmods/README.secureboot
```

Now you need to enroll the public key in MOK, this process is described below:

- Ask MOK to enroll new keypair with certificate with the command
                   `mokutil --import /etc/pki/akmods/certs/public_key.der`.
- mokutil asks to generate a password to enroll the public key.
- Rebooting the system is needed for MOK to enroll the new public key.
- On next boot MOK Management is launched and you have to choose
                  "Enroll MOK".
- Choose "Continue" to enroll the key or "View key 0" to show the keys already enrolled.
- Confirm enrollment by selecting "Yes".
- You will be invited to enter the password generated above. 
                  WARNING: keyboard is mapped to QWERTY!
- The new key is enrolled, and system ask you to reboot.
 
You can confirm the enrollment of the new keypair once the system rebooted with:

`mokutil --list-enrolled | grep Issuer`

or with:

`mokutil --test-key /etc/pki/akmods/certs/public_key.der` -> is already enrolled

Regards.,

drindt, 2024-06-30:
I just did this on a Fedora 40 installation. The key is enrolled, but I still can't load the module:
`modprobe: ERROR: could not insert 'v4l2loopback': Key was rejected by service`
Are there any other steps I need to take?

umlaeute, 2024-06-30:
You probably should read the kernel module documentation of your distribution.

Signing modules is really outside the scope of this project.