v4l2loopback/v4l2loopback #398: Last version has been released 10 months ago

alesguzik, 2021-02-05:
Currently there are no released versions that include dynamic device management. I want to update NixOS package to have it, but would prefer not to resort to packaging HEAD version. Judging by the commit history, previously new versions has been released more often, so maybe it is time to cut a new release.

Thanks.

umlaeute, 2021-02-05:
i very much want to have #359 included.

however, i think this will require some API changes, so the plan is to not release until we have basic split devices working.

evelikov, 2022-01-21:
Don't have much say what happens in the project. Although personally if there are doubts about instability or potential regressions, one can simply do a release with the `REMAKE_INITRD` change alone.

It is perfectly safe and will keep the log free of annoying deprecation warnings.

alesya-h, 2022-05-22:
@umlaeute there has been no activity on #359 for over a year, and last release was more than 2 years ago. Would it be possible to release 0.12.6 with current updates and release 0.13.0 when/if split devices are implemented?

umlaeute, 2022-05-23:
@alesya-h which features would you need in 0.12.6?

alesya-h, 2022-05-28:
@umlaeute I personally mostly need dynamic device management and v4l2loopback-ctl.

BenBE, 2022-05-28:
With #359 being mostly dead by now, what sense does it make to wait for it any longer and thus delay the release of the other patches that accumulated by now?

BenBE, 2022-09-01:
The last release has been `0.12.7` about 2 weeks ago, but without several code quality fixes. That said, you should probably notify that PPA to include the changes from `0.12.6` and `0.12.7` as they fix one CVE as well as consolidate some other issues reported from downstream maintainers (as I understood the changelog).

FWIW: Care to link the PPA?

jbrains, 2022-09-02:
@BenBE Thank you for clarifying the situation for me. I'm discovering it as I go.

The PPA in question: https://launchpad.net/~flexiondotorg/+archive/ubuntu/v4l2loopback

I'll get this message to them somehow.

closed 2024-03-19 by umlaeute