v4l2loopback/v4l2loopback #399: v4l2loopback on host does not work for Virtualbox

attilakinali, 2021-02-09:
Host: Debian/stable (10), Kernel 5.6.11, v4l2loopback from git (2021-02-05 10e1c61)

Feeding the loopback device from ffmpeg or obs (with v4l2sink) works fine on the host. All host applications that I tested it with work fine and can read the video back.

Getting it into Virtualbox doesn't work. I can connect the loopback device, it becomes visible in the guest but does not seem to work. Depending on application i get "no such devices" (eg. cheese) to just looping and not doing anything (e.g. vlc). This is independent of the guest OS, same for different distributions of Linux, with different kernel versions and Windows 10.

Using a hardware webcam (logitech c920) as v4l2 device works fine. There is nothing in the log files. As far as I can tell, both the loopback device and the webcam behave the same (from the perspective of what ends up in the log files and using v4l2-dbg).

My guess is, that v4l2loopback behaves slightly different in some corner case that virtualbox seems to trigger, but I have no idea what that could be or how I could help debug this.

umlaeute, 2021-02-09:
afaict this is an issue with VirtualBox rather than `v4l2loopback`.
VBox assumes to have lower-level access to the video-device that is passed through than `v4l2loopback` can expose (at least that's what I assume)

attilakinali, 2021-02-09:
Is there anything I could help with that? Unfortunately, the source code of the extension pack, where the webcam interface resides in, is not open source.

umlaeute, 2021-02-09:
nag oracle and create bug-reports for VirtualBox.

attilakinali, 2021-02-09:
Already on it! :-)

theKidOfArcrania, 2021-03-05:
Do you have a link to the bug report? seems like i also ran into this issue earlier today and would like to know what the status on oracle's end is.

jinliming2, 2021-03-05:
@theKidOfArcrania https://www.virtualbox.org/ticket/20176

Townsheriff, 2021-03-22:
Hello, I'm also trying to loop video to guest OS. Here a person is saying it is possible - https://forums.virtualbox.org/viewtopic.php?f=7&t=96552 I'm a bit confused, since I'm also getting black screen. To me it makes no sense that it is working for him.

attilakinali, 2021-03-22:
This has been recently fixed in virtualbox (see https://www.virtualbox.org/ticket/20176 ) you have to wait for the next release (probably 6.1.20) of vbox, which should happen very soon now [tm].

umlaeute, 2024-03-19:
just tested VirtualBox 7.0 (on a Debian host), and it works.

closed 2024-03-19 by umlaeute