v4l2loopback/v4l2loopback #400: kernel warning about ignored parameters

weberk, 2021-02-10:
Seems like my kernel is not able to hand over the configuration parameters to the module.

```
chronos@localhost / $ uname -ra
Linux localhost 5.4.75-brunch-sebanc #1 SMP PREEMPT Sun Dec 27 08:32:30 CET 2020 x86_64 Intel(R) Core(TM) i7-4650U CPU @ 1.70GHz GenuineIntel GNU/Linux
```
I am using a version of v4l2loopback cloned from the HEAD in github:
chronos@localhost /usr/local/bin/v4l2loopback $ git describe --tags
v0.12.5-217-g10e1c61

I wonder if my the MODULE_PARM- macros from the kernel header file my not be used adequate?
The only messages from the module out of DMSG
```
[   14.096742] v4l2loopback: unknown parameter 'debug' ignored
[   14.096745] v4l2loopback: unknown parameter 'video_nr' ignored
[   14.096747] v4l2loopback: unknown parameter 'card_label' ignored
```

No additional video device is created (/dev/video1).

Any hints what I could try to support narrowing down the error?

BenBE, 2021-02-10:
What is the command line used to load this module? (`modprobe …` / `/etc/modprobe.d/<v4l2loopback>.conf`)

umlaeute, 2021-02-10:
- what is the version of `v4l2loopback` you are *loading*
- how do you load the module?
- why on earth do you have a directory `v4l2loopback` under `/usr/local/bin/` that contains source code?

weberk, 2021-02-10:
The module is loaded by the following file:
```
localhost /etc/init # cat v4l2loopback.conf 
start on started boot-services
script
 modprobe v4l2loopback
end script
```
fetching config data from:
```
localhost /etc/modprobe.d # cat v4l2loopback.conf 
options v4l2loopback debug=9
options v4l2loopback video_nr=-1
options v4l2loopback card_label="virtualdevice"
```

The version is rather current (2 day old from git):
```
chronos@localhost /usr/local/bin/v4l2loopback $ git describe --tags
v0.12.5-217-g10e1c61
```
Since this is ChromeOS (brunch) I am not allowed to execute shell scripts on every partition. For /usr/local it is allowed - therefore I am compiling my packages from source there. An additional advantage is, that it survives CromeOS updates wiping other partitions.

weberk, 2021-02-10:
If I remove the config file I do not see any feedback in dmesg, but also no new /dev/video1 is created. Can I enforce verbosity of the module during load?

umlaeute, 2021-02-10:
`git describe` does not show the version of the module as it is loaded. it shows the version of some git repository lying on your harddisk.
the only way to know which version you are running is to watch out for a "v4l2loopback driver version" in the `dmesg` output.

> For /usr/local it is allowed

but you put a directory into `/usr/local/bin` while *this* directory is supposed to **only** hold executable files (and no directories).
just put it into */usr/local/src` or somesuch.

i don't see why you would need to have the *v4l2loopback* sources on a partition that allows you to execute binaries.
*compiling* does not require executing scripts.

weberk, 2021-02-10:
Ok, I moved my sources to /usr/local/src ;-)

There is nothing about a driver version output in the dmesg output, but the module fails receiving its parameters:
```
....
[   13.578753] Bluetooth: btbcm_setup_apple() hci0: BCM20702B0 Generic USB Class 1 @ 20 MHz
[   13.669064] usbcore: registered new interface driver snd-usb-audio
[   13.712010] usb 1-3.2: USB disconnect, device number 8
[   13.972868] v4l2loopback: unknown parameter 'debug' ignored
[   13.972870] v4l2loopback: unknown parameter 'video_nr' ignored
[   13.972871] v4l2loopback: unknown parameter 'card_label' ignored
[   14.008525] pci 0000:02:00.0: attach allowed to drvr facetimehd [trusted device]
...
```
There some other observation pointing on some incompatibility:

```
chronos@localhost /usr/local/src/v4l2loopback $ sudo modinfo v4l2loopback
filename:       /lib/modules/5.4.75-brunch-sebanc/extra/v4l2loopback.ko
license:        GPL
description:    Sound driver for My Chip
depends:        
retpoline:      Y
name:           v4l2loopback
vermagic:       5.4.75-brunch-sebanc SMP preempt mod_unload 
chronos@localhost /usr/local/src/v4l2loopback $ sudo modprobe -vf v4l2loopback
insmod /lib/modules/5.4.75-brunch-sebanc/extra/v4l2loopback.ko debug=9 video_nr=-1 card_label="virtualdevice" 
modprobe: ERROR: could not insert 'v4l2loopback': Exec format error
chronos@localhost /usr/local/src/v4l2loopback $ uname -rv
5.4.75-brunch-sebanc #1 SMP PREEMPT Sun Dec 27 08:32:30 CET 2020
```

weberk, 2021-02-11:
success:
```
[ 2026.520412] v4l2-loopback[2405]: creating v4l2loopback-device #-1
[ 2026.520420] /usr/local/src/v4l2loopback/v4l2loopback.c:2437[v4l2_loopback_add]
[ 2026.520423] /usr/local/src/v4l2loopback/v4l2loopback.c:2452[v4l2_loopback_add]
[ 2026.520426] /usr/local/src/v4l2loopback/v4l2loopback.c:2253[init_vdev]
[ 2026.520429] /usr/local/src/v4l2loopback/v4l2loopback.c:2286[init_vdev]
[ 2026.520432] /usr/local/src/v4l2loopback/v4l2loopback.c:2292[init_capture_param]
[ 2026.520435] /usr/local/src/v4l2loopback/v4l2loopback.c:2473[v4l2_loopback_add]
[ 2026.520449] /usr/local/src/v4l2loopback/v4l2loopback.c:2153[try_free_buffers]
[ 2026.520451] /usr/local/src/v4l2loopback/v4l2loopback.c:2136[free_buffers]
[ 2026.520456] v4l2-loopback[2137]: freeing image@0000000000000000 for dev:00000000c8217222
[ 2026.520460] /usr/local/src/v4l2loopback/v4l2loopback.c:2238[allocate_timeout_image]
[ 2026.520463] v4l2-loopback[2530]: buffer_size = 0 (=0)
[ 2026.520466] /usr/local/src/v4l2loopback/v4l2loopback.c:2164[allocate_buffers]
[ 2026.520977] /usr/local/src/v4l2loopback/v4l2loopback.c:2545[v4l2_loopback_add]
[ 2026.520981] v4l2-loopback[2863]: module installed
[ 2026.520984] v4l2loopback driver version 0.12.5 loaded
[ 2026.539072] /usr/local/src/v4l2loopback/v4l2loopback.c:2000[v4l2_loopback_open]
[ 2026.539080] v4l2-loopback[2026]: opened dev:00000000c8217222 with image:0000000000000000
[ 2026.539082] /usr/local/src/v4l2loopback/v4l2loopback.c:2027[v4l2_loopback_open]
[ 2026.539098] video0: VIDIOC_QUERYCAP: driver=v4l2 loopback, card=virtualdevice, bus=platform:v4l2loopback-000, version=0x0005044b, capabilities=0x85208003, device_caps=0x05208003
[ 2026.539208] /usr/local/src/v4l2loopback/v4l2loopback.c:2036[v4l2_loopback_close]
[ 2026.539212] /usr/local/src/v4l2loopback/v4l2loopback.c:2153[try_free_buffers]
[ 2026.539214] /usr/local/src/v4l2loopback/v4l2loopback.c:2136[free_buffers]
[ 2026.539217] v4l2-loopback[2137]: freeing image@0000000000000000 for dev:00000000c8217222
[ 2026.539221] /usr/local/src/v4l2loopback/v4l2loopback.c:2058[v4l2_loopback_close]

chronos@localhost /usr/local/src/v4l2loopback $ v4l2-ctl --list-devices
Apple Facetime HD (PCI:0000:02:00.0):
        /dev/video1
virtualdevice (platform:v4l2loopback-000):
        /dev/video0
```
Thanks for your patience!

umlaeute, 2021-02-11:
so what was the problem in the first place?

since you solved your problem, would you mind sharing what you did so others can learn from that as well?

weberk, 2021-02-12:
Ok, so in order to get this compiled and successfully accepted by the kernel I had to modify the git repo as follows:

* In order to get rid of a linker error (undefined symbol: __stack_chk_guard) I added some dummy implementation I found on Google for such cases.
* In order to get the kernel check the symbols of the new module I had to add depmod -A to the Makefile.

```
chronos@localhost /usr/local/src/v4l2loopback $ git diff 
diff --git a/Makefile b/Makefile
index b895546..3cf0f47 100644
--- a/Makefile
+++ b/Makefile
@@ -49,6 +50,7 @@ install:
        @echo ""
        @echo "SUCCESS (if you got 'SSL errors' above, you can safely ignore them)"
        @echo ""
+       depmod -A
 
 install-utils: utils/v4l2loopback-ctl
        $(INSTALL_DIR) "$(DESTDIR)$(BINDIR)"
diff --git a/v4l2loopback.c b/v4l2loopback.c
index a750fd3..840a9fe 100644
--- a/v4l2loopback.c
+++ b/v4l2loopback.c
@@ -279,6 +279,19 @@ MODULE_PARM_DESC(max_height,
 static DEFINE_IDR(v4l2loopback_index_idr);
 static DEFINE_MUTEX(v4l2loopback_ctl_mutex);
 
+unsigned long __stack_chk_guard;
+void __stack_chk_guard_setup(void)
+{
+     __stack_chk_guard = 0xBAAAAAAD;//provide some magic numbers
+     dprintk("__stack_chk_guard reached\n");
+}
+
+void __stack_chk_fail(void)
+{
+ /* Error message */
+     dprintk("__stack_chk_fail reached\n");
+}// will be called when guard variable is corrupted
+
 /* control IDs */
 #ifndef HAVE__V4L2_CTRLS
 #define V4L2LOOPBACK_CID_BASE (V4L2_CID_PRIVATE_BASE)

```

weberk, 2021-02-12:
Whats now working is to copy over the camera stream of my v4linux driven device (video0) to the virtual loopback device (video1).
`gst-launch-1.0 v4l2src device=/dev/video0 ! v4l2sink device=/dev/video1
`
I was hunting for v4l2loopback, because I have read, that the virtual device is exposed as a UVC compliant device. This is not the case and therefore is also ignored by "cros-camera" on ChromeOS.

Now I am heading towards adopting uvcgadget for wrapping my v4l2 device as a UVC webcam using kernel modules
```
/lib/modules/5.4.75-brunch-sebanc/kernel/drivers/usb/gadget/legacy/g_webcam.ko
/lib/modules/5.4.75-brunch-sebanc/kernel/drivers/usb/gadget/function/usb_f_uvc.ko

```

umlaeute, 2021-02-14:
thanks for your additional info.

just some notes:

>    In order to get rid of a linker error (undefined symbol: __stack_chk_guard) I added some dummy implementation I found on Google for such cases.

that seems to the wrong solution.
typically any symbols that starts with double-underscore (`__`) is *reserved* for the compiler/linker's internal usage.
by providing your own implementation you are fuddling with the internals, and are likely to break your system severely.

it might be that this is because your system (i think you mentioned ChromeOS)  applies special hardening to the kernel.
in this case i think that the build system of your system is simply broken, as it doesn't export the necessary compiler and/or linker flags to properly build modules.
you should make a bug at your distribution (rather than here).

 >   In order to get the kernel check the symbols of the new module I had to add `depmod -A` to the Makefile.

alternatively you could have read the [README](https://github.com/umlaeute/v4l2loopback#install)

anyhow, good luck with your further pursuit.

closed 2021-02-14 by umlaeute