v4l2loopback/v4l2loopback #403: [bug] Failing to build of Fedora34

aenertia, 2021-02-14:
#### Make fails on Fedora34 Beta with `cc1: error: incompatible gcc/plugin versions

  * `v4l2loopback` version: commit 10e1c61cde2a237d8eb61078edd1cb9bb1b6e8b9 (HEAD -> main, origin/main, origin/HEAD)
  * kernel version: 5.11.0-0.rc7.149.fc34.x86_64
  * Distribution (+version): Fedora 34

#### Steps to reproduce:

  1. Pull Fresh git clone
  2. Run Make
  
#### Observed Results:

 ```
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/aenertia/localbuild/v4l2loopback modules
make[1]: Entering directory '/usr/src/kernels/5.11.0-0.rc7.149.fc34.x86_64'
  CC [M]  /home/aenertia/localbuild/v4l2loopback/v4l2loopback.o
cc1: error: incompatible gcc/plugin versions
cc1: error: failed to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[2]: *** [scripts/Makefile.build:279: /home/aenertia/localbuild/v4l2loopback/v4l2loopback.o] Error 1
make[1]: *** [Makefile:1811: /home/aenertia/localbuild/v4l2loopback] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.11.0-0.rc7.149.fc34.x86_64'
make: *** [Makefile:44: v4l2loopback.ko] Error 2
```

umlaeute, 2021-02-14:
> cc1: error: incompatible gcc/plugin versions

looks like you are using the wrong compiler (e.g. not the one that was used for building the kernel).

This is explained in the [README](https://github.com/umlaeute/v4l2loopback/#dependencies)

closed 2021-02-14 by umlaeute

aenertia, 2021-02-14:
Grub really wanted to stick to the fc33 I had to manually move it to the fc34 one. I wonder if that is related. Possibly the fc34 kernel is not matched to the userspace in the beta.