v4l2loopback/v4l2loopback #404: OBS Virtual Cam doesn't activate (Ubuntu 20.04)

Strecx, 2021-02-19:
I installed v4l2loopback-dkms in order to add the Virtual cam option to obs. But when I try to activate the virtual cam it doesn't actually activate.
[image]
[image]
[image]
I already reported this bug in OBS Linux support, but they told me to report the bug here.

Strecx, 2021-02-19:
This is what happens to me when trying to load the v4l2loopback module 
[image]

umlaeute, 2021-02-19:
please paste error messages instead of screenshots of them.

the screenshot adds zero information and cannot be searched. on my mobile github client it doesn't show at all.
such screenshots are therefore discouraged (that's why this is explicitly mentioned in the template form).

apart from that: the template that opens for every new issue asks for a number of specific things to help triage any problem, like your system,kernel and module version.

it seems I (or you) those questions.

finally: please paste the output of `dmesg` when trying to load the module (as text).

umlaeute, 2021-02-19:
ah, I now see there's a bit of info in the subject.
it doesn't tell me the kernel though (and I'm lazy enough to ask you to look this up yourself).

Strecx, 2021-02-19:
modprobe: ERROR: could not insert 'v4l2loopback': Operation not permitted

umlaeute, 2021-02-19:
thanks for the text-output.
you might want to *replace* the image in https://github.com/umlaeute/v4l2loopback/issues/404#issuecomment-782124974, though

however

1. there's still

    >  it doesn't tell me the kernel though (and I'm lazy enough to ask you to look this up yourself).

2. and speaking of which: I don't know which version of `v4l2loopback` is buried in the `v4l2loopback-dkms` packge of Ubuntu-20.04. please specify

2. and finally:

    > finally: please paste the output of dmesg when trying to load the module (as text).

umlaeute, 2021-02-20:
btw, with "the output of `dmesg`" I really meant: the *relevant* output of `dmesg` ( the lines that are added when you try to `modprobe` the module.

ewennedelec, 2021-04-08:
Hi,

First, thanks a lot for your great job.

I am facing a very similar issue. 

My configuration : Ubuntu 18.04 LTS - Kernel 5.4.0-70-generic

I have installed OBS 26.1.1 and I want to use the vitual cam option.

Therefore, I have installed v4l2loopback-dkms :
$ sudo apt install v4l2loopback-dkms

I got this output :
"
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  v4l2loopback-dkms
0 upgraded, 1 newly installed, 0 to remove and 8 not upgraded.
Need to get 23.3 kB of archives.
After this operation, 112 kB of additional disk space will be used.
Get:1 http://ch.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 v4l2loopback-dkms all 0.10.0-1ubuntu1.2 [23.3 kB]
Fetched 23.3 kB in 0s (115 kB/s)             
Selecting previously unselected package v4l2loopback-dkms.
(Reading database ... 272714 files and directories currently installed.)
Preparing to unpack .../v4l2loopback-dkms_0.10.0-1ubuntu1.2_all.deb ...
Unpacking v4l2loopback-dkms (0.10.0-1ubuntu1.2) ...
Setting up v4l2loopback-dkms (0.10.0-1ubuntu1.2) ...
Loading new v4l2loopback-0.10.0 DKMS files...
Building for 5.4.0-70-generic
Building initial module for 5.4.0-70-generic
Secure Boot not enabled on this system.
Done.

v4l2loopback:
Running module version sanity check.
 - Original module
 - Installation
   - Insta_lling to /lib/mod_ules/5.4.0-70-generic/updates/dkms/

depmod...

DKMS: install completed.
"

At this stage, OBS displays the virtual cam button.
When pressing the "start virtual cam" button, I am prompted to enter the admin password, but then, nothing happens (and vitual cam is not started).

When trying to run v4l2loopback in terminal using the following command : 
sudo modprobe v4l2loopback video_nr=10 card_label="OBS Video Source" exclusive_caps=1

I get the following error message :
modprobe: ERROR: could not insert 'v4l2loopback': Bad address

Here is the dmesg output :
"
[jeu avr  8 08:57:12 2021] ------------[ cut here ]------------
[jeu avr  8 08:57:12 2021] WARNING: CPU: 7 PID: 6142 at /build/linux-hwe-5.4-p6lk6L/linux-hwe-5.4-5.4.0/drivers/media/v4l2-core/v4l2-dev.c:863 __video_register_device+0x54/0x6e0 [videodev]
[jeu avr  8 08:57:12 2021] Modules linked in: v4l2loopback(OE+) rfcomm thunderbolt ccm cmac bnep snd_hda_codec_hdmi intel_rapl_msr mei_hdcp intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm iwlmvm nls_iso8859_1 mac80211 crct10dif_pclmul crc32_pclmul ghash_clmulni_intel libarc4 aesni_intel crypto_simd cryptd glue_helper snd_hda_codec_realtek rapl snd_hda_codec_generic ledtrig_audio iwlwifi intel_cstate snd_hda_intel snd_intel_dspcfg snd_usb_audio snd_hda_codec joydev snd_usbmidi_lib snd_hda_core serio_raw intel_wmi_thunderbolt mxm_wmi cfg80211 wmi_bmof snd_hwdep snd_seq_midi snd_pcm snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer btusb uvcvideo btrtl btbcm videobuf2_vmalloc btintel videobuf2_memops videobuf2_v4l2 bluetooth videobuf2_common snd videodev input_leds mei_me ecdh_generic mc ecc soundcore mei acpi_pad mac_hid nvidia_uvm(OE) sch_fq_codel tuxedo_keyboard(OE) parport_pc ppdev lp parport ip_tables x_tables autofs4 btrfs xor zstd_compress raid6_pq
[jeu avr  8 08:57:12 2021]  libcrc32c dm_mirror dm_region_hash dm_log hid_generic usbhid hid nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) drm_kms_helper nvme syscopyarea psmouse sysfillrect sysimgblt fb_sys_fops sdhci_pci drm cqhci nvme_core ahci alx sdhci mdio libahci wmi video
[jeu avr  8 08:57:12 2021] CPU: 7 PID: 6142 Comm: modprobe Tainted: P        W  OE     5.4.0-70-generic #78~18.04.1-Ubuntu
[jeu avr  8 08:57:12 2021] Hardware name: Notebook                         P7xxDM3(-G)                     /P7xxDM3(-G)                     , BIOS 1.06.09 03/20/2017
[jeu avr  8 08:57:12 2021] RIP: 0010:__video_register_device+0x54/0x6e0 [videodev]
[jeu avr  8 08:57:12 2021] Code: 48 8b 87 90 04 00 00 49 89 ff 48 85 c0 0f 84 80 05 00 00 83 fe 03 89 d3 41 89 ce 4d 89 c4 74 68 8b 8f b0 01 00 00 85 c9 75 16 <0f> 0b 41 bc ea ff ff ff 5b 44 89 e0 41 5c 41 5d 41 5e 41 5f 5d c3
[jeu avr  8 08:57:12 2021] RSP: 0018:ffff9f5fca10fbf0 EFLAGS: 00010246
[jeu avr  8 08:57:12 2021] RAX: ffff8e776b894000 RBX: 000000000000000a RCX: 0000000000000000
[jeu avr  8 08:57:12 2021] RDX: 000000000000000a RSI: 0000000000000000 RDI: ffff8e76bb40b800
[jeu avr  8 08:57:12 2021] RBP: ffff9f5fca10fc18 R08: ffffffffc23a2180 R09: ffff8e7778407640
[... 141 lines cut ...]
[jeu avr  8 08:57:12 2021]  sysfs_remove_file_ns+0x15/0x20
[jeu avr  8 08:57:12 2021]  device_remove_file+0x15/0x20
[jeu avr  8 08:57:12 2021]  free_devices+0xa6/0x103 [v4l2loopback]
[jeu avr  8 08:57:12 2021]  init_module+0x6df/0x1000 [v4l2loopback]
[jeu avr  8 08:57:12 2021]  ? 0xffffffffc23a6000
[jeu avr  8 08:57:12 2021]  do_one_initcall+0x4a/0x200
[jeu avr  8 08:57:12 2021]  ? _cond_resched+0x19/0x40
[jeu avr  8 08:57:12 2021]  ? kmem_cache_alloc_trace+0x213/0x230
[jeu avr  8 08:57:12 2021]  do_init_module+0x5f/0x22e
[jeu avr  8 08:57:12 2021]  load_module+0x1d1c/0x2190
[jeu avr  8 08:57:12 2021]  __do_sys_finit_module+0xfc/0x120
[jeu avr  8 08:57:12 2021]  ? __do_sys_finit_module+0xfc/0x120
[jeu avr  8 08:57:12 2021]  __x64_sys_finit_module+0x1a/0x20
[jeu avr  8 08:57:12 2021]  do_syscall_64+0x57/0x190
[jeu avr  8 08:57:12 2021]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[jeu avr  8 08:57:12 2021] RIP: 0033:0x7f2dad697639
[... cut ...]
[jeu avr  8 08:57:12 2021] RAX: ffffffffffffffda RBX: 000055ba5de33df0 RCX: 00007f2dad697639
[jeu avr  8 08:57:12 2021] RDX: 0000000000000000 RSI: 000055ba5de344c0 RDI: 0000000000000003
[jeu avr  8 08:57:12 2021] RBP: 000055ba5de344c0 R08: 0000000000000000 R09: 000000000000003a
[jeu avr  8 08:57:12 2021] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000000000
[jeu avr  8 08:57:12 2021] R13: 000055ba5de33f00 R14: 0000000000040000 R15: 000000000000003a
[jeu avr  8 08:57:12 2021] ---[ end trace bb2c5b14ec38ac70 ]---
"

umlaeute, 2021-04-09:
@ewennedelec please try a newer version of `v4l2loopback`.

- Ubuntu/focal comes with [v4l2loopback-0.12.3](https://packages.ubuntu.com/focal-updates/v4l2loopback-dkms)
- Ubuntu/groovy comes with [v4l2loopback-0.12.5](https://packages.ubuntu.com/groovy/v4l2loopback-dkms)

There are no external dependencies of *v4l2loopback*, so you should be able to install the packages from the newer releases without any problems (just download the `.deb` files manually and install them).

ewennedelec, 2021-04-09:
Hi,

Thanks a lot for your very appreciated support.
I could download and install the deb file of the v4l2loopback-0.12.5 version.

I now get a "Start vitual cam" button on OBS working, which creates an OBS Virtual Camera device (so things are already better).

v4l2-ctl --list-devices gives me the following output :

"
OBS Virtual Camera (platform:v4l2loopback-000):
	/dev/video2

Chicony USB 2.0 Camera: Chicony (usb-0000:00:14.0-10):
	/dev/video0
	/dev/video1
"

But I do not get any output from this device. 

Trying with Guvcview, I get the following message : 
"
Guvcview error
Could not start a video stream in the device
"
Trying with Jitsi meet, the OBS device is not proposed within the list.

Another strange symptom is that my build-in webcam is not accessible anymore after having hit the Start Virtual Cam button once (it's OK before).

champost, 2021-04-28:
@umlaeute : Excellent suggestion to use the .deb files and most importantly the mention of "_no external dependencies_". I was thus able to get v4l2loopback-0.12.5 running on Bionic Beaver (18.04 LTS) using the Groovy (20.10) .deb packages !

umlaeute, 2021-10-09:
Then it is a problem with Ubuntu specific modifications (see the version number), and you should report the problem *there*.

ccoenen, 2021-11-19:
in my case, the virtual camera from OBS only works, if I unload the module (`rmmod`) and let obs load it itself (start virtual camrea button will ask for sudo password). I have not yet had the time to dig into this further, or if it is the exact same issue, though.

vicamo, 2021-12-28:
Copy my comments from https://bugs.launchpad.net/ubuntu/+source/v4l2loopback/+bug/1946660/comments/9:
The first open has:
```
[17047.368576] v4l2-loopback[1992]: opened dev:00000000c093a43d with image:0000000000000000
[17047.368577] /var/lib/dkms/v4l2loopback/0.12.5/build/v4l2loopback.c:1993[v4l2_loopback_open]
[17047.368582] video10: VIDIOC_QUERYCAP: driver=v4l2 loopback, card=OBS Video Source, bus=platform:v4l2loopback-000, version=0x00050f07, capabilities=0x
85200002, device_caps=0x05200002
```

The second open has:
```
[17091.427287] /var/lib/dkms/v4l2loopback/0.12.5/build/v4l2loopback.c:1964[v4l2_loopback_open]
[17091.427292] v4l2-loopback[1992]: opened dev:00000000c093a43d with image:0000000000000000
[17091.427293] /var/lib/dkms/v4l2loopback/0.12.5/build/v4l2loopback.c:1993[v4l2_loopback_open]
[17091.427297] video10: VIDIOC_QUERYCAP: driver=v4l2 loopback, card=OBS Video Source, bus=platform:v4l2loopback-000, version=0x00050f07, capabilities=0x85200000, device_caps=0x05200000
[17091.427304] video10: VIDIOC_G_FMT: error -22: type=vid-out, width=4294966992, height=4294967295, pixelformat=.... little-endian (0x00000000), field=any, bytesperline=1401200640, sizeimage=4288352138, colorspace=-744513136, flags=0xfffffed0, ycbcr_enc=0, quantization=0, xfer_func=0
```

As indicated in the capabilities field returned from the `VIDIOC_QUERYCAP` call, 0x85200000 means no `V4L2_CAP_VIDEO_CAPTURE`(0x01) neither `V4L2_CAP_VIDEO_OUTPUT`(0x02) is set. And since it reports neither, kernel `check_fmt` in *drivers/media/v4l2-core/v4l2-ioctl.c* will evaluate variable **is_vid** to false, and therefore returns `-EINVAL`.

This is a direct result from https://github.com/umlaeute/v4l2loopback/commit/9a77eb6 merged in https://github.com/umlaeute/v4l2loopback/pull/415.

This behavior is actually a bug in the upstream v4l2loopback source, function v4l2_loopback_write:
```
/* there's at least one writer, so don'stop announcing output capabilities */
dev->ready_for_output = 0;
```
Which set `dev->ready_for_output` to zero without correcting its `dev->type`.

vicamo, 2021-12-29:
> This behavior is actually a bug in the upstream v4l2loopback source, function v4l2_loopback_write:
> ```
> /* there's at least one writer, so don'stop announcing output capabilities */
> dev->ready_for_output = 0;
> ```
> Which set `dev->ready_for_output` to zero without correcting its `dev->type`.

To resolve this, it involves the designed behavior of read/writing a fd and their effects on `dev->type`. Again, in kernel V4L2 drivers that utilize videobuf2 framework (and it should be), a V4L2 fd cannot be both capturing and outputting using streaming or file I/O. The definition of the right behaviors for v4l2loopback will define how this issue should be solved, and that's also the ultimate problem I want to address in #359.

vicamo, 2022-01-04:
> To resolve this, it involves the designed behavior of read/writing a fd and their effects on `dev->type`. Again, in kernel V4L2 drivers that utilize videobuf2 framework (and it should be), a V4L2 fd [...]

#457 targets similar thing. Multiple simultaneous streaming (2 reads, 2 writes, read & write) is not allowed in a formal Linux V4L2 driver. And to correctly set the right states for this issue, I think this must be enforced to avoid creating further troubles.

rciorba, 2022-03-01:
> in my case, the virtual camera from OBS only works, if I unload the module (`rmmod`) and let obs load it itself (start virtual camrea button will ask for sudo password). I have not yet had the time to dig into this further, or if it is the exact same issue, though.

In my case the module was not loaded at all. Loading it (`modprobe v4l2loopback`) was enough to get the virtual camera working in OBS. Thanks for your comment @ccoenen, otherwise I probably wouldn't have thought to try this.

WolfdaleA3, 2022-07-22:
Strangely enough, I've only experienced this issue just recently after upgrading my PC and moving to Jammy. This hasn't happened to me while I was using the OBS virtual camera in Focal. I could toggle it on and off repeatedly without any problems. It does ask for sudo for the first activation though.

stephematician, 2024-10-18:
Is this still an issue? Running Ubuntu 22.04, OBS 30.2.2, v4l2loopback (latest), I can start and stop a Virtual Camera no problem. I can watch the Virtual Camera with `ffplay /dev/video2`.