v4l2loopback/v4l2loopback #405: Firefox: "Used by" miscounting / "modprobe: FATAL: Module v4l2loopback is in use."

T-X, 2021-02-24:
### Environment:

  * `v4l2loopback` version: 10e1c61

          sudo dmesg  | grep -i v4l2loopback
          [328911.997193] v4l2loopback: loading out-of-tree module taints kernel.
          [328911.998540] v4l2loopback driver version 0.12.5 loaded

  * kernel version: 5.7.19

          uname -a
          Linux linus-work 5.7.19 #7 SMP Sun Nov 8 00:36:09 CET 2020 x86_64 GNU/Linux

  * Distribution (+version): Debian Sid

          lsb_release -a
          No LSB modules are available.
          Distributor ID: Debian
          Description:    Debian GNU/Linux bullseye/sid
          Release:        unstable
          Codename:       sid

  * Browser: Firefox  83.0 (64-bit)

### Step 3: Describe the problem:

After trying to access the video device created by v4l2loopback via Firefox I can't unload the kernel module anymore. The "Used by" decrements only once after closing the according browser tab, but incremented 3 times when trying to access the device. Only closing all Firefox processes reduces the "Used by" count back to zero.

#### Steps to reproduce:

  1. Load v4l2loopack via "modprobe v4l2loopback"
  2. Start Firefox
  3. Check 'lsmod | grep "^v4l2loopback"', it should look like `v4l2loopback           49152  0`
  4. Open a website which tries to access a webcam, like https://meet.ffmuc.net/v4l2loopback-test-bug (a Jitsi meet instance)
  5. Select the v4l2loopback device as "Camera to share" and click "Allow".
  6. Quickly check the "Used by" counter in 'lsmod | grep "^v4l2loopback"', it increased by 1: `v4l2loopback           49152  1`
  7. Wait a few seconds until Jitsi complains with "Failed to access your camera".
  8. Check the "Used by" counter again, it increased by 2: `v4l2loopback           49152  3`
  9. Close the browser tab, the "Used by" counter only decreases by one: `v4l2loopback           49152  2`
  10. Try to unload v4l2loopback via "modprobe -r v4l2loopback", it will complain with `modprobe: FATAL: Module v4l2loopback is in use.`
  11. Kill all Firefox instances via "killall -s KILL firefox": Now the used by count is 0 again. The module can now be successfully unloaded.
  
#### Observed Results:

  * The v4l2loopback "Used by" counter increased by 3 and only decreased by one after closing the browser tab.
  
#### Expected Results:

  * The v4l2loopback "Used by" counter should decrease by the same amount it was incremented after closing the according browser tab which created the increments.
  
I'm not 100% sure if this is a bug in Firefox, if Firefox in fact keeps hogging the device. Or if v4l2loopback might be miscounting or misinterpreting the events triggered by Firefox. Or might be calling some wrong Linux ABI functions to free the usage?

umlaeute, 2021-02-24:
i'd rather suspect that firefox is the culprit.
after all: other applications (that focus more on video) like VLC, ffmpeg, gstreamer,... don't show this problem.

the fact that killing firefox frees the dangling ressources, seems to back this theory.

(of course it's easier to point fingers than to admit bugs)

how does chrome handle this?

umlaeute, 2021-03-18:
i just checked with some normal client (like pd/Gem), where I know that I can properly close the device with the application still running, and the device is freed.

also: v4l2loopback operates entirely in kernel-space. if the handle count would be broken, a simple kill of the a user-space application wouldn't fix the problem (and instead you would need a reboot or at least a forced module-unload).

so I'm closing this as i don't think this is an issue in `v4l2loopback`.
(if you have evidence that the module is the culprit, do not hesitate to ping me back here so I can re-open the issue)

closed 2021-03-18 by umlaeute