v4l2loopback/v4l2loopback #406: Implement videobuf2 api (need to remove support for older kernels)

wt, 2021-02-25:
### My environment

  * `v4l2loopback` version: 0.12.5 (latest git)
  * kernel version: 5.12.0-0.rc0.20210222git31caf8b2a847.158.fc35.x86_64
  * Distribution (+version): fedora rawhide

### v4l2loopback doesn't implement the videobuf2 api

According to #268, the videobuf2 api is a requirement for upstream consideration. I would like to remove support for kernels that do not support the videobuf2 api. I believe that 3.6.1 is a safe point to cut since I think it was released after the videobuf2 api landed. For the record, 3.6.1 was released in Oct 7, 2012. Removing support for prior kernels would allow the implementation of the videobuf2 api as a discrete step towards improving code quality for possible upstream acceptance while removing support only for extremely old kernels.

#### RFC

I would like to get feedback and see if this work is worth doing.

umlaeute, 2021-02-26:
go ahead.
i think we can safely purge support for kernel lines that have seen the EOL of the LTS branches.

wt, 2021-02-28:
Just so you know, I have local changes that cut kernel support at 3.6.1 and allocate, initialize, and free the vb2_queue objects. I'm currently working on migrating buffer allocation and free to the queue objects' implementations of the operations.