v4l2loopback/v4l2loopback #410: Does not compile on Fedora 34 with kernel 5.11

j1cken, 2021-03-08:
### Step 2: Describe your environment

  * `v4l2loopback` version: _____

```
'master' git pull from today
```

  * kernel version: _____

          uname -a

```
Linux fedora 5.11.3-300.fc34.x86_64 #1 SMP Thu Mar 4 19:03:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

  * Distribution (+version): _____

          lsb_release -a
```
LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	Fedora
Description:	Fedora release 34 (Thirty Four)
Release:	34
Codename:	ThirtyFour
```
  
### Step 3: Describe the problem:

#### Steps to reproduce:

  1. make clean && make
  
#### Observed Results:

  * What happened?  This could be a description, log output, etc.

```
rm -f *~
rm -f Module.symvers Module.markers modules.order
make -C /lib/modules/`uname -r`/build M=/home/torben/development/v4l2loopback clean
make[1]: Entering directory '/usr/src/kernels/5.11.3-300.fc34.x86_64'
make[1]: Leaving directory '/usr/src/kernels/5.11.3-300.fc34.x86_64'
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/home/torben/development/v4l2loopback modules
make[1]: Entering directory '/usr/src/kernels/5.11.3-300.fc34.x86_64'
  CC [M]  /home/torben/development/v4l2loopback/v4l2loopback.o
cc1: error: incompatible gcc/plugin versions
cc1: error: failed to initialize plugin ./scripts/gcc-plugins/structleak_plugin.so
make[2]: *** [scripts/Makefile.build:279: /home/torben/development/v4l2loopback/v4l2loopback.o] Error 1
make[1]: *** [Makefile:1801: /home/torben/development/v4l2loopback] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.11.3-300.fc34.x86_64'
make: *** [Makefile:43: v4l2loopback.ko] Error 2
```
  
#### Expected Results:

  * What did you expect to happen?

```
Successful compilation
```
  
#### Relevant Code:

  ```
see log output above
  ```

umlaeute, 2021-03-08:
> cc1: error: incompatible gcc/plugin versions

that looks very much like a problem with your installed gcc.
make sure you have the [same compiler used for building the kernel](https://github.com/umlaeute/v4l2loopback#dependencies)

can you build other out-of-tree modules?

umlaeute, 2021-03-08:
and please use a full commitish rather than "`master` git pull from today". *today* is a fickle thing, and it's not so much fun tracking down a problem if you first have to find out when the OP created the ticket.

kwizart, 2021-03-09:
https://ask.fedoraproject.org/t/fedora-34-beta-and-oot-kmod-nvidia-virtualbox-v4l2loopback-etc/12778

This should work:
dnf donwgrade gcc

closed 2021-03-09 by umlaeute