v4l2loopback/v4l2loopback #411: Support changing FPS without root

letmaik, 2021-03-09:
Currently, FPS can be changed with root permissions by writing to `/sys/devices/virtual/video4linux/video0/format`. Most other settings don't require root, like setting the frame size, pixel format etc. If technically possible, it would be nice to be able to set FPS without root.

umlaeute, 2021-03-09:
how about this?

~~~sh
sudo chgrp video /sys/devices/virtual/video4linux/video*/format
sudo chmod g+w   /sys/devices/virtual/video4linux/video*/format
~~~

umlaeute, 2021-03-09:
to make this persistent, you would use something like `udev`.

and, lo-and-behold! there's an [example udev rule](https://github.com/umlaeute/v4l2loopback/blob/main/udev/60-persistent-v4l2loopback.rules) in the repository that does *exactly* that.

https://github.com/umlaeute/v4l2loopback/blob/f62fb9076b6313e5eb82fdcaceadb6b3052f346e/udev/60-persistent-v4l2loopback.rules#L9-L12

closed 2021-03-09 by umlaeute

letmaik, 2021-03-09:
Is there a reason why this isn't the default? Seems if `/dev/videox` is in group `video` maybe the rest should also be?

umlaeute, 2021-03-09:
afaict, this is technically not possible.