v4l2loopback/v4l2loopback #416: Wrong data in `/sys/module/v4l2loopback/parameters/video_nr` (version 0.12.5)

tovrstra, 2021-03-17:
### Describe your environment

  * `v4l2loopback` version: 0.12.5
  * kernel version: `Linux localhost.localdomain 5.10.22-200.fc33.x86_64 #1 SMP Tue Mar 9 22:05:08 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`
  * Distribution (+version): Fedora 33
```
LSB Version:	:core-4.1-amd64:core-4.1-noarch
Distributor ID:	Fedora
Description:	Fedora release 33 (Thirty Three)
Release:	33
Codename:	ThirtyThree
```

### Describe the problem

#### Steps to reproduce

```bash
# unload v4l2loopback to have a clean start
modprobe -r v4l2loopback
# The modprobe line from the plugin
modprobe v4l2loopback exclusive_caps=1 card_label='OBS Virtual Camera'
```

#### Observed Results

Output of `cat /sys/module/v4l2loopback/parameters/video_nr` is `-1,-1,-1,-1,-1,-1,-1,-1`

Output of `v4l2-ctl --list-devices`:

```
OBS Virtual Camera (platform:v4l2loopback-000):
	/dev/video6
```
  
#### Expected Results

Output of `cat /sys/module/v4l2loopback/parameters/video_nr` should be `6,-1,-1,-1,-1,-1,-1,-1`

umlaeute, 2023-05-02:
In the meantime I'm no longer convinced that the expected behaviour is really to be expected... After all, the *module parameters* in your example did not set any device numbers explicitly, and if we assume that `/sys/module/v4l2loopback/parameters/video_nr` reflects the module parms than they should be like they currently are.

I figure that what you really want is a way to enumerate the loopback devices, which probably should be solved with the dynamic device management (`/dev/v4l2loopback`)

tovrstra, 2023-05-06:
It's been too long ago to remember the details, so I'm happy to trust your judgement.

closed 2023-05-06 by tovrstra