v4l2loopback/v4l2loopback #426: -Wformat= in `make` step at v4l2loopback-ctl compilation

NarvyAutonomus, 2021-04-26:
Sometimes, when I recompile (after a clean on the modules), the v4l2loopback-ctl fails at compilation; I had a lucky time when it not happened, but I am stuck again on this:

```
v4l2loopback-ctl.c:441:27: warning: format ‘%s’ expects argument of type ‘char *’, but argument 2 has type ‘struct v4l2_loopback_config *’ [-Wformat=]
  441 |   printf("configuration: %s\n", cfg);
      |                          ~^     ~~~
      |                           |     |
      |                           |     struct v4l2_loopback_config *
      |                           char *
v4l2loopback-ctl.c: In function ‘parse_caps’:
v4l2loopback-ctl.c:660:24: warning: format ‘%c’ expects argument of type ‘char *’, but argument 3 has type ‘char (*)[5]’ [-Wformat=]
  660 |  if (sscanf(buffer, "%4c:%dx%d@%d/%d", &fourcc, &caps->width,
      |                      ~~^               ~~~~~~~
      |                        |               |
      |                        char *          char (*)[5]

```
I tried a few changes, but I got nothing; how could I correct this in the source file I have in my computer?

umlaeute, 2021-04-26:
these are *warnings*, and as such they are not fatal to the build process.
unless you (or something else; but nothing on the side of this repository) marks warnings as errors with `-Werror`.

i'm closing this, as not actable upon.
if you can come up with actual *error*s, please don't hesitate to add them to this ticket and I will consider re-opening it.

closed 2021-04-26 by umlaeute

BenBE, 2021-04-26:
Looking at https://github.com/umlaeute/v4l2loopback/blob/f62fb9076b6313e5eb82fdcaceadb6b3052f346e/utils/v4l2loopback-ctl.c#L437-L444 I see a straight forward type confusion. You are writing memory from a structure, when you'd actually want to format the structure's content into something readable for the user.

For the second warning you'd probably want to use `%4s` instead of the `%4c` the compiler complains about. Also the `&fourcc` should likely just be `fourcc`.

Please take warnings seriously.

umlaeute, 2021-04-26:
> Oh, and on second glance I also see a NULL dereference

ah well, i figure that `"%s", 0` would print as `(null)`. but of course this nevertheless should read `%p` instead of `%s`.

> For the second warning you'd probably want to use `%4s` instead of the `%4c`

most certainly not. the FOURCC string can contain (trailing) spaces (e.g. `V4L2_PIX_FMT_Y16`)

> Also the `&fourcc` should likely just be `fourcc`.

yes.

NarvyAutonomus, 2021-04-26:
the `%p` for `cfg` at line 441 and `fourcc` instead of `&fourcc` at line 660 worked for compiling the code. Thank you.