v4l2loopback/v4l2loopback #427: Enhancement: Revisit change resolution on-the fly, also solves producer disconnect.

xxxajk, 2021-04-26:
* `v4l2loopback` version: github-master-pull
  * kernel version: 5.11.x 5.8.x
  * Distribution (+version): mint/ubuntu mixed
### Step 3: Describe the problem:
Can't have a query return multiple supported resolutions set by producer, and be able to request a size change.

From brief Email discussion:
```
The producer could just query every so often using the IOCTL VIDEOC_G_FMT, when there is a change requested, it will be able to see the change, and do what it needs to do using VIDEOC_S_FMT.

On the consumer side -> request change -> we make a note to pass to the producer. VIDEOC_G_FMT will act like nothing has happened YET and will report back the OLD resolution.

So how do we know which FD is a producer? That's pretty easy, the producer can set this as a fact by using VIDEOC_S_EXT_CTRLS and set a special control there, which can even be named :-) Since we will know which FD made the request, the rest is pretty simple.

```
Question from reply:
```
so how would that work for any standard-compliant consumer?
i mean: there's actual data flowing between the producer and consumer,
and the format of the data has been pre-negotiated.
no consumer actually polls the format before receiving a new frame.

and what if the producer does not query every so often?

one of the two will access out-of-bounds memory.

i don't see how this could work in practice.

```

- On first start, use a sane default (which is 640x480, etc currently), We can use a special VIDEOC_[G|S]_EXT_CTRLS to check /populate/update a list of formats/resolutions/etc to give the query choices to the program making the request. These IOCTLs are restricted to the producer only, and won't show for the consumer, unless we add capability there. One nice thing that we can do is also have the ability to add extra controls on the fly, such as setting focal distance on a motorized focus camera, setting auto focus, and other settings. The producer can check what is there already to sync in the event it gets restarted. If a control isn't listed, the producer will know to populate the list with whatever defaults.
- A blank or placeholder frame for the requested resolution. This should also be done if there isn't a producer, instead of just a timeout. There's no option for a plain blank, only for an image currently, and it seems that it is required to fit the resolution. A blank screen option that takes an html color number would be ideal, or, just what many other capture cards do when there is no signal, the blue screen of dispair.
- As soon as the consumer makes a request, blank frames are sent as above, until producer sets the new values. All pending and incoming frames are disposed of/ignored, and prevents any out-of-bounds memory issues, etc.
- If producer thread monitoring for changes fails, the consumer will get the blank screen, if the producer is restarted, the producer uses the setting from VIDEOC_G_FMT instead of a preset in the producer. This keeps things in sync with the kernel and requests, and prevents any out-of-bounds issues.

How to tell old v.s. new producers.
Old producers won't be affected by the changes, since there's nothing new in order to stream. Simply lock everything if there's no request for the mode using  VIDEOC_S_EXT_CTRLS. This will also allow an interesting trick where a consumer and producer can effectively swap roles. Currently producers only open in WO mode, consumers in RO mode. Allowing an RW mode would allow role swapping if it is found to be useful, but RW mode can be the difference to say we are a new-style producer. Might be a useful? Not sure.
New producers can check if they can use the new IOCTLs, and set modes. If this fails, it can revert back and spew a warning, or bail out.

xxxajk, 2025-03-07:
It can *query* the format, but can't renegotiate after? 
What if you close and re-open? 
The problem is that the format is fixed to a single value, and the enumeration is a list of one format. There is no way to ask for any more than one format, which should be possible to do, and is the entire point.

umlaeute, 2025-03-07:
What is "it"?

Obviously, a v4l2 client can query the format and re-negotiate it. 

However, once a client has negotiated a format, the device must keep its promise and deliver the data in the negotiated format. 

Or to put otherwise: only the client can initiate a format renegotiation.

So if a client has negotiated a format, or hold a lock on the format. 

With the loopback device, we have two clients.
As soon as one has negotiated a format, the other cannot change it, until all format locks are released again... which basically means that the other client has to close the device to release the lock.

So the problem is, that you need a way to tell the other client to release the lock, and v4l2 (the in kennel framework, not this driver) does not provide sich a way.

stephematician, 2025-03-07:
Agree with umlaeute.

It seems, to me, like something that should be negotiated between client applications (e.g. producer and consumer) directly with each other; rather than via the loopback node.

xxxajk, 2025-03-08:
Well, since producer and consumer are already talking to each other, and are aware of the connection, what's preventing forwarding the the producer the request from the consumer that was originated by the v4l2 client? 

It would seem to me that this would be totally possible, since the consumer could forward the queries... 
That request for format or video frame size or whatever could be forwarded to the producer, and if the producer doesn't understand the request, no big deal, you ignore it and keep pumping the same just as usual. Even if this would be something out-of-band, or an ioctl to check, it should be totally possible, since software is malleable. The producer should also be able to tell the consumer (kernel input side of video) what it can do and enumerate it as well.

stephematician, 2025-03-08:
It is possible but I don't think it's practical.

If the consumer and producer want to talk; there are plenty of IPC approaches available that would be cleaner than using an IOCTL of a V4L2 node.

Maybe take this back a step: What do you want to be able to do with a loopback node that you can't do already?

xxxajk, 2025-03-09:
Application gets a list of formats. Currently it's pretty much locked down to a single resolution.
Application requests resolution wanted, this again is locked to whatever the producer is sending.
Application wants to change resolution -- it can't, since only one is listed.
What should happen is that the producer should be able to give a list of available resolutions to pass to the application, which can ask for a change. This would be no different than requesting a change from 1k to 2k resolution on a capture board or USB camera. 

The problems are:
There is no way to tell the input side of loopback what all available resolutions are.
There is no way to tell the input side of loopback what all available formats  are at the input, e.g. YUYV, RGB, mjpeg.
There is no way for the output side of the loopback to relay back to the input side any kind of resolution changes.
There is no way for the output side of the loopback to relay back to the input side any kind of codec (format) changes.

Usually there is no need to change formats, but in some specific cases it can matter, e.g. simulating zoom by cutting resolution. Some cams can do this, they just drop resolution and clip the frame to fake zooming.

stephematician, 2025-03-09:
> What should happen is that the producer should be able to give a list of available resolutions to pass to the application, which can ask for a change.

There is nothing stopping any existing producer application from implementing an API that would allow a consumer application to request a change in format. It doesn't need loopback IOCTLs for this.

I still think that, by far, the producer application is the best place to handle this because it would be the _most_ aware of what the source's constraints are and what the producer itself _can produce_.

umlaeute, 2025-03-10:
> Think more about this as a networking loopback device, they don't only communicate in a single direction. It's should be the same with video, regardless of it's source, which, it isn't at present

sure, you might wish for that.
but the `v4l2` API does not provide a means for it.

there is simply no way to make your suggestion work with a *general* v4l2 consumer (or a general v4l2 producer).
so you need to write special code in your clients to support the on-the-fly format changing.

but if both your consumer and your producers need special code to support that feature, there are already other solutions that allow you to do this (and which run in user-space rather than kernel-space!).

Just use pipewire!

closed 2025-03-10 by umlaeute

xxxajk, 2025-03-10:
And this is why I stopped doing kernel... Nobody actually sees what things are, and really don't for whatever reason understand plain language anymore.

stephematician, 2025-03-10:
I think you've probably missed the point of my question, too. I don't mind what you are proposing conceptually about what  loopback "should" be - but it's very subjective.

It would be easier if you told me something like "I am writing application XYZ and I need to be able to do ABC". Then I would understand where you are stuck and how to find a solution for missing functionality. Instead, what I'm reading is "I want this API, because it should be this way".

umlaeute, 2025-03-10:
> And this is why I stopped doing kernel... Nobody actually sees what things are, and really don't for whatever reason understand plain language anymore.

I honestly think that I understand what you are trying to say.

it's just that I *also* think that this is out-of-scope for this project.

(and I totally fail to see what is wrong with using pipewire for your use case)

stephematician, 2025-03-10:
Thanks for the further explanation.

Probably the only way forward for this - if you want loopback node-mediated producer-consumer format negotiation; would be your own fork. I'd recommend waiting a bit as there might be some further strengthening of compliance of _v4l2loopback_ by using [_videobuf2_](https://www.kernel.org/doc/html/latest/driver-api/media/v4l2-videobuf2.html) buffers; otherwise there is a potential in-tree version: [linux-video-loop](https://codeberg.org/jarkko/linux-video-loop) lead by others).

A loopback device, to me, is a way to present a stream of video as a V4L2 (compliant) device. It isn't a full or partial virtualisation of some underlying physical (or otherwise) device. That means that, from my perspective, it doesn't _have_ to provide anything other than what it needs to provide the stream from one _application_ (the producer) to another _application_ (the consumer) in line with the V4L2 UAPI.

In your example:

1.  We have an IP camera that produces UDP video.
2.  Some _application_ (producer) negotiates a format with the loopback node (e.g. _gstreamer_, or perhaps your own? you didn't specify).
3.  The producer sends (stream) the formatted video to the loopback node.
4.  Another _application_ (consumer) can access the stream as through it is a V4L2-compliant video device;
    -   It has to accept the stream in the existing format because "loopback" is almost always synonymous with "unmodified".

I don't see any missing functionality given my understanding of what a loopback feed _should_ be; i.e. a representation of a stream of video as a V4L2-compliant video device.

stephematician, 2025-03-11:
FWIW @xxxajk : I would implement/design it a little differently:

1.  The producer, that hasn't S_FMT'd or REQBUFS (allocated buffers), gets exclusive priority (I think this first part is safest if _no one_ else is querying the node).
2.  The producer would query the loopback device (node) if 'format forwarding' is available via an EXT_CTRL.
3.  The producer then 'activates' format forwarding (another EXT_CTRL), and uses another set of EXT_CTRLS for 'setting' the available formats:
    -   They'd maybe have names like S_ENUM_FWD_FMT, S_ENUM_FWD_FRAMESIZES, and probably S_ENUM_FWD_FRAMEINTERVALS.
    -   These will control what is returned by ENUM_FMT, ENUM... for both the producer and consumer.
4.  The producer could then lower its priority so others can query the device, it can then select a format and start a stream as usual.
5.  Eventually a consumer queries and set formats as though the node were any other device.
6.  When the consumer REQBUFS (or maybe S_FMT) while a producer is streaming a _different_ format (and format forwarding is active):
    -   An "orphaned producer stream" state is set that can be accessed (by the producer) via an IOCTL.
    -   The node short-circuits streaming IOCTLs for the producer except for STREAMOFF and REQBUFS with count equal to zero (i.e. release buffers).
    -   Once the producer releases buffers; the "orphaned producer stream" state is cleared.
    -   The producer can poll the "orphaned producer" state; and it can respond accordingly (i.e. release buffers, etc).
7.  A producer, provided no buffers are allocated (?), can deactivate format forwarding (possibly only when it has exclusive priority?).

stephematician, 2025-03-11:
You could also clear the "orphaned producer stream" if the consumer closes the fd or switches to the producers _existing_ format.

A producer can take care of any "in progress" images if there is a lengthy delay in changing formats and it can do that work asynchronously.

umlaeute, 2025-03-11:
>  I would implement/design it a little differently:

i still totally fail to see why you would want your applications (both producer and consumer) to limit themselves to a very specific implementation that supports some `EXT_CTRL`s that no one else is using, for something that is already solved  in PipeWire.

whatever clever scheme we can come up with, applications would need to explicitly implement support for it.
imho, that time is better spent implementing support for a framework that natively supports what you want (and more, like a built-in tee)

or to put otherwise: `v4l2loopback` is mostly useful for feeding arbitrary video streams into applications that are impractical to extend beyond their existing support for v4l2.

xxxajk, 2025-03-21:
@umlaeute so what you are basically saying is that it's not extendable because it's not necessarily connected to actual hardware, which technically shouldn't be an issue whatsoever. However the problem occurs when it is, and that I'd have to duplicate everything in the new driver set to emulate something that's already 50% emulated. 

If it is to look no different to the app that is processing the output, which normally has multiple settings, but currently can never have, then it's only a half implementation. No special IOCTLs would actually need to be added to any old producer. If the producer side COULD give one OR MORE selections, and the app gets to pick then it would need a renegotiation request when it changes, and that would be driven by the app doing the request. Again, this would not affect anything already in existence, since this already happens, but its limited to a single choice made during the beginning instead of all possible choices. It would seem rather obvious to me that if only one was given, then there's only one choice, and we're done, no change to anything on either side. No current producer code would need to know anything new, nothing broken, still only one selection. 

If there's multiple choices, then obviously there's something that can accept a renegotiation, and that would/could be done via an IOCTL that already exists in v4l2... how else does an app pick new resolutions and whatnot? That's easy, it uses an IOCTL that already exists, and the hardware changes the output. This action  won't exist or even occur if there is only one choice, nothing is passed back to the producer, we'd know that, the producer did the setup, offered only one. In the case of multiple choice, the IOCTL request could simply be looped back to the producer, and that would give you a full loopback. No new IOCTL needed, instead it is just passed to the producer, which, since it has multiple choices would be written to understand that something like that has to be polled for. This isn't all that difficult to understand.