v4l2loopback/v4l2loopback #443: Ubuntu - Error loading v4l2loopback with Secure Boot (Solution)

CdtDelta, 2021-09-01:
This isn't an issue/bug but if you want to include it in the documentation for the project here's how I was able to get this to work.

I'm running Ubuntu 18.04 and I've had a similar issue with VMWare and loading some of the drivers, and came across this:

https://askubuntu.com/questions/1096052/vmware-15-error-on-ubuntu-18-4-could-not-open-dev-vmmon-no-such-file-or-dire

I realized when I was looking through the reported issues for this project it seemed to be a similar issue.  There was a closed issue that pointed to a Github gist, but it appears that is gone. Although I did find another one with similar steps:

https://gist.github.com/lijikun/22be09ec9b178e745758a29c7a147cc9

Here's what I had to do in order to get it working...
(note the name OBS.priv and OBS.der are the names I just gave it, you can call it whatever you want)

The first six steps you only have to do once:
1. openssl req -new -x509 -newkey rsa:2048 -keyout OBS.priv -outform DER -out OBS.der -nodes -days 36500 -subj "/CN=OBS/"
2. sudo /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 ./OBS.priv ./OBS.der $(modinfo -n v4l2loopback)
3. tail $(modinfo -n v4l2loopback) | grep "Module signature appended"
4. sudo mokutil --import OBS.der
5. Reboot; You may have to go through a MOK enrollment process in the BIOS (I have to)
6. mokutil --test-key OBS.der (this just confirms it's enrolled)
7. sudo /usr/src/linux-headers-$(uname -r)/scripts/sign-file sha256 ./OBS.priv ./OBS.der $(modinfo -n v4l2loopback)

I don't know if step 7 is needed the first time (since you do it in step 2), but I had it in my notes for VMWare modules so I'm including it here.

Now every time there's a kernel update, or the v4l2loopback module is updated, you'll have to go through step 7 at least again and reboot. However this is the first time I'm using the v4l2loopback module so I'm not 100% sure if you have to reboot after each update.

umlaeute, 2021-09-01:
thanks for your research.

i'm reluctant to add this to the readme, but i think it would be a great addition to the [wiki](https://github.com/umlaeute/v4l2loopback/wiki) - feel free to add it!

closed 2021-09-01 by umlaeute

CdtDelta, 2021-09-01:
> but i think it would be a great addition to the [wiki](https://github.com/umlaeute/v4l2loopback/wiki) - feel free to add it!

Done!

umlaeute, 2021-09-02:
Since the Wiki is large, here's a [direct link](https://github.com/umlaeute/v4l2loopback/wiki/Secure-Boot)

thanks again.