v4l2loopback/v4l2loopback #448: Wrong incoming frame length

niXman, 2021-11-04:
hi,

I'm using ubuntu-20.04 with v4l2loopback-0.12.3.

I have two v4l2 loopback devices with the following characteristics:
```
ffprobe /dev/video0

Input #0, video4linux2,v4l2, from '/dev/video0':
Duration: N/A, start: 17479.219834, bitrate: 331776 kb/s
Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p, 1280x720, 331776 kb/s, 30 fps, 30 tbr, 1000k tbn, 1000k tbc
```
```
ffprobe /dev/video1

Input #0, video4linux2,v4l2, from '/dev/video1':
Duration: N/A, start: 17481.786751, bitrate: 117964 kb/s
Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p, 640x512, 117964 kb/s, 30 fps, 30 tbr, 1000k tbn, 1000k tbc
```
but I ran into a problem with the frame size in webrtc source code: https://webrtc.googlesource.com/src/+/refs/heads/main/modules/video_capture/video_capture_impl.cc#132

for I420 the frame size in bytes should be `1280×720+640×360×2 == 1382400` but was read 1384448...

IncomingFrame() called from here: https://webrtc.googlesource.com/src/+/refs/heads/main/modules/video_capture/linux/video_capture_linux.cc#411

`struct v4l2_buffer` filled here: https://webrtc.googlesource.com/src/+/refs/heads/main/modules/video_capture/linux/video_capture_linux.cc#398

but pay attention, webrtc works great for my second video stream! (/dev/video1: yuv420p, 640x512)
so I think the problem is not in the webrtc code.

any ideas?

best!

niXman, 2021-11-05:
looks like the problem is in gstreamer's v4l2sink plugin...

mariannasb, 2021-12-08:
can you comment on what you found on v4l2sink?

umlaeute, 2023-05-02:
Here's the relevant bugreport for the `v4l2sink`: https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/2532

commit 175adaf9 (2023-05-10): v4l2loopback: Fixup bytesused field when writer sends a too large value

Gstreamer's v4l2sink is known to submit buffers with bytesused set to
the length of the buffer instead of the size of the actual image-data
within the buffer which is typically smaller due to buffer sizes
being rounded op to a page-size:

https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/2532

Despite this being a long standing issue and their being 2 merge-reqs:

https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/3713
https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/4527

to  [... cut]
PR #543 (2023-05-10): v4l2loopback: Fixup bytesused field when writer sends a too large value
Gstreamer's v4l2sink is known to submit buffers with bytesused set to the length of the buffer instead of the size of the actual image-data within the buffer which is typically smaller due to buffer sizes being rounded op to a page-size:

https://gitlab.freedesktop.org/gstreamer/gstreamer/-/issues/2532

Despite this being a long standing issue and their being 2 merge-reqs:

https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/3713 https://gitlab.freedesktop.org/gstreamer/gstreamer/-/merge_requests/4527

to t [... cut]