v4l2loopback/v4l2loopback #454: GStreamer: All access to video loopback device close if one closes using VIDIOC_REQBUFS with count = 0 in teardown

dabrain34, 2022-01-03:
### Describe your environment

  * `v4l2loopback` version: 0.12.5(git version)

  * kernel version: 5.11.0-43-generic

  * Distribution (+version): Ubuntu 20.04

  
### Step 3: Describe the problem:

When running a multiple access to a loopback device and closing one GStreamer pipeline accessing the device will close all the other pipelines.

#### Steps to reproduce:

  1. sudo insmod v4l2loopback.ko video_nr="42" 'card_label=mccam' exclusive_caps=1 max_buffers=2
  2. gst-launch-1.0 -v -e videotestsrc ! video/x-raw,width=320,height=200 ! timeoverlay ! videoconvert ! v4l2sink device=/dev/video42
  3. Run Gstreamer playback pipelines to access the video device:
     1.  gst-launch-1.0 v4l2src device=/dev/video42 ! videoconvert !  autovideosink
     2.  gst-launch-1.0 v4l2src device=/dev/video42 ! videoconvert !  autovideosink
  
#### Observed Results:

  * When closing the second playback pipeline only, the first pipeline gets closed as well with this error.
```
0:00:04.948341830 28762 0x55eeb31e0400 ERROR          v4l2allocator gstv4l2allocator.c:1405:gst_v4l2_allocator_dqbuf:<v4l2src0:pool0:src:allocator> failed dequeuing a mmap buffer: Invalid argument
0:00:04.948410752 28762 0x55eeb31e0400 ERROR          v4l2allocator gstv4l2allocator.c:1415:gst_v4l2_allocator_dqbuf:<v4l2src0:pool0:src:allocator> The buffer type is not supported, or the index is out of bounds, or no buffers have been allocated yet, or the userptr or length are invalid.
0:00:04.948466153 28762 0x55eeb31e0400 WARN                 v4l2src gstv4l2src.c:1146:gst_v4l2src_create:<v4l2src0> error: Failed to allocate a buffer
```
  
#### Expected Results:

The second pipeline should continue to run.

#### Additional information

As with mplayer the problem does not happen
```
$ mplayer tv:// -tv device=/dev/video42
```
I started to investigate the way GStreamer was cleaning up the v4l2 device usage and discovered that removing the VIDIOC_REQBUFS with count = 0 in https://gitlab.freedesktop.org/gstreamer/gstreamer/-/blob/main/subprojects/gst-plugins-good/sys/v4l2/gstv4l2allocator.c#L790 was fixing the issue but according to the [documentation](https://www.kernel.org/doc/html/v4.10/media/uapi/v4l/vidioc-reqbufs.html), this call is valid, that's why I started to investigate in v4l2loopback.

vicamo, 2022-01-03:
Please have a check of [kernel V4L2 API document](https://www.kernel.org/doc/html/v5.0/media/uapi/v4l/vidioc-reqbufs.html), which has stated clearly **VIDIOC_REQBUFS** with zero count has its special meaning:
```
A count value of zero frees or orphans all buffers, after aborting or finishing any DMA in progress, an implicit VIDIOC_STREAMOFF.
```
The current implementation is correct.

This is a non-issue for me.

vicamo, 2022-01-03:
> #### Steps to reproduce:
> 3. Run Gstreamer playback pipelines to access the video device:
>    
>    1. `gst-launch-1.0 v4l2src device=/dev/video42 ! videoconvert !  autovideosink`
>    2. `gst-launch-1.0 v4l2src device=/dev/video42 ! videoconvert !  autovideosink`
> 
> #### Observed Results:
> * When closing the second playback pipeline only, the first pipeline gets closed as well with this error.

No. You should not open the v4l2 device for read twice. In a formal kernel V4L2 driver, each `video_device` would own a `vb2_queue` structure; for each fd opened, when you're trying to activate streaming, the `q->streaming` must be **false**, or a `-EBUSY` is returned. It follows, you can always perform one streaming action (via file I/O or buffer exchanging) on one `video_device` no matter how much fd instances you created. So, the operation here is invalid from a formal kernel driver's point of view, and the expectation is therefore also wrong.

vicamo, 2022-01-03:
You'll need https://gstreamer.freedesktop.org/documentation/coreelements/tee.html if you really need a duplicate stream.

dabrain34, 2022-01-04:
Thanks @vicamo for your feedback and your confirmation about the work-by-accident behavior. 

I was also expecting that one v4l2 device should be open exclusively. I got confused because of the comment: `/* do nothing here, buffers are always allocated */`

So I guess the v4l2loopback device should not accept multiple access then.

Regarding the tee solution, here is the solution I found:

`$  sudo insmod v4l2loopback.ko devices=2 video_nr=10,9 exclusive_caps=1`

Server:

`$ gst-launch-1.0 -v -e videotestsrc pattern=ball ! video/x-raw,width=320,height=200 ! timeoverlay ! videoconvert ! tee name=t ! queue ! v4l2sink device=/dev/video9 t. ! queue ! v4l2sink device=/dev/video10`

Clients:

`$ gst-launch-1.0 v4l2src device=/dev/video9 ! videoconvert !  autovideosink`
`$ gst-launch-1.0 v4l2src device=/dev/video10 ! videoconvert !  autovideosink`

closed 2022-01-04 by dabrain34