v4l2loopback/v4l2loopback #456: Frames getting distorted after passing through v4l2loopback

nishanth-hon, 2022-01-04:
### Step 2: Describe your environment

  * `v4l2loopback` version: 0.12.3

  * kernel version: `Linux buildroot 4.19.81-g7fe3e10f1fdf-dirty #2 SMP PREEMPT Mon Dec 20 12:19:27 IST 2021 aarch64 GNU/Linux`

  * Distribution (+version): (not a standard distribution, custom buildroot os)
   
### Step 3: Describe the problem:

I am planning to use camera from two different 
when trying to read frames / write video through opencv-python, after passing through v4l2loopback, frames are getting distorted

#### Steps to reproduce:

  1. modprobe v4l2loopback
  2. gst-launch1.0 v4l2src device=/dev/video0 ! v4l2sink device=/dev/video1
  3. write video using opencv-python 3.4.9 ( pasted code below)
  
#### Observed Results:

sample 1:
[image]

sample 2:
[image]

 sample 3: (not all frames look distorted)
[image]

#### Expected Results:

  Normal frames without distortion
  
#### Relevant Code:

code I am using to write video

  ```
import cv2
import time

cap = cv2.VideoCapture(1)
fourcc = cv2.VideoWriter_fourcc(*'mp4v')
out = cv2.VideoWriter('output.mp4', fourcc, 24, (1920, 1080))
print(cap.isOpened())

start = time.time()
while (time.time() - start) < 10:
    ret, frame = cap.read()
    if not ret:
        break
    
    out.write(frame)

out.release()
cap.release()

  ```

I understand if this is not reproducible, because I am not using a standard x86_64 distro.
If that is the case, Any suggestions on how to work around this problem would be very useful.

Thanks in advance.

umlaeute, 2022-04-06:
> I am planning to use camera from two different processes.

does that mean you try to attach to consumers (aka: capture processes) to a single `/dev/video` (loopback) device?
this is not supported (see https://github.com/umlaeute/v4l2loopback/pull/447#issuecomment-1002160400)

stephematician, 2024-12-13:
This could be related to #607, see PR #608 for a potential fix.

stephematician, 2025-02-06:
@nishanth-hon - if you're still around, this is likely closed as of 24cc1e8.