v4l2loopback/v4l2loopback #457: A device should not accept multiple open/read attempt

dabrain34, 2022-01-04:
### Describe your environment

  * `v4l2loopback` version: 0.12.5(git version)

  * kernel version: 5.11.0-43-generic

  * Distribution (+version): Ubuntu 20.04

  
### Describe the problem:

The device should not accept a multiple access for read

#### Steps to reproduce:

  1. `sudo insmod v4l2loopback.ko video_nr="42" 'card_label=mccam' exclusive_caps=1 max_buffers=2`
  2. `gst-launch-1.0 -v -e videotestsrc ! video/x-raw,width=320,height=200 ! videoconvert ! v4l2sink device=/dev/video42`
  3. Run Gstreamer playback pipelines to access the video device:
     1.  `gst-launch-1.0 v4l2src device=/dev/video42 ! videoconvert !  autovideosink`
     2.  `gst-launch-1.0 v4l2src device=/dev/video42 ! videoconvert !  autovideosink`

#### Expected Results:

On the second attempt to read/open the device, the driver should return -EBUSY.

dabrain34, 2022-01-04:
see #454

dabrain34, 2022-01-04:
I'm curious why max_openers has been set to 10 by default.

dabrain34, 2022-01-04:
@vicamo it should be set to 2 to avoid confusion.

vicamo, 2022-01-04:
> @vicamo it should be set to 2 to avoid confusion.

Not really. My original words are multiple streaming, not multiple opens. A third V4L2 client may *retrieve* the states and *listen* to their changes. It would be correct to me if you update the title to 'read/write' instead of 'open/read'.

bmghirbi, 2022-01-05:
imho, this should be an optional behavior. 
The kernel documentation states: `EBUSY: The driver does not support multiple read streams and the device is already in use.` 
I can imagine a use case where multiple reader are using the same driver at the same time.

vicamo, 2022-01-05:
https://www.kernel.org/doc/html/v5.0/media/uapi/v4l/open.html#multiple-opens
> Once an application has allocated the memory buffers needed for streaming data (by calling the ioctl VIDIOC_REQBUFS or ioctl VIDIOC_CREATE_BUFS ioctls, or implicitly by calling the read() or write() functions) that application (filehandle) becomes the owner of the device. It is no longer allowed to make changes that would affect the buffer sizes (e.g. by calling the VIDIOC_S_FMT ioctl) and other applications are no longer allowed to allocate buffers or start or stop streaming. The EBUSY error code will be returned instead.

So, no, multiple read/write is not optional and is not allowed.

closed 2023-05-02 by umlaeute