v4l2loopback/v4l2loopback #459: VIDIOC_DEQUE : Invalid argument

GU0816, 2022-01-18:
Hello everyone,
Two processes of ffmpeg are reading at the same time for one loopback device.
When ffmpeg first accesses, `VIDIOC_REQBUFS` is used to get the number of buffers.
At that time, `init_buffers` is called, but the flag of v4l2_buffer is cleared to 0 in it,
and if ffmpeg in operation requests `VIDIOC_DEQBUF` immediately after the flag is cleared,
the following error occurs.

`vidioc_dqbuf trying to return not mapped buf[4]`

When multiple processes access, I think `VIDIOC_REQBUF` should be called only once,
but since ffmpeg is OSS, the code cannot be changed easily.

Is this a usage issue?
Or is it a bug in the loopback device?

umlaeute, 2022-01-18:
both. but mostly a usage error:

1. it's not really allowed to have multiple readers for a v4l2-device.
see https://github.com/umlaeute/v4l2loopback/issues/454#issuecomment-1004214649

2. nevertheless, afaik there are some issues with dequeing (e.g. #304)

>  but since `ffmpeg` is OSS, the code cannot be changed easily.

this does not correlate with my world.
in my experience, OSS can be changed much easier that closed source software.

closed 2022-01-18 by umlaeute

GU0816, 2022-01-19:
Hi umlaeute.
Thank you for the information.
I understand.