v4l2loopback/v4l2loopback #462: README doesn't specify how to check which video device is added

abitrolly, 2022-02-15:
> https://stackoverflow.com/questions/tagged/v4l2loopback

As a completely new user, I could ask there, but after like 30 minutes reading various docs I still don't have an answer how to check if `modprobe v4l2loopback` command actually added any devices, and what are their numbers. There are three devices.

```
➜  ~ ls -a  /dev | grep video
video0
video1
video2
```
Running `modprobe v4l2loopback` again doesn't add anything new.

It is not a problem to send a PR to add the missing info what is the right way to see that `modprobe v4l2loopback` command added devices successfully, how to check which devices were added, and how to check they actually work. The missing thing is the knowledge of the right way.

### Step 2: Describe your environment

  * `v4l2loopback` version: _____

```
[    5.603674] v4l2loopback: loading out-of-tree module taints kernel.
[    5.603723] v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel
[    5.604675] v4l2loopback driver version 0.12.5 loaded
```

  * kernel version: 5.16.8-200.fc35.x86_64
  * Distribution (+version): Fedora 35

umlaeute, 2022-02-15:
> I still don't have an answer how to check if `modprobe v4l2loopback` command actually added any devices,

of course it did.
why would it not?

>  and what are their numbers. 

that is more tricky.

in general, `v4l2loopback` follows how other video devices do that.
if you plugin your USB-camera you do not know whether that actually added any devices (though you assume it did), and what the resulting device number is.

the standard way is to just query the existing devices with `v4l2-ctl` and check their names/capabilities.

the other way is to check the sysfs nodes that were created - which btw is documented in the [README](https://github.com/umlaeute/v4l2loopback#attributes).

the README doesn't explicitly say that this is a way to list virtual devices, but i guess it could be guessed `/sys/devices/virtual/video4linux/video*` lists *virtual* devices of the *video4linux* kind: unless you want to discriminate between different v4l2loopback implementations, that should be good enough.

abitrolly, 2022-02-15:
Thanks. It appears that virtual device is `video0`, but there is no link back to `/dev/video0`, so it is hard to tell if it is the same device or not.

```
➜  ~ ls -1 /sys/devices/virtual/video4linux
video0
```

> the standard way is to just query the existing devices with `v4l2-ctl` and check their names/capabilities.

I had to install `v4l-utils` to get `v4l2-ctl` on Fedora 35.

```
➜  ~ v4l2-ctl --list-devices
OBS Virtual Camera (platform:v4l2loopback-000):
	/dev/video0

Integrated Camera: Integrated C (usb-0000:00:14.0-8):
	/dev/video1
	/dev/video2
	/dev/media0
```
Not very machine readable, but solves the issue. Surprised to see laptop integrated camera being listed three times, and not as `video0`. Perhaps Fedora 35 enables autoload `v4l2loopback`. Need to reboot to check that.

Also strange that the device is called "OBS Virtual Camera". I don't see that this project is associated with OBS.

> of course it did.
> why would it not?

I didn't know all kernel modules add devices when `modprobe`d.
What happens if it couldn't add the device.. due to some bug perhaps? :bug: Will the `modprobe` fail with an error message?

umlaeute, 2022-02-15:
>  It appears that virtual device is `video0`, but there is no link back to `/dev/video0`, 

yes.
it's `video0`. and it points to `/dev/video0` via its name.

with "link" i guess you mean a *symlink*.
why would you trust a symlink more than a filename?

> Also strange that the device is called "OBS Virtual Camera". I don't see that this project is associated with OBS.

you are right that this project is not *associated* with OBS.
However, OBS uses the `v4l2loopback` module to provide a live output.
OBS loads the module (i was under the impression it only does so if the user requests a live output),
and while doing so also sets the name to "OBS Virtual Camera" (somewhat annoyingly, as the user has no way to change that name).

probably you used OBS before plugging in your camera (this would explain why `/dev/video0` is the loopback device).
or OBS persists the module loading (and the parameters), so they survive a reboot.
I don't think that Fedora would autoloads the module by default.

> I didn't know all kernel modules add devices when `modprobe`d.

not *all* modules.
but this one does.
(this is going to change somewhat with the next major release that will see dynamic loopback devices (that can be created/destroyed on demand).

>  What happens if it couldn't add the device.. due to some bug perhaps? Will the `modprobe` fail with an error message?

that's the plan.

abitrolly, 2022-02-15:
> why would you trust a symlink more than a filename?

Symlink connects the filenames, so it becomes obvious without prior knowledge that they are related. There is no reason to assume that `video0` should mean a different device if placed into another dir, but as I am new to this stuff, I have doubts.

> probably you used OBS before plugging in your camera (this would explain why /dev/video0 is the loopback device).

This is the embedded camera in Lenovo X250 laptop, so I can not plug or unplug it that easily. I tried OBS long time ago, but hadn't touch it recently. I found that Fedora package ships with `/usr/lib/modprobe.d/98-v4l2loopback.conf` file that explicitly sets the name to be "OBS Virtual Camera".

```
options v4l2loopback exclusive_caps=1 card_label="OBS Virtual Camera"
```

https://download1.rpmfusion.org/free/fedora/releases/35/Everything/x86_64/os/Packages/v/v4l2loopback-0.12.5-4.fc35.noarch.rpm

And that probably explains why virtual camera driver is loaded before real camera. Or doesn't?

> (this is going to change somewhat with the next major release that will see dynamic loopback devices (that can be created/destroyed on demand)

:+1:

PR #463 (2022-02-23): README.md Add info how to check the device number
Closes #462.

Although this shows how to figure out the device number, it still doesn't reveal how to test that the loopback device works.
commit 2199283e (2022-02-23): README.md Add info how to check the device number

Closes #462