v4l2loopback/v4l2loopback #469: Can't create devices with v4l2loopback-ctl

azertyalex, 2022-03-31:
I could create device the 'old' way, with v0.6, but wanted the newer features such as on the fly device creation and deletion. I then installed from source and everything broke.

### Step 2: Describe your environment

  * `v4l2loopback` version: [  635.727601] v4l2loopback driver version 0.12.5 loaded

  * kernel version: Linux pop-os 5.16.11-76051611-generic #202202230823~1646248261~20.04~2b22243-Ubuntu SMP PREEMPT Thu Ma x86_64 x86_64 x86_64 GNU/Linux

  * Distribution pop:
Description:	Pop!_OS 20.04 LTS
Release:	20.04
Codename:	focal

          lsb_release -a

### Step 3: Describe the problem:

#### Steps to reproduce:

  1. install from source following the steps in readme.md
  2. try to create a device
  3. fail

#### Observed Results:

```bash
sudo v4l2loopback-ctl add -n "Webcam 1" /dev/video3
unable to open control device '/dev/v4l2loopback': No such file or directory
```

#### Expected Results:

 created a device

umlaeute, 2022-03-31:
that most likely means that you have loaded  a wrong version of the module (most likely: the one installed by the `v4l2loopback-dkms` package of your distribution).
try uninstalling the `v4l2loopback-dkms` package and reload the module.

PS: it's a bit unfortunate that the last released version is `0.12.5` and the development version uses the same specifier, so it's hard to tell which one you actually loaded.

azertyalex, 2022-03-31:
I followed your advice, changed the strings myself to 0.12.6 and reinstalled/reloaded all the modules.
demsg was still saying i was using `0.12.6`

I looked further and noticed this:
```bash
$ find /lib/modules/$(uname -r) -type f -name '*.ko' | grep v4l2loopback
/lib/modules/5.16.11-76051611-generic/kernel/v4l2loopback/v4l2loopback.ko
/lib/modules/5.16.11-76051611-generic/extra/v4l2loopback.ko
```

so I removed those and tried to load kernel module, got not found error.

I then manually copied from the `extra` folder to the `kernel` folder. lo and behold
```bash
$ sudo dmesg  | grep -i v4l2loopback
[  635.727601] v4l2loopback driver version 0.12.5 loaded
[ 5942.758320] v4l2loopback driver version 0.12.5 loaded
[ 6025.262163] v4l2loopback driver version 0.12.5 loaded
[ 6318.440831] v4l2loopback driver version 0.12.6 loaded
```
```bash
$ sudo v4l2loopback-ctl add -n "Webcam 1" /dev/video3
/dev/video3
```

umlaeute, 2022-03-31:
now that's just fine.
you'r problem is simply an outdated cache.

running `depmod -a` after "installling the from-source version of `v4l2loopback` and removing the version from .../extra" would probably have done the trick.

in general it is not very advisable to just remove random files from your kernel-tree.
better to find out which package installed the file (`dpkg -S /lib/modules/5.16.11-76051611-generic/extra/v4l2loopback.ko`) and remove the entire package (so the system can keep track of the file which it thinks it has installed). of course this only works if removing that package doesn't remove half of your system (e.g. the kernel itself :-P))

anyhow: glad it works now.

closed 2022-03-31 by umlaeute