v4l2loopback/v4l2loopback #472: Chrome doesn't see the loopback device ("Camera muted by the system")

kuba-orlik, 2022-04-11:
* `v4l2loopback` version: `0.12.5.r253` (gd01c20c)
  * kernel version: `5.17.1-arch1`
  * Distribution (+version): Arch Linux

#### Steps to reproduce:

```
# create a new loopback device: 

sudo v4l2loopback-ctl add \
     -n "loopy doopy" \
     -w 853 \
     -h 480 \
     -x 1 \
     -v \
     /dev/video1

# set proper caps
sudo v4l2loopback-ctl set-caps /dev/video1 YU12:853x480@25

# stream a scaled version of the /dev/video0 device
ffmpeg -f v4l2 -pix_fmt nv12 -i /dev/video0 -f v4l2 -vf "scale=853:480,format=yuv420p" -fflags nobuffer -r $framerate /dev/video1
```

#### Observed Results:

  * When trying to use this camera in Chrome, it fails. Google Meet says "Camera muted by system". Jitsi says "Unable to access camera"

#### Expected Results:

  * I expected the camera to be available in Chrome

cfbraun, 2022-06-09:
1. Chrome needs 'exclusive_caps=1'
2. Chrome rounds the image size differently, so odd widths lead to 'Dequeued v4l2 buffer contains invalid length' errors

umlaeute, 2024-03-19:
currently, the width of a yuv420p image must be even (that is: 854 is ok, but 853 is not)

closed 2024-03-19 by umlaeute