v4l2loopback/v4l2loopback #475: V4L2loopback for Home Assistant with Add-on

pergolafabio, 2022-05-11:
Hi, anyone succeed to run this on Home Assistant as Add-on with HassOS as OS?

I was trying to create addon, also enabled the kernel-modules and the enabled priviliged, and disabled protection mode..
Basicly enabled all settins to access host :-)

```
  "kernel_modules": true,
  "host_network": true,
  "full_access": true,
  "privileged": ["NET_ADMIN", "SYS_ADMIN", "SYS_RAWIO", "SYS_TIME", "SYS_NICE", "SYS_RESOURCE", "SYS_PTRACE", "SYS_MODULE", "DAC_READ_SEARCH"],
```

Addon Dockerfile to start:

```
ARG BUILD_FROM
FROM $BUILD_FROM
RUN apk add git gcc make
RUN git clone https://github.com/umlaeute/v4l2loopback
```

but the command "make" gives me error below:

```
bash-5.1# make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/app/v4l2loopback KCPPFLAGS="-DSNAPSHOT_VERSION='"v0.12.5-253-gd01c20c"'" modules
make[1]: *** /lib/modules/5.10.108/build: No such file or directory.  Stop.
make: *** [Makefile:51: v4l2loopback.ko] Error 2
```
allthough that  that 5.10.108 directory exist?

```
bash-5.1# cd /lib/modules/5.10.108/
bash-5.1# ls
extra                      modules.alias.bin          modules.builtin.bin        modules.dep.bin            modules.softdep
kernel                     modules.builtin            modules.builtin.modinfo    modules.devname            modules.symbols
modules.alias              modules.builtin.alias.bin  modules.dep                modules.order              modules.symbols.bin
```

thnx for help

umlaeute, 2022-05-11:
*that* directory is `/lib/modules/5.10.108/build` and it does **not** exist.

the `build` directory is contains the kernel-headers (or rather: it is a symlink to the kernel headers), which - [as per the documentation](https://github.com/umlaeute/v4l2loopback#dependencies) - are a hard dependency for building `v4l2loopback`.

so: please install the kernel headers.

if your kernel-headers are somewhere else (thought really they shouldn't) you can specify the directory with the `KERNEL_DIR` variable.

closed 2022-05-11 by umlaeute

pergolafabio, 2022-05-11:
ah, thats interesting, 
when i do: apk add linux-headers-$(uname -r)

```
ERROR: unable to select packages:
  linux-headers-5.10.108 (no such package):
    required by: world[linux-headers-5.10.108]
```

i had the same message with a debian based container, but then the linux-headers were installed
so i can maybe manage with that KERNEL_DIR variable .... is that somewhere described? or is that a variable from the "make" app ?

umlaeute, 2022-05-11:
you *must* have the kernel headers installed.

consult the documentation of your distribution on how to install them.

pergolafabio, 2022-05-11:
yes, i now grabbedl the .apk from alpine packakes (latest one from 3.15)
did now : "apk add linux-headers-5.10.41-r0.apk"

that installed me linux-headers

now what ? :-)

pergolafabio, 2022-05-11:
allthough i still dont see /lib/modules/5.10.41 folder

i still see there /lib/modules/5.10.108

pergolafabio, 2022-05-11:
ok, i i updated my host os, to latest kernel, now when i do run "apk add linux-headers" it succeeds it seems

But see below, when i do a make, allthough that dir 5.15.38 is there, but the linux-headers are 5.10.41 ? how can i proceed?
```
➜  ~ docker exec -it addon_local_v4l2 /bin/bash
bash-5.1# apk add linux-headers
(1/1) Installing linux-headers (5.10.41-r0)
OK: 151 MiB in 55 packages
bash-5.1# uname -r
5.15.38
bash-5.1# uname -a
Linux local-v4l2 5.15.38 #1 SMP Wed May 11 09:52:49 UTC 2022 x86_64 Linux
bash-5.1# cd v4l2loopback/
bash-5.1# make
Building v4l2-loopback driver...
make -C /lib/modules/`uname -r`/build M=/app/v4l2loopback KCPPFLAGS="-DSNAPSHOT_VERSION='"v0.12.5-253-gd01c20c"'" modules
make[1]: *** /lib/modules/5.15.38/build: No such file or directory.  Stop.
make: *** [Makefile:51: v4l2loopback.ko] Error 2
bash-5.1# cd /lib/modules
bash-5.1# ls
5.15.38
bash-5.1# cd 5.15.38/
bash-5.1# ls
extra                      modules.alias              modules.builtin            modules.builtin.bin        modules.dep                modules.devname            modules.softdep            modules.symbols.bin
kernel                     modules.alias.bin          modules.builtin.alias.bin  modules.builtin.modinfo    modules.dep.bin            modules.order              modules.symbols
bash-5.1#
```

pergolafabio, 2022-05-11:
Ok, found something, seems the build folder is indeed located somewhere else on Alpine...

https://www.linuxquestions.org/questions/linux-newbie-8/lib-modules-4-15-0-29-generic-build-no-such-file-or-directory-4175640971/

Gonna try it again tomorrow and change that kernel dir variable to that new folder...

pergolafabio, 2022-05-19:
hey @umlaeute , i was trying this setup again, i was able to create the "v4l2loopback.ko" file by the "make" command,

I changed indeed this one below to another path:

KERNEL_DIR	?= /srv/linux-4.9.4

but the command "sudo make install" is failing me, because of read only  system, is there another way or can i also point to another DIR?

```
/app/v4l2loopback # make
Building v4l2-loopback driver...
make -C /srv/linux-4.9.4 M=/app/v4l2loopback KCPPFLAGS="-DSNAPSHOT_VERSION='"v0.12.5-253-gd01c20c-dirty"'" modules
make[1]: Entering directory '/srv/linux-4.9.4'
  CC [M]  /app/v4l2loopback/v4l2loopback.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /app/v4l2loopback/v4l2loopback.mod.o
  LD [M]  /app/v4l2loopback/v4l2loopback.ko
make[1]: Leaving directory '/srv/linux-4.9.4'
make -C utils
make[1]: Entering directory '/app/v4l2loopback/utils'
cc  -I..   v4l2loopback-ctl.c   -o v4l2loopback-ctl
make[1]: Leaving directory '/app/v4l2loopback/utils'
/app/v4l2loopback # sudo make install
make -C /srv/linux-4.9.4 M=/app/v4l2loopback modules_install
make[1]: Entering directory '/srv/linux-4.9.4'
mkdir: can't create directory '/lib/modules/4.9.4/': Read-only file system
Makefile:1502: recipe for target '_emodinst_' failed
make[1]: *** [_emodinst_] Error 1
make[1]: Leaving directory '/srv/linux-4.9.4'
Makefile:55: recipe for target 'install' failed
make: *** [install] Error 2
/app/v4l2loopback #
```

umlaeute, 2022-05-19:
1. you don't need to edit the Makefile in order to change the `KERNEL_DIR`. instead use:
   ~~~sh
   make KERNEL_DIR=/srv/linux-4.9.4
   ~~~
2. `make install` tries to install the kernel modules in their proper place. According to *your* kernel headers this is `/lib/modules/4.9.4/`. However, on your system this directory is **read-only** (see the error logs). If you cannot write to the directory even as `root`, then most likely the entire filesystem that hosts the directory is read-only.
   Either re-mount the directory as `r/w` and run again, or manually copy the files (with whatever magic involved) to the proper places.

pergolafabio, 2022-05-19:
ok, thnx, i'm on the host now, but still read/only when i manually want to copy file, gonna check with home assistant how to load it indeed as r/w
hopefully i get some magic
thnx

pergolafabio, 2022-05-25:
I was able to rebuild the OS , based on buildroot, enabled v4l2loopback, after the compile was done, copied over the .ko file to my live system, .... With insmod I was able to load the object!!!

Great, now I can use it with Home Assistant also ;-)

Thnx for all the help, really appreciated