v4l2loopback/v4l2loopback #476: Module fails to load on 11th+ gen Intel processors since kernel 5.18

justmoon, 2022-06-02:
Apparently Intel Indirect Branch Tracking feature was enabled in kernel 5.18 for Intel 11th and 12th gen processors and newer. This makes `v4l2loopback` not load correctly with error:

```
kernel: traps: Missing ENDBR: init_module+0x0/0x1000 [v4l2loopback]
```

<details>
  <summary>Full kernel log (dmesg)</summary>
<code><pre>
Jun 02 07:03:06.069090 perdix kernel: v4l2loopback: loading out-of-tree module taints kernel.
Jun 02 07:03:06.069161 perdix kernel: v4l2loopback: module verification failed: signature and/or required key missing - tainting kernel
Jun 02 07:03:06.069175 perdix kernel: traps: Missing ENDBR: init_module+0x0/0x1000 [v4l2loopback]
Jun 02 07:03:06.069188 perdix kernel: ------------[ cut here ]------------
Jun 02 07:03:06.069415 perdix kernel: kernel BUG at arch/x86/kernel/traps.c:252!
Jun 02 07:03:06.069494 perdix kernel: invalid opcode: 0000 [#1] PREEMPT SMP NOPTI
Jun 02 07:03:06.083762 perdix kernel: CPU: 0 PID: 1604 Comm: modprobe Tainted: G           OE     5.18.1-arch1-1 #1 aeb6a372044721fe869dfc17901d8ed9fc452f1a
Jun 02 07:03:06.083811 perdix kernel: Hardware name: Framework Laptop/FRANBMCP08, BIOS 03.02 07/01/2021
Jun 02 07:03:06.083830 perdix kernel: RIP: 0010:exc_control_protection+0xc2/0xd0
Jun 02 07:03:06.083843 perdix kernel: Code: 8b 93 80 00 00 00 be f9 00 00 00 48 c7 c7 83 ab e6 ab e8 d1 f1 4f ff e9 72 ff ff ff 48 c7 c7 6a ab e6 ab e8 c7 21 fb ff 0f 0b <0f> 0b 66 66 2e 0f 1f 84 0>
Jun 02 07:03:06.083977 perdix kernel: RSP: 0018:ffffb9f4c2443cf8 EFLAGS: 00010002
Jun 02 07:03:06.083992 perdix kernel: RAX: 000000000000003b RBX: ffffb9f4c2443d18 RCX: 0000000000000027
Jun 02 07:03:06.084005 perdix kernel: RDX: 0000000000000000 RSI: 0000000000000001 RDI: ffff9fe06fa216a0
Jun 02 07:03:06.084015 perdix kernel: RBP: 0000000000000003 R08: 0000000000000000 R09: ffffb9f4c2443b18
Jun 02 07:03:06.084025 perdix kernel: R10: 0000000000000003 R11: ffffffffac6caa08 R12: 0000000000000000
Jun 02 07:03:06.084040 perdix kernel: R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
Jun 02 07:03:06.084051 perdix kernel: FS:  00007f0a00827740(0000) GS:ffff9fe06fa00000(0000) knlGS:0000000000000000
Jun 02 07:03:06.084063 perdix kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jun 02 07:03:06.084077 perdix kernel: CR2: 00007f0a00239000 CR3: 0000000600dc8001 CR4: 0000000000f70ef0
Jun 02 07:03:06.084087 perdix kernel: PKRU: 55555554
Jun 02 07:03:06.084098 perdix kernel: Call Trace:
Jun 02 07:03:06.084107 perdix kernel:  <TASK>
Jun 02 07:03:06.084117 perdix kernel:  asm_exc_control_protection+0x22/0x30
Jun 02 07:03:06.084128 perdix kernel: RIP: 0010:init_module+0x0/0x1000 [v4l2loopback]
Jun 02 07:03:06.084138 perdix kernel: Code: Unable to access opcode bytes at RIP 0xffffffffc09f3fd6.
Jun 02 07:03:06.084148 perdix kernel: RSP: 0018:ffffb9f4c2443dc0 EFLAGS: 00010246
Jun 02 07:03:06.084158 perdix kernel: RAX: 0000000000000000 RBX: ffffffffc09f4000 RCX: 0000000000000000
Jun 02 07:03:06.084170 perdix kernel: RDX: 0000000000000000 RSI: ffffffffc09f4000 RDI: ffffb9f4c2443da8
Jun 02 07:03:06.084180 perdix kernel: RBP: ffffb9f4c2443dc8 R08: 0000000000000010 R09: ffff9fd5f07152c0
Jun 02 07:03:06.084190 perdix kernel: R10: ffff9fd5fecd6b80 R11: 0000000000000100 R12: 0000000000000000
Jun 02 07:03:06.084201 perdix kernel: R13: 00007f09fff0bb50 R14: 00005625081d8c20 R15: ffffb9f4c2443e60
Jun 02 07:03:06.084210 perdix kernel:  ? 0xffffffffc09f4000
Jun 02 07:03:06.084219 perdix kernel:  ? 0xffffffffc09f4000
Jun 02 07:03:06.084227 perdix kernel:  do_one_initcall+0x5a/0x220
Jun 02 07:03:06.084238 perdix kernel:  do_init_module+0x4a/0x240
Jun 02 07:03:06.084252 perdix kernel:  __do_sys_init_module+0x138/0x1b0
Jun 02 07:03:06.084261 perdix kernel:  do_syscall_64+0x5c/0x90
Jun 02 07:03:06.084292 perdix kernel:  ? exc_page_fault+0x74/0x170
Jun 02 07:03:06.084310 perdix kernel:  entry_SYSCALL_64_after_hwframe+0x44/0xae
Jun 02 07:03:06.084322 perdix kernel: RIP: 0033:0x7f0a00112c3e
Jun 02 07:03:06.084337 perdix kernel: Code: 48 8b 0d 5d b1 0e 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c>
Jun 02 07:03:06.084347 perdix kernel: RSP: 002b:00007ffec61e0c98 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
Jun 02 07:03:06.084357 perdix kernel: RAX: ffffffffffffffda RBX: 00005625081d8ce0 RCX: 00007f0a00112c3e
Jun 02 07:03:06.084369 perdix kernel: RDX: 00005625081d8c20 RSI: 0000000000028b40 RDI: 00007f09ffee3010
Jun 02 07:03:06.084378 perdix kernel: RBP: 00007f09ffee3010 R08: 0000000000049000 R09: 0000000000000000
Jun 02 07:03:06.084387 perdix kernel: R10: 0000000000004681 R11: 0000000000000246 R12: 00005625081d8c20
Jun 02 07:03:06.084398 perdix kernel: R13: 00005625081d8e10 R14: 00005625081d8ce0 R15: 00005625081d9410
Jun 02 07:03:06.084407 perdix kernel:  </TASK>
Jun 02 07:03:06.084438 perdix kernel: Modules linked in: v4l2loopback(OE+) ccm rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ip>
Jun 02 07:03:06.084497 perdix kernel:  x86_pkg_temp_thermal ledtrig_audio ee1004 iTCO_vendor_support hid_multitouch joydev snd_hda_intel intel_powerclamp coretemp mac80211 snd_intel_dspcfg snd_inte>
Jun 02 07:03:06.084521 perdix kernel:  video acpi_pad int3400_thermal acpi_thermal_rel vfat fat dm_multipath ipmi_devintf ipmi_msghandler sg crypto_user fuse ip_tables x_tables ext4 crc32c_generic >
Jun 02 07:03:06.084533 perdix kernel: ---[ end trace 0000000000000000 ]---
Jun 02 07:03:06.084546 perdix kernel: RIP: 0010:exc_control_protection+0xc2/0xd0
Jun 02 07:03:06.084553 perdix kernel: Code: 8b 93 80 00 00 00 be f9 00 00 00 48 c7 c7 83 ab e6 ab e8 d1 f1 4f ff e9 72 ff ff ff 48 c7 c7 6a ab e6 ab e8 c7 21 fb ff 0f 0b <0f> 0b 66 66 2e 0f 1f 84 0>
Jun 02 07:03:06.084563 perdix kernel: RSP: 0018:ffffb9f4c2443cf8 EFLAGS: 00010002
Jun 02 07:03:06.084572 perdix kernel: RAX: 000000000000003b RBX: ffffb9f4c2443d18 RCX: 0000000000000027
Jun 02 07:03:06.084582 perdix kernel: RDX: 0000000000000000 RSI: 0000000000000001 RDI: ffff9fe06fa216a0
Jun 02 07:03:06.084590 perdix kernel: RBP: 0000000000000003 R08: 0000000000000000 R09: ffffb9f4c2443b18
Jun 02 07:03:06.084597 perdix kernel: R10: 0000000000000003 R11: ffffffffac6caa08 R12: 0000000000000000
Jun 02 07:03:06.084604 perdix kernel: R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
Jun 02 07:03:06.084611 perdix kernel: FS:  00007f0a00827740(0000) GS:ffff9fe06fa00000(0000) knlGS:0000000000000000
Jun 02 07:03:06.084619 perdix kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jun 02 07:03:06.084628 perdix kernel: CR2: ffffffffc09f3fd6 CR3: 0000000600dc8001 CR4: 0000000000f70ef0
Jun 02 07:03:06.084637 perdix kernel: PKRU: 55555554
</pre></code>
</details>

The workaround is to disable IBT with the `ibt=off` kernel parameter.

See: https://bbs.archlinux.org/viewtopic.php?id=276822

  * `v4l2loopback` version: 0.12.5
  * kernel version: 5.18.1-arch1-1
  * Distribution (+version): Arch Linux
 
#### Steps to reproduce:

  1. Have an Intel 11th or 12th gen processor
  2. Install kernel 5.18 with IBT enabled (default)
  3. Run `modprobe v4l2loopback`

#### Observed Results:

* `v4l2loopback` module is added to kernel, but OBS is not able to create a virtual webcam
* Kernel log shows `kernel: traps: Missing ENDBR: init_module+0x0/0x1000 [v4l2loopback]`

#### Expected Results:

  * `v4l2loopback` should load successfully and operate normally
  * No kernel bug is logged

pswiatki, 2022-06-05:
Just in case it could help:

```
Jun 05 09:41:15 sdr-aptv kernel: traps: Missing ENDBR: init_module+0x0/0x1000 [v4l2loopback]
Jun 05 09:41:15 sdr-aptv kernel: ------------[ cut here ]------------
Jun 05 09:41:15 sdr-aptv kernel: kernel BUG at arch/x86/kernel/traps.c:252!
Jun 05 09:41:15 sdr-aptv kernel: invalid opcode: 0000 [#1] PREEMPT SMP NOPTI
Jun 05 09:41:15 sdr-aptv kernel: CPU: 3 PID: 2310 Comm: modprobe Tainted: G           OE     5.18.1-arch1-1 #1 aeb6a372044721fe869dfc17901d8ed9fc452f1a
Jun 05 09:41:15 sdr-aptv kernel: Hardware name: Dell Inc. Precision 3561/03G764, BIOS 1.8.0 12/10/2021
Jun 05 09:41:15 sdr-aptv kernel: RIP: 0010:exc_control_protection+0xc2/0xd0
Jun 05 09:41:15 sdr-aptv kernel: Code: 8b 93 80 00 00 00 be f9 00 00 00 48 c7 c7 83 ab c6 8e e8 d1 f1 4f ff e9 72 ff ff ff 48 c7 c7 6a ab c6 8e e8 c7 21 fb ff 0f 0b <0f> 0b 66 66 2e 0f 1f 84 00 00 00 00 00 90 66 0f 1f 00 55 53 48 89
Jun 05 09:41:15 sdr-aptv kernel: RSP: 0018:ffffabf1c3b4fc58 EFLAGS: 00010002
Jun 05 09:41:15 sdr-aptv kernel: RAX: 000000000000003b RBX: ffffabf1c3b4fc78 RCX: 0000000000000000
Jun 05 09:41:15 sdr-aptv kernel: RDX: 0000000000000000 RSI: ffff91b54f4e16a0 RDI: ffff91b54f4e16a0
Jun 05 09:41:15 sdr-aptv kernel: RBP: 0000000000000003 R08: 0000000000000000 R09: ffffabf1c3b4fa78
Jun 05 09:41:15 sdr-aptv kernel: R10: 0000000000000003 R11: ffffffff8f4caa08 R12: 0000000000000000
Jun 05 09:41:15 sdr-aptv kernel: R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
Jun 05 09:41:15 sdr-aptv kernel: FS:  00007f862b8e4740(0000) GS:ffff91b54f4c0000(0000) knlGS:0000000000000000
Jun 05 09:41:15 sdr-aptv kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jun 05 09:41:15 sdr-aptv kernel: CR2: 00007f862b039000 CR3: 0000000116aae001 CR4: 0000000000f70ee0
Jun 05 09:41:15 sdr-aptv kernel: PKRU: 55555554
Jun 05 09:41:15 sdr-aptv kernel: Call Trace:
Jun 05 09:41:15 sdr-aptv kernel:  <TASK>
Jun 05 09:41:15 sdr-aptv kernel:  asm_exc_control_protection+0x22/0x30
Jun 05 09:41:15 sdr-aptv kernel: RIP: 0010:init_module+0x0/0x1000 [v4l2loopback]
Jun 05 09:41:15 sdr-aptv kernel: Code: Unable to access opcode bytes at RIP 0xffffffffc09d6fd6.
Jun 05 09:41:15 sdr-aptv kernel: RSP: 0018:ffffabf1c3b4fd28 EFLAGS: 00010246
Jun 05 09:41:15 sdr-aptv kernel: RAX: 0000000000000000 RBX: ffffffffc09d7000 RCX: 0000000000000000
Jun 05 09:41:15 sdr-aptv kernel: RDX: 0000000000000000 RSI: ffffffffc09d7000 RDI: ffffabf1c3b4fd10
Jun 05 09:41:15 sdr-aptv kernel: RBP: ffffabf1c3b4fd30 R08: 0000000000000010 R09: ffff91a664e7ab10
Jun 05 09:41:15 sdr-aptv kernel: R10: ffff91a632765940 R11: 0000000000000100 R12: 0000000000000000
Jun 05 09:41:15 sdr-aptv kernel: R13: 00007f862afd8b50 R14: 00005570e42d6d70 R15: ffffabf1c3b4fdc8
Jun 05 09:41:15 sdr-aptv kernel:  ? 0xffffffffc09d7000
Jun 05 09:41:15 sdr-aptv kernel:  ? 0xffffffffc09d7000
Jun 05 09:41:15 sdr-aptv kernel:  do_one_initcall+0x5a/0x220
Jun 05 09:41:15 sdr-aptv kernel:  do_init_module+0x4a/0x240
Jun 05 09:41:15 sdr-aptv kernel:  __do_sys_init_module+0x138/0x1b0
Jun 05 09:41:15 sdr-aptv kernel:  do_syscall_64+0x5c/0x90
Jun 05 09:41:15 sdr-aptv kernel:  ? syscall_exit_to_user_mode+0x26/0x50
Jun 05 09:41:15 sdr-aptv kernel:  ? do_syscall_64+0x6b/0x90
Jun 05 09:41:15 sdr-aptv kernel:  ? syscall_exit_to_user_mode+0x26/0x50
Jun 05 09:41:15 sdr-aptv kernel:  ? do_syscall_64+0x6b/0x90
Jun 05 09:41:15 sdr-aptv kernel:  ? syscall_exit_to_user_mode+0x26/0x50
Jun 05 09:41:15 sdr-aptv kernel:  ? do_syscall_64+0x6b/0x90
Jun 05 09:41:15 sdr-aptv kernel:  entry_SYSCALL_64_after_hwframe+0x44/0xae
Jun 05 09:41:15 sdr-aptv kernel: RIP: 0033:0x7f862b312c3e
Jun 05 09:41:15 sdr-aptv kernel: Code: 48 8b 0d 5d b1 0e 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 2a b1 0e 00 f7 d8 64 89 01 48
Jun 05 09:41:15 sdr-aptv kernel: RSP: 002b:00007ffce1ba3df8 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
Jun 05 09:41:15 sdr-aptv kernel: RAX: ffffffffffffffda RBX: 00005570e42d6e30 RCX: 00007f862b312c3e
Jun 05 09:41:15 sdr-aptv kernel: RDX: 00005570e42d6d70 RSI: 0000000000028b40 RDI: 00007f862afb0010
Jun 05 09:41:15 sdr-aptv kernel: RBP: 00007f862afb0010 R08: 0000000000049000 R09: 0000000000000000
Jun 05 09:41:15 sdr-aptv kernel: R10: 0000000000004211 R11: 0000000000000246 R12: 00005570e42d6d70
Jun 05 09:41:15 sdr-aptv kernel: R13: 00005570e42d6f40 R14: 00005570e42d6e30 R15: 00005570e42d7540
Jun 05 09:41:15 sdr-aptv kernel:  </TASK>
Jun 05 09:41:15 sdr-aptv kernel: Modules linked in: v4l2loopback(OE+) xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc overlay rfcomm ccm cmac algif_hash algif_skcipher af_alg bnep bbswitch(OE) joydev snd_hda_codec_hdmi snd_sof_pci_intel_tgl dell_rbtn snd_sof_intel_hda_common hid_sensor_custom soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda hid_sensor_hub snd_ctl_led snd_sof_pci snd_sof_xtensa_dsp snd_sof intel_ishtp_hid snd_sof_utils snd_soc_hdac_hda dell_laptop snd_hda_ext_core hid_multitouch spi_nor uvcvideo iTCO_wdt snd_soc_acpi_intel_match dell_wmi intel_pmc_bxt mei_hdcp mei_pxp mei_wdt mtd ee1004 iTCO_vendor_support videobuf2_vmalloc snd_soc_acpi snd_hda_codec_realtek dell_smbios btusb pmt_telemetry videobuf2_memops pmt_class intel_rapl_msr soundwire_bus videobuf2_v4l2 dell_wmi_sysman intel_tcc_cooling btrtl dcdbas
[... cut ...]
Jun 04 21:49:31 sdr-aptv kernel:  reg_todo+0x1da/0x230 [cfg80211 6c8a3866f60be53418f2b19fc604cae5d6ef316a]
Jun 04 21:49:31 sdr-aptv kernel:  process_one_work+0x1c4/0x380
Jun 04 21:49:31 sdr-aptv kernel:  worker_thread+0x51/0x380
Jun 04 21:49:31 sdr-aptv kernel:  ? rescuer_thread+0x3a0/0x3a0
Jun 04 21:49:31 sdr-aptv kernel:  kthread+0xdb/0x110
Jun 04 21:49:31 sdr-aptv kernel:  ? kthread_complete_and_exit+0x20/0x20
Jun 04 21:49:31 sdr-aptv kernel:  ret_from_fork+0x1f/0x30
Jun 04 21:49:31 sdr-aptv kernel:  </TASK>
```

ProfeJulianLasso, 2022-06-21:
i have the same problem :(

MauroSoli, 2022-08-05:
Same for me.

GeorgFoc, 2022-08-06:
And I am also affected.

umlaeute, 2022-08-16:
i'm sure it is unrelated to "11th+ gen Intel processors since kernel 5.18".

```sh
$ lscpu 
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         39 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  8
  On-line CPU(s) list:   0-7
Vendor ID:               GenuineIntel
  Model name:            11th Gen Intel(R) Core(TM) i5-1145G7 @ 2.60GHz
$ uname -a
Linux umlautT 5.18.0-3-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.14-1 (2022-07-23) x86_64 GNU/Linux
$ sudo modprobe v4l2loopback
$ sudo dmesg | tail -1
[430342.292879] v4l2loopback driver version 0.12.7 loaded
```

umlaeute, 2022-08-16:
the problem seems to be the enabling of `IBT` on Arch-kernels.

As a temporary workaround, you can disable it by adding `ibt=off` to your kernel params.
see https://github.com/NVIDIA/open-gpu-kernel-modules/issues/256#issuecomment-1138851460

xnox, 2022-08-16:
v0.12.7 contains https://github.com/umlaeute/v4l2loopback/commit/ddce39b5eb667d3a5db0009f96f6050496bd58ad which should resolve this. For me I am able to build this module on IBT-enabled kernel and load it.

closed 2022-08-17 by umlaeute