v4l2loopback/v4l2loopback #484: Can't write video to /dev/device since commit 3312a6d

azertyalex, 2022-06-22:
### environment

  * v4l2loopback driver version 0.12.5 (v0.12.5-254-g3312a6d) loaded

  * `Linux pop-os 5.17.5-76051705-generic #202204271406~1651504840~20.04~63e51bd-Ubuntu SMP PREEMPT Wed Ma x86_64 x86_64 x86_64 GNU/Linux`

```bash
No LSB modules are available.
Distributor ID:	Pop
Description:	Pop!_OS 20.04 LTS
Release:	20.04
Codename:	focal
```

### Problem:

Can't write video with ffmpeg

#### Steps to reproduce:

  1. checkout commit 3312a6d0461d6d2d82a411f26c34e380bd3cee27
  2. build and install kernerl module
  3. write video to device

#### Observed Results:

`Error: av_interleaved_write_frame(): Invalid argument`

#### Expected Results:

There should be no errors

#### Relevant Code:

  ```bash
ffmpeg -re -i upload.mp4 -f v4l2 /dev/video3
  ```

BenBE, 2022-06-22:
The mentioned patch was a result of another bug mentioned in #470 and addressed as part of #477.

Please have a look at the commit prior to this change and see if you can reproduce the behavior reported in that issue (with exclusive_caps enabled `ffmpeg` will only run once for this output device and caps will be lost after `ffmpeg` terminates).

I'll have to take another look at this situation, as the implementation before this patch was much more broken (causing hard-to-debug side-effects), while the current situation at least clearly indicates where things went wrong (this is intentional and documented in the commit message).

I'll check if there's a reasonable way to restore the previous behavior (though triggering a warning in the kernel log) by trying to upgrade the opener to a WRITER if possible (and only fail if this upgrade fails). That's not ideal, but should lessen the pain somewhat and mitigate the original issue from re-appearing.

azertyalex, 2022-06-22:
1. ffmpeg version 4.2.4-1ubuntu0.1
2. fixed
3.
```
ioctl(2, TCGETS, 0x7ffe6b18be50)        = -1 ENOTTY (Inappropriate ioctl for device)
ioctl(2, TCGETS, 0x7ffe6b18be10)        = -1 ENOTTY (Inappropriate ioctl for device)
write(2, "ffmpeg version 4.2.4-1ubuntu0.1", 31ffmpeg version 4.2.4-1ubuntu0.1) = 31
write(2, " Copyright (c) 2000-2020 the FFm"..., 46 Copyright (c) 2000-2020 the FFmpeg developers) = 46
write(2, "\n", 1
write(2, "  built with gcc 9 (Ubuntu 9.3.0"..., 44  built with gcc 9 (Ubuntu 9.3.0-10ubuntu2)
write(2, "  configuration: --prefix=/usr -"..., 1234  configuration: --prefix=/usr --extra-version=1ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-nvenc --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
write(2, "  WARNING: library configuration"..., 42  WARNING: library configuration mismatch
write(2, "  avcodec     configuration: --p"..., 1437  avcodec     configuration: --prefix=/usr --extra-version=1ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-nvenc --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --enable-version3 --disable-doc --disable-programs --enable-libaribb24 --enable-liblensfun --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libtesseract --enable-libvo_amrwbenc
write(2, "  libavutil      56. 31.100 / 56"..., 41  libavutil      56. 31.100 / 56. 31.100
write(2, "  libavcodec     58. 54.100 / 58"..., 41  libavcodec     58. 54.100 / 58. 54.100
write(2, "  libavformat    58. 29.100 / 58"..., 41  libavformat    58. 29.100 / 58. 29.100
write(2, "  libavdevice    58.  8.100 / 58"..., 41  libavdevice    58.  8.100 / 58.  8.100
write(2, "  libavfilter     7. 57.100 /  7"..., 41  libavfilter     7. 57.100 /  7. 57.100
write(2, "  libavresample   4.  0.  0 /  4"..., 41  libavresample   4.  0.  0 /  4.  0.  0
write(2, "  libswscale      5.  5.100 /  5"..., 41  libswscale      5.  5.100 /  5.  5.100
write(2, "  libswresample   3.  5.100 /  3"..., 41  libswresample   3.  5.100 /  3.  5.100
write(2, "  libpostproc    55.  5.100 / 55"..., 41  libpostproc    55.  5.100 / 55.  5.100
ioctl(0, TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(0, TCGETS, {B38400 opost isig icanon echo ...}) = 0
ioctl(0, SNDCTL_TMR_START or TCSETS, {B38400 opost isig -icanon -echo ...}) = 0
ioctl(0, TCGETS, {B38400 opost isig -icanon -echo ...}) = 0
write(2, "Input #0, mov,mp4,m4a,3gp,3g2,mj"..., 54Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'upload.mp4':
write(2, "  Metadata:\n", 12  Metadata:
write(2, "    major_brand     : ", 22    major_brand     : )  = 22
write(2, "isom", 4isom)                     = 4
write(2, "\n", 1
write(2, "    minor_version   : ", 22    minor_version   : )  = 22
write(2, "512", 3512)                      = 3
write(2, "\n", 1
write(2, "    compatible_brands: ", 23    compatible_brands: ) = 23
write(2, "isomiso2avc1mp41", 16isomiso2avc1mp41)        = 16
write(2, "\n", 1
write(2, "    encoder         : ", 22    encoder         : )  = 22
write(2, "Lavf58.29.100", 13Lavf58.29.100)           = 13
write(2, "\n", 1
write(2, "  Duration: ", 12  Duration: )            = 12
write(2, "00:00:28.91", 1100:00:28.91)             = 11
write(2, ", start: ", 9, start: )                = 9
write(2, "0.000000", 80.000000)                 = 8
write(2, ", bitrate: ", 11, bitrate: )             = 11
write(2, "1600 kb/s", 91600 kb/s)                = 9
write(2, "\n", 1
[... cut ...]
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=24 value=24
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
				1: Manual Mode
				3: Aperture Priority Mode
         exposure_time_absolute 0x009a0902 (int)    : min=1 max=10000 step=1 default=166 value=166 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
```

BenBE, 2022-06-22:
Not sure if I'm reading this right, but looking at the [kernel docs](https://www.kernel.org/doc/html/v4.9/media/uapi/v4l/vidioc-streamon.html), this looks like a bug in `ffmpeg`:

> Capture hardware is disabled and no input buffers are filled (if there are any empty buffers in the incoming queue) until `VIDIOC_STREAMON` has been called. **Output hardware is disabled and no video signal is produced until `VIDIOC_STREAMON` has been called.** The `ioctl` will succeed when at least one output buffer is in the incoming queue.
> 
> **Memory-to-memory devices will not start until `VIDIOC_STREAMON` has been called for both the capture and output stream types.**

(Emphasis mine)

BenBE, 2022-06-22:
@azertyalex Please have a look at the patch in #485 which should address the issue.

Looking at the code for `ffmpeg` it seems no `ioctl` for `VIDIOC_STREAMON` is issued when a `v4l2` device is used for video output, despite the documentation suggesting that both the video producer and consumer need to initiate the stream before any video is produced.

Please report back if the linked patch solves your issue.

closed 2022-06-29 by umlaeute

stephematician, 2025-03-16:
> [@azertyalex](https://github.com/azertyalex) Please have a look at the patch in [#485](https://github.com/v4l2loopback/v4l2loopback/pull/485) which should address the issue.
>
> Looking at the code for [...]

@BenBE - FYI; the call to VIDIOC_STREAMON is actually _non-compliant_ when using file I/O mode of a V4L2 device (i.e. when using `read()` and `write()`) The only time that STREAMON is allowed is when the file (device) is opened and ready for memory mapped (or other streaming) I/O modes. I'm just in the final stages of fixing this up in OBS.

PR #485 (2022-06-29): Reintroduce previous behavior for write call
This allows for implicit start of streaming if no other writer is currently registered for the device.

This still fails the write operation if calling the (implicit) VIDIOC_STREAMON ioctl fails.

Fixes: #484