v4l2loopback/v4l2loopback #494: ffplay can not play video of V4L2 virtual camera with YUYV422 format

BarleyXu, 2022-07-21:
I modified a line of code in scrcpy to generate a stream in YUYV422 format and output it to the V4L2 virtual camera. When I play it with ffplay, an error occurs.

1. app/src/v4l2_sink.c
[image]

2. create a virtual camera
sudo v4l2loopback-ctl add /dev/video0 -b 8

3. set format of virtual camera
sudo v4l2loopback-ctl set-caps /dev/video0 "YUYV:640x480"

4. scrcpy to virtual camera
scrcpy --v4l2-sink /dev/video0

5. test with ffplay
ffplay -pixel_format yuyv422 -i /dev/video0

6. errors
[video4linux2,v4l2 @ 0x7f9654000b80] ioctl(VIDIOC_QUERYBUF): Invalid argument
/dev/video0: Invalid argument

umlaeute, 2022-07-21:
please replace your images with text.

umlaeute, 2022-07-21:
also: what makes you think that the issue is with `v4l2loopback`, rather than with your modified version of `ffmpeg` or with `scrcpy`?

does it work with other consumers? (e.g. `gst-launch v4l2sink ! autovideosink`)

BarleyXu, 2022-07-21:
> also: what makes you think that the issue is with `v4l2loopback`, rather than with your modified version of `ffmpeg` or with `scrcpy`?
> Since I can't be sure that it is the problem of v4l2loopback, I asked the same question in scrcpy, I guess it is caused by the mismatch between the encoding format of scrcpy and the settings of v4l2loopback? Looking forward to your suggestions

> does it work with other consumers? (e.g. `gst-launch v4l2sink ! autovideosink`)
I will try it

umlaeute, 2022-07-22:
what's the point of patching `scrcpy`? why can't you use YUV420P natively (as explained in the scrcpy README)?

BarleyXu, 2022-07-23:
> what's the point of patching `scrcpy`? why can't you use YUV420P natively (as explained in the scrcpy README)?
My requirement is to use unity to play the content of the virtual camera, but unity does not seem to support YUV420P.
https://forum.unity.com/threads/webcamtexture-yuyv-in-videocapture.982866/