v4l2loopback/v4l2loopback #495: Add SECURITY.md

JamieSlome, 2022-07-29:
Hello 👋

I run a security community that finds and fixes vulnerabilities in OSS. A researcher (@benbe) has found a potential issue, which I would be eager to share with you.

Could you add a `SECURITY.md` file with an e-mail address for me to send further details to? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) a security policy to ensure issues are responsibly disclosed, and it would help direct researchers in the future.

Looking forward to hearing from you 👍

(cc @huntr-helper)

umlaeute, 2022-08-01:
Fixed with 432e9a4caf2e70bf0f7f8c0820bfbf57447db6d9

closed 2022-08-01 by umlaeute

JamieSlome, 2022-08-02:
@umlaeute - the URL you provided in the `SECURITY.md` is not working?

Just for reference, you can find the report directly here:
https://huntr.dev/bounties/1b055da5-7a9e-4409-99d7-030280d242d5/

It is private and only accessible to you 👍 If you do not want to sign-up, let me know and I can provide you with a magic URL to view the contents of the report.

umlaeute, 2022-08-02:
the URL should be working (in general); we just had a temporary server problem today.

JamieSlome, 2022-08-03:
@umlaeute - no worries, thanks for the heads up :)

BenBE, 2022-08-03:
The second part of the patch is missing from the recent release. Please recheck the second location mentioned in the report.

umlaeute, 2022-08-03:
@BenBE are you sure? (i think you are confusing the `main` branch with `v0.12.6` (which only contains the fix with respect to `v0.12.5`).

(btw: this ticket seems to be the wrong place to discuss the patch)