v4l2loopback/v4l2loopback #496: Installation failed on CentOS 7

Baiyixuan1113, 2022-08-03:
env:
  CentOS Linux release 7.4.1708 (Core)
  kernel: 3.10.0-693.el7.x86_64
  
I was trying to install v4l2loopback on CentOS 7, and I ran this command:
```
git clone https://github.com/umlaeute/v4l2loopback.git
cd v4l2loopback
make install
```
And I got:
```
make -C /lib/modules/`uname -r`/build M=/root/workspace/v4l2loopback modules_install
make[1]: Entering directory `/usr/src/kernels/3.10.0-693.el7.x86_64'
  DEPMOD  3.10.0-693.el7.x86_64
make[1]: Leaving directory `/usr/src/kernels/3.10.0-693.el7.x86_64'

SUCCESS (if you got 'SSL errors' above, you can safely ignore them)
```
But when I ran:
```
depmod -a
modprobe v4l2loopback
```
I got an ERROR:
```
modprobe: FATAL: Module v4l2loopback not found.
```
After reviewing some issues, I ran this:
```
sudo make install V=1; echo $?
```
And I got this ERROR:
```
make -C /lib/modules/`uname -r`/build M=/root/workspace/v4l2loopback modules_install
make[1]: Entering directory `/usr/src/kernels/3.10.0-693.el7.x86_64'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /lib/modules/3.10.0-693.el7.x86_64/extra
make -f ./scripts/Makefile.modinst
  /bin/sh ./scripts/depmod.sh /sbin/depmod 3.10.0-693.el7.x86_64 ""
make[1]: Leaving directory `/usr/src/kernels/3.10.0-693.el7.x86_64'

SUCCESS (if you got 'SSL errors' above, you can safely ignore them)

0
```
And I can't find `v4l2loopback.ko` in `/lib/modules/3.10.0-693.el7.x86_64/extra/`.

Looking forward to your reply.

umlaeute, 2022-08-03:
we are not really supporting linux-3.x anymore... :thinking:

Baiyixuan1113, 2022-08-04:
> we are not really supporting linux-3.x anymore... 🤔

ok, thanks for your reply, I tried to install on Ubuntu18.04 and it was successful.
And of course, thank you very much for your project.

closed 2022-08-04 by Baiyixuan1113