v4l2loopback/v4l2loopback #498: Commit 4b05265da broke v4l2loopback-dkms on Debian unstable

AlexTMjugador, 2022-08-05:
#### Description

I've just updated my Debian unstable system to the latest kernel and v4l2loopback-dkms versions, and to my unpleasant surprise I was greeted with the fact that the v4l2loopback module now fails to build because the dkms.conf configuration thinks that my kernel is missing the needed V4L2 configuration options:

[image]

Needless to say, this makes the v42loopback module not build at all. I've looked at the `/lib/modules/5.18.0-3-amd64/build/.config` file and confirmed that `CONFIG_VIDEO_V4L2` is not explicitly mentioned, but other V4L2-related configurations were enabled, which I think that transitively imply `CONFIG_VIDEO_V4L2` anyway. It's worth mentioning that everything worked fine on a previous v4l2loopback-dkms version with the 5.18.0-2 kernel, and I doubt that Debian really stopped enabling the necessary kernel features in such a small version bump.

##### Steps to reproduce

  1. Install the v4l2loopback-dkms on an updated Debian unstable system
  2. Watch how it refuses to build due to dkms.conf settings

##### Expected results

It should just work.

#### Environment

  * `v4l2loopback` version: 0.12.7

  * kernel version: `Linux hostname 5.18.0-3-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.14-1 (2022-07-23) x86_64 GNU/Linux`

  * Distribution (+version):

```
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux bookworm/sid
Release:        unstable
Codename:       sid
```

umlaeute, 2022-08-05:
the issue seems to be that with 5.18, the kernel config variables for enabling v4l2 has been renamed from `CONFIG_VIDEO_V4L2` to `CONFIG_VIDEO_DEV`).

Go figure.

AlexTMjugador, 2022-08-05:
Thank you for looking into this. Would you like that submitting a PR that also checks for `CONFIG_VIDEO_DEV`? 😄 

> this is actually an issue with the Debian package (so you should have reported it there..)

I've thought about that, but the Debian package didn't patch the dkms.conf file with the check, so I figured out that the problem was present upstream and that some paperwork could be saved by reporting the issue here.

closed 2022-08-05 by umlaeute