v4l2loopback/v4l2loopback #500: New Feature Release?

xnox, 2022-08-16:
```
$ git diff --stat v0.12.7 main
...
 35 files changed, 2623 insertions(+), 638 deletions(-)
```

Are you going to make a new major release from main branch soon, which includes all accumulated features and changes since last v0.12.7 point release?

I am being asked to ship 0e12ea4fe3920a574adc1eba5c6cfd992a8a52a7 /dev/v4l2loopback in ubuntu, and I feel uneasy to introduce that as a stable ABI when it hasn't yet been released upstream as part of any tagged releases.

Do you recommend shipping main branch snapshot as stable?

umlaeute, 2022-08-16:
the problem is, that split-devices (separate `CAPTURE` and `OUTPUT` devices) will break the ABI for interfacing with `/dev/v4l2loopback`.

this is the main reason why there hasn't been a release yet.

closed 2022-08-16 by xnox

BenBE, 2022-08-16:
As far as I understand the situation the PR for split device support (#359) has been stalled for over a year now and its author noted that he was unlikely to complete that PR any time soon (https://github.com/umlaeute/v4l2loopback/pull/359#issuecomment-1133943818). Thus unless someone was to take over the development of that PR you are kinda deadlocking any new releases indefinitely.

Is there a rough list of where the required ABI changes will likely be breaking ones? How likely is it that through some planning both the current and new behavior could be accommodated through the same ABI (be it with changes to the current ABI)? If necessary let's take that discussion over to PR #359 …

xnox, 2022-08-17:
separately, if we want to publish a different module with different abi we could start a different project v4l2splitloopback and publish that too. it seems like currently module is good for many people and it makes sense to keep it stable. and if there are new usecases for different way of doing things, it may make sense to publish that as a separate brand new namespaced module too.