v4l2loopback/v4l2loopback #501: [feature request] Support V4L2_MEMORY_DMABUF

daijh, 2022-08-17:
Ask V4L2_MEMORY_DMABUF feature support in v4l2loopback. It could be helpful for hardware accelerator usage, e.g. GL and HW encoder.

daijh, 2022-08-17:
WIP a POC to implement this feature. Will submit PR for review soon.

daijh, 2022-11-14:
Sorry for late reply.
I had an under-development (buggy and limitation) driver which is able to accept dmabuf import.
It changes a few things over original driver.

It may need ~3 week to cleanup for experiment usage.

chrisiberg, 2022-11-23:
Would be great, can't await to test!

umlaeute, 2023-05-26:
afaic the lack of support for `V4L2_MEMORY_DMABUF` is the reason why people cannot use the proprietary `nvcamerasrc` for nVidia's Tegra/Jetson platform (which [depends on `V4L2_MEMORY_DMABUF`](https://docs.nvidia.com/jetson/archives/r35.3.1/DeveloperGuide/text/SD/Multimedia/AcceleratedGstreamer.html?highlight=nvv4l2camerasrc#progressive-capture-using-nvv4l2camerasrc))

activey, 2023-09-03:
Yeah, it would be great to have it because of #128

chrisiberg, 2023-09-15:
Did someone try daijh expbuf-dmabuf?
https://github.com/daijh/v4l2loopback/tree/expbuf-dmabuf

I had no luck with it, error compiling in yocto with kernel 5.15.94
---
v4l2loopback.c:3593:28: error: storage size of 'map' isn't known
|  3593 |         struct dma_buf_map map;
---

daijh, 2023-09-18:
> Did someone try daijh expbuf-dmabuf? https://github.com/daijh/v4l2loopback/tree/expbuf-dmabuf
> 
> I had no luck with it, error compiling in yocto with kernel 5.15.94
> 
> v4l2loopback.c:3593:28: error: storage size of 'map' isn't known | 3593 | struct dma_buf_map map;

I am glad to hear that you are interested in the expbuf-dmabuf repo. 
It has been verified to work on Intel MIPI cameras, and probably with USB camera,

It required significant changes to the original v4l2loopback code, making it difficult to backport.

May I know how you plan to use expbuf-dmabuf?

chrisiberg, 2023-09-18:
@daijh 
My plans are to use it for RTSP and NDI streams, decode it via VAAPI to view it in chromium with low latency.
It's now working without DMA but is limited in resolution because of the high CPU usage, without DMA I just can use SW-decoding.
Is it possible to connect to Intel VAAPI with DMA, or is it just limited to devices as input?

daijh, 2023-09-19:
Could you share if any GStreamer command? I could do some experiment on it.

ksquen, 2024-07-16:
hi @daijh,

I'm currently use your version of  v4l2loopback expbuf-dmabuf with gstreamer.

I can't use this pipeline to use as producer like in the wiki: gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video1 . Maybe it only support vma ?
But ffmpeg use ok ( ffmpeg -re -i h264-hd-30.mp4 -f v4l2 /dev/video3 )
Can you give me some examples on what you use it for and how to use it.

Thanks.

stephematician, 2024-12-16:
Is there a simpler way to do this via  `udmabuf` to wrap the buffers, see https://github.com/torvalds/linux/blob/master/drivers/dma-buf/udmabuf.c?

stephematician, 2024-12-17:
I'll start a branch and work on this for a bit; seems like a good challenge. I'll poach a lot of inspiration from "udmabuf" as I sort through the `dma_buf_ops` struct. If anyone has any thoughts, feel free to add to the discussion here: https://github.com/stephematician/v4l2loopback/discussions/1