v4l2loopback/v4l2loopback #505: GStreamer v4l2sink not able to write to virtual device

ivanzivotic, 2022-10-12:
Hi,
I'm running the v4l2loopback on the system that has Ubuntu 18.04.6 with GStreamer version 1.14.5 and kernel version 5.4.0-126-generic. I'm trying to receive MPEG-TS video stream from a UDP source, demux it using tsdemux, decode video using vaapih264dec (with Intel UHD Graphics card). The pipeline looks like this: `gst-launch-1.0 udpsrc port=11024 ! tsdemux ! multiqueue ! h264parse config-interval=-1 ! vaapih264dec base-only=true low-latency=true ! v4l2sink device=/dev/video42 sync=false async=false`. The error message I'm getting looks like this:
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Got context from element 'vaapidecode_h264-0': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayGBM\)\ gldisplaygbm0";
Got context from element 'vaapidecode_h264-0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayDRM\)\ vaapidisplaydrm1";
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstUDPSrc:udpsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstUDPSrc:udpsrc0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.561290621
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
Although it looks like the error is in the `udpsrc` element, it works correctly when `fakesink` element is used instead of `v4l2sink`. The configuration file used when loading the module is:
```
options v4l2loopback video_nr=42
options v4l2loopback card_label="virtcam"
options v4l2loopback exclusive_caps=1
options v4l2loopback max_buffers=2
```
When the system boots up and the kernel module loads, before running any pipeline, the output of the command `v4l2-ctl --all -d /dev/video42` looks like this:
```
Driver Info (not using libv4l2):
      Driver name   : v4l2 loopback
      Card type     : virtcam
      Bus info      : platform:v4l2loopback-000
      Driver version: 5.4.203
      Capabilities  : 0x85200002
        Video Output
        Read/Write
        Streaming
        Extended Pix Format
        Device Capabilities
      Device Caps   : 0x05200002
        Video Output
        Read/Write
        Streaming
        Extended Pix Format
Priority: 2
Video output: 0 (loopback in)
Format Video Output:
      Width/Height      : 0/0
      Pixel Format      : 'BGR4'
      Field             : None
      Bytes per Line    : 0
      Size Image        : 0
      Colorspace        : sRGB
      Transfer Function : Default (maps to sRGB)
      YCbCr/HSV Encoding: Default (maps to ITU-R 601)
      Quantization      : Default (maps to Full Range)
      Flags             :
Streaming Parameters Video Capture:
      Frames per second: 30.000 (30/1)
      Read buffers     : 2
Streaming Parameters Video Output:
      Frames per second: 30.000 (30/1)
      Write buffers    : 2

User Controls

                    keep_format 0x0098f900 (bool)   : default=0 value=0
              sustain_framerate 0x0098f901 (bool)   : default=0 value=0
                        timeout 0x0098f902 (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io 0x0098f903 (bool)   : default=0 value=0
```
When I run the pipeline `gst-launch-1.0 videotestsrc ! video/x-raw,width=1280,height=720 ! tee ! v4l2sink device=/dev/video42` the output of the previous command looks like this (as expected):
```
Driver Info (not using libv4l2):
      Driver name   : v4l2 loopback
      Card type     : virtcam
      Bus info      : platform:v4l2loopback-000
      Driver version: 5.4.203
      Capabilities  : 0x85200002
        Video Output
        Read/Write
        Streaming
        Extended Pix Format
        Device Capabilities
      Device Caps   : 0x05200002
        Video Output
        Read/Write
        Streaming
        Extended Pix Format
Priority: 2
Video output: 0 (loopback in)
Format Video Output:
      Width/Height      : 1280/720
      Pixel Format      : 'YUYV'
      Field             : None
      Bytes per Line    : 2560
      Size Image        : 1843200
      Colorspace        : sRGB
      Transfer Function : sRGB
      YCbCr/HSV Encoding: ITU-R 601
      Quantization      : Limited Range
      Flags             :
Streaming Parameters Video Capture:
      Frames per second: 30.000 (30/1)
      Read buffers     : 2
Streaming Parameters Video Output:
      Frames per second: 30.000 (30/1)
      Write buffers    : 2

User Controls

                    keep_format 0x0098f900 (bool)   : default=0 value=0
              sustain_framerate 0x0098f901 (bool)   : default=0 value=0
                        timeout 0x0098f902 (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io 0x0098f903 (bool)   : default=0 value=0
```

On the other hand, if I try running the desired pipeline, the output of the `v4l2-ctl` command looks like this:
```
Driver Info (not using libv4l2):
      Driver name   : v4l2 loopback
      Card type     : virtcam
      Bus info      : platform:v4l2loopback-000
      Driver version: 5.4.203
      Capabilities  : 0x85200002
        Video Output
        Read/Write
        Streaming
        Extended Pix Format
        Device Capabilities
      Device Caps   : 0x05200002
        Video Output
        Read/Write
        Streaming
        Extended Pix Format
Priority: 2
Video output: 0 (loopback in)
Format Video Output:
      Width/Height      : 48/32
      Pixel Format      : 'YVYU'
      Field             : None
      Bytes per Line    : 96
      Size Image        : 3072
      Colorspace        : sRGB
      Transfer Function : Default (maps to sRGB)
      YCbCr/HSV Encoding: Default (maps to ITU-R 601)
      Quantization      : Default (maps to Limited Range)
      Flags             :
Streaming Parameters Video Capture:
      Frames per second: 30.000 (30/1)
      Read buffers     : 2
Streaming Parameters Video Output:
      Frames per second: 30.000 (30/1)
      Write buffers    : 2

User Controls

                    keep_format 0x0098f900 (bool)   : default=0 value=0
              sustain_framerate 0x0098f901 (bool)   : default=0 value=0
                        timeout 0x0098f902 (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io 0x0098f903 (bool)   : default=0 value=0
```

You can see that the resolution of the device is set to 48x32, although the video that is sent has 1280x720 resolution. I've tried adding the `tee` element in the pipeline between `vaapih264dec` and `v4l2sink` elements, but the issue still occurs.

This type of pipeline used to work correctly before for us and we suspect that the issue may be occurring due to Linux kernel version update, although we are not sure if the video being sent to us may have changed (in terms of format), which may also be the cause of this.

Thanks in advance

umlaeute, 2022-10-13:
> Although it looks like the error is in the `udpsrc` element, it works correctly when `fakesink` element is used instead of `v4l2sink`. 

this really doesn't mean much: the `fakesink` element accepts (by definition) everything. the `v4l2sink` can only accept some video formats.
a better sink element for comparing would be some video sink (e.g. `autovideosink`)

you should start with finding out which caps the elements agree on, and then check what the differences are between a working and a non-working pipeline.

ivanzivotic, 2022-10-14:
UPDATE:

The issue was that the framerate property of the video was not being set in the caps of the elements (it stayed 0) so the `v4l2sink` could not accept it. After adding `videorate ! video/x-raw,framerate=30/1` before the `v4l2sink` element it is working correctly.

closed 2022-10-14 by ivanzivotic