v4l2loopback/v4l2loopback #506: build failure in RHEL7.9

aash666, 2022-10-18:
### Step 1: Describe your environment

  * `v4l2loopback` version: 0.12.5
  * kernel version: 3.10.0-1160.76.1.el7.x86_64
  * Distribution (+version): Red Hat Enterprise Linux Server release 7.9 (Maipo

### Step 2: Describe the problem:
Unable to build v4l2loopback in RHEL7.9

#### Steps to reproduce:

```
version=0.12.5
curl -L https://github.com/umlaeute/v4l2loopback/archive/v${version}.tar.gz | sudo tar xvz -C /usr/src
sudo dkms add -m v4l2loopback -v ${version}
sudo dkms build -m v4l2loopback -v ${version}
```

#### Observed Results:

```
# sudo dkms build -m v4l2loopback -v ${version}
Sign command: /lib/modules/3.10.0-1160.76.1.el7.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Deprecated feature: REMAKE_INITRD

Building module:
Cleaning build area...
make -j2 KERNELRELEASE=3.10.0-1160.76.1.el7.x86_64 KERNEL_DIR=/lib/modules/3.10.0-1160.76.1.el7.x86_64/build all...(bad exit status: 2)
Error! Bad return status for module build on kernel: 3.10.0-1160.76.1.el7.x86_64 (x86_64)
Consult /var/lib/dkms/v4l2loopback/0.12.5/build/make.log for more information.
```

```
# cat /var/lib/dkms/v4l2loopback/0.12.5/build/make.log
DKMS make.log for v4l2loopback-0.12.5 for kernel 3.10.0-1160.76.1.el7.x86_64 (x86_64)
Tue Oct 18 16:31:11 UTC 2022
Building v4l2-loopback driver...
make -C /lib/modules/3.10.0-1160.76.1.el7.x86_64/build M=/var/lib/dkms/v4l2loopback/0.12.5/build modules
make[1]: Entering directory `/usr/src/kernels/3.10.0-1160.76.1.el7.x86_64'
  CC [M]  /var/lib/dkms/v4l2loopback/0.12.5/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.12.5/build/v4l2loopback.c: In function ‘vidioc_querycap’:
/var/lib/dkms/v4l2loopback/0.12.5/build/v4l2loopback.c:701:11: error: ‘struct video_device’ has no member named ‘device_caps’
  dev->vdev->device_caps = cap->device_caps = cap->capabilities = capabilities;
           ^
make[2]: *** [/var/lib/dkms/v4l2loopback/0.12.5/build/v4l2loopback.o] Error 1
make[1]: *** [_module_/var/lib/dkms/v4l2loopback/0.12.5/build] Error 2
make[1]: Leaving directory `/usr/src/kernels/3.10.0-1160.76.1.el7.x86_64'
make: *** [v4l2loopback.ko] Error 2
```

#### Expected Results:

  * Expected to build v4l2loopback successfully in RHEL7.9
  
#### Relevant Code:

 * NA

aash666, 2022-10-20:
Tried the same with 0.12.7 version and build is failing with the same error. I could see that 0.12.4 is building successfully.

umlaeute, 2022-10-20:
> kernel version: 3.10.0-1160.76.1.el7.x86_64

i think this is why the [README](https://github.com/umlaeute/v4l2loopback#kernels) says:
> = `3.0.0` might work

i'm sorry to say, but there are no plans to actively support kernels of the 3.x line on a system that has entered the "product retirement" phase of its life cycle.

closed 2022-10-20 by umlaeute