v4l2loopback/v4l2loopback #508: Linux 6.1 support: 'BUILD_EXCLUSIVE directive which does not match this kernel/arch'

Firestar99, 2022-11-01:
### Step 2: Describe your environment

  * `v4l2loopback` version: 0.12.5-1ubuntu5 (from apt show v4l2loopback-dkms)
  * kernel version: 6.1.0-060100rc1-generic
  * Distribution (+version): Ubuntu 22.04 "jammy"

### Step 3: Describe the problem:

#### Steps to reproduce:

  1. install Ubuntu kernel 6.1-rc3 for amd64 from https://kernel.ubuntu.com/~kernel-ppa/mainline/v6.1-rc3/
  2. sudo apt-get install v4l2loopback-dkms

#### Observed Results:

Running `sudo apt-get install v4l2loopback-dkms`:
```
Setting up v4l2loopback-dkms (0.12.5-1ubuntu5) ...
Loading new v4l2loopback-0.12.5 DKMS files...
Building for 6.1.0-060100rc1-generic 6.1.0-060100rc3-generic
Building initial module for 6.1.0-060100rc1-generic
Error! The /var/lib/dkms/v4l2loopback/0.12.5/6.1.0-060100rc1-generic/x86_64/dkms.conf for module v4l2loopback includes a BUILD_EXCLUSIVE directive which does not match this kernel/arch.
This indicates that it should not be built.
Skipped.
Building initial module for 6.1.0-060100rc3-generic
Error! The /var/lib/dkms/v4l2loopback/0.12.5/6.1.0-060100rc3-generic/x86_64/dkms.conf for module v4l2loopback includes a BUILD_EXCLUSIVE directive which does not match this kernel/arch.
This indicates that it should not be built.
Skipped.
```

#### Expected Results:

Working Linux 6.1 support :D

#### Additional Context:
I want to use obs virtual camera and need to use Linux 6.1 as it fixes vendor ACPI suspend bugs on Ryzen 6000 mobile series processors.

Thanks for any support you can offer :D

umlaeute, 2022-11-01:
i currently don't have access to a machine with a Linux-6.1 kernel, but the `BUILD_EXCLUSIVE` directive should only be triggered if your kernel does not have v4l2 support.
can you confirm that your kernel has been build with `CONFIG_VIDEO_DEV` support?

Firestar99, 2022-11-02:
I'm not exactly sure ow to check these prebuild kernels, but I can do this:
```
$ modinfo videodev
filename:       /lib/modules/6.1.0-060100rc3-generic/kernel/drivers/media/v4l2-core/videodev.ko
alias:          char-major-81-*
license:        GPL
description:    Video4Linux2 core driver
author:         Alan Cox, Mauro Carvalho Chehab <mchehab@kernel.org>, Bill Dirks, Justin Schoeman, Gerd Knorr
srcversion:     97D12D428AABEE85C79C48B
depends:        mc
retpoline:      Y
intree:         Y
name:           videodev
vermagic:       6.1.0-060100rc3-generic SMP preempt mod_unload modversions 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        14:A6:FA:5F:FF:B6:8A:39:A4:7B:7E:71:FB:49:9B:F8:17:C7:BB:15
sig_hashalgo:   sha512
signature:      21:E5:66:6F:82:0C:B9:37:CA:A4:63:72:73:15:F5:33:61:30:53:B9:
                92:FD:0D:C7:E0:0D:2D:0B:4C:9F:6B:94:E6:94:DA:37:F6:1C:70:2E:
                E3:0B:EB:48:16:32:A7:E5:56:2D:11:41:E9:AD:F7:F4:C4:48:CE:D7:
                35:24:2E:EA:84:FC:78:42:C9:0F:A2:5C:B0:08:79:94:6E:D5:59:4C:
                59:48:68:DD:DB:10:1F:B1:C1:9A:F6:0B:0F:32:29:B6:F5:53:15:EE:
                35:A0:02:2F:73:F4:CB:4F:CF:AB:5F:D2:AB:26:E7:BB:E7:8D:E2:8B:
                AA:7E:71:96:70:1F:E4:AB:14:22:B6:F9:72:ED:54:67:34:6E:49:15:
                6E:6E:B2:B2:EA:69:BB:5F:54:05:15:F6:A3:0E:3A:54:E3:F9:E4:48:
                E9:C0:24:7E:39:15:D9:74:7E:EC:43:1C:34:AF:14:7B:AB:01:8D:FE:
                64:04:4D:87:FC:B7:75:03:D6:EB:93:35:53:D4:51:48:4D:1B:19:0F:
                AF:AE:E6:54:A2:60:78:70:14:A5:8A:BA:1D:73:5C:33:C2:88:F4:58:
                73:D4:B9:FB:05:D9:0F:38:12:C5:8F:AD:AE:13:44:0C:CF:DA:43:F4:
                74:06:AB:10:B9:B2:0E:B8:54:13:1A:08:A1:38:A8:BB:36:4F:7F:47:
                45:BE:74:AB:58:3F:29:5B:32:AE:B4:B6:E7:AE:57:2C:18:E4:F0:5A:
                4C:5F:EE:BD:64:1A:E7:E7:91:5E:6F:A6:12:01:4C:5C:EE:FA:68:F7:
                DC:CF:92:94:7A:88:52:CA:8C:D4:A3:9F:D5:87:DA:BD:5C:33:3B:CD:
                BF:31:CF:BF:8C:47:4A:14:36:92:C9:03:C0:3C:6A:EF:D3:44:07:9C:
                CB:C6:E5:82:C1:7C:6A:3B:C8:7C:6A:8A:0D:10:F5:11:0E:A2:D7:6E:
                23:59:7F:EE:28:1F:A2:DD:29:6D:B9:59:2F:94:7E:A3:5B:B0:28:80:
                1E:DC:9F:6D:F3:2A:5F:08:EB:22:44:B8:C6:F0:78:18:B4:84:F9:46:
                B0:29:69:CD:0A:33:EF:1A:D6:16:70:5A:DE:6F:6E:A3:4D:76:3A:1D:
                96:83:5B:C3:22:1A:FF:61:1C:EB:2F:76:D7:8E:A1:0F:F5:6F:06:30:
                39:BE:56:EA:D0:D4:16:29:D3:88:07:32:66:93:76:DF:9A:15:10:D9:
                58:FC:7D:59:43:90:05:BD:75:4C:95:F6:12:A7:35:3B:03:18:E8:5A:
                3C:15:D9:DA:0C:CC:2E:A3:7A:92:C6:49:CA:12:53:0E:5F:3D:FB:78:
                14:C4:A4:80:A3:84:A7:A5:6A:0C:3F:D8
```

So I assume it must be present. Also I can use my webcam in vlc and obs just fine.

rauldipeas, 2022-11-15:
> i currently don't have access to a machine with a Linux-6.1 kernel, but the `BUILD_EXCLUSIVE` directive should only be triggered if your kernel does not have v4l2 support. can you confirm that your kernel has been build with `CONFIG_VIDEO_DEV` support?

I'm getting the same error when try to install `v4l2loopback-dkms` on Kubuntu 22.04 with latest mainline kernel, 6.1.0-rc5 for at this moment.

I've checked for this `CONFIG_VIDEO_DEV` on kernel build config and it's set to as `CONFIG_VIDEO_DEV=m`.

The `dkms` log show an information about `gcc` mismatch.

umlaeute, 2022-11-15:
i've just installed the `linux-headers-6.1.0-0-amd64` as it is currently available in Debian/experimental (`6.1~rc3-1~exp1`), and `v4l2loopback-dkms` builds and installs fine:

```
# dkms build v4l2loopback/0.12.7
Sign command: /usr/lib/linux-kbuild-6.0/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j8 KERNELRELEASE=6.0.0-2-amd64 KERNEL_DIR=/lib/modules/6.0.0-2-amd64/build all...
Signing module /var/lib/dkms/v4l2loopback/0.12.7/build/v4l2loopback.ko
Cleaning build area...
#
```

so I'm inclined to close this as "works-for-me".
however, I'm a bit unsure whether the actual problem you are facing is
- because of an outdated version of `v4l2loopback` (e.g. Ubuntu Jammy/22.04 ships with `v4l2loopback-0.12.5`; at the time of writing, the last release here (and which I test) is `v4l2loopback-0.12.7`)
- an issue with the kernel packaging ("6.1.0-060100rc1-generic" resp "mainline kernel 6.1.0-rc5" sounds all very experimental to me; it seems that there are no (official?) Ubuntu packages for these kernels *at all*).

probably it's both.
you should first try to install `v4l2loopback-0.12.7` (the package from kinetic should work just fine; so download and install it manually). the changelog of the 0.12.7 package says "* Fix DKMS build on linux>=5.18" which *might* be related to your issue.
if this doesn't work try some officially packaged kernel.

closed 2022-11-15 by umlaeute

Firestar99, 2022-11-15:
In the mean time I updated to Ubuntu 22.10 kinetic and it works just fine!
I can use virtual camera in OBS and can watch it in discord (didn't show up in vlc for some reason).

Installed it from the official kinetic repos: package `v4l2loopback-dkms` version `0.12.7-2ubuntu2`.

rauldipeas, 2022-11-15:
This is how I fixed it on Kubuntu 22.04 with Linux 6.1.0-rc5...
```
pkcon install gcc-12
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-12 12
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 11
wget -qc --show-progress http://mirrors.kernel.org/ubuntu/pool/universe/v/v4l2loopback/$(wget -qO- http://mirrors.kernel.org/ubuntu/pool/universe/v/v4l2loopback/|grep v4l2loopback-dkms|grep .deb|tail -n4|head -n1|cut -d '"' -f2)
pkcon install-local ./v4l2loopback-dkms*.deb
```