v4l2loopback/v4l2loopback #510: can not change the format to V4L2_PIX_FMT_Y10

Katze719, 2022-11-08:
sudo v4l2-ctl -d /dev/video7 --all

Driver Info:
        Driver name      : v4l2 loopback
        Card type        : loopy doopy
        Bus info         : platform:v4l2loopback-007
        Driver version   : 5.15.68
        Capabilities     : 0x85200002
                Video Output
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x05200002
                Video Output
                Read/Write
                Streaming
                Extended Pix Format
Priority: 2
Video output: 0 (loopback in)
Format Video Output:
        Width/Height      : 640/480
        Pixel Format      : 'BGR4' (32-bit BGRA/X 8-8-8-8)
        Field             : None
        Bytes per Line    : 2560
        Size Image        : 1228800
        Colorspace        : sRGB
        Transfer Function : Default (maps to sRGB)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Full Range)
        Flags             :
Streaming Parameters Video Capture:
        Frames per second: 25.000 (25/1)
        Read buffers     : 2
Streaming Parameters Video Output:
        Frames per second: 25.000 (25/1)
        Write buffers    : 2

User Controls

                    keep_format 0x0098f900 (bool)   : default=0 value=1
              sustain_framerate 0x0098f901 (bool)   : default=0 value=0
                        timeout 0x0098f902 (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io 0x0098f903 (bool)   : default=0 value=0

sudo v4l2loopback-ctl set-caps /dev/video7 "Y10:640x480"

nothing changed

umlaeute, 2022-11-09:
which versio nof `v4l2loopback` is this? (if compiled from the git-repository, i would need the exact commitish)

umlaeute, 2022-11-09:
ah, you must pass the exact FOURCC-value, which is `Y10 ` (note the trailing space).

```sh
$ v4l2loopback-ctl set-caps /dev/video0 "Y10 :640x480"
$ v4l2-ctl -d /dev/video0 --get-fmt-video-out
Format Video Output:
	Width/Height      : 640/480
	Pixel Format      : 'Y10 ' (10-bit Greyscale)
	Field             : None
	Bytes per Line    : 800
	Size Image        : 384000
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
```

closed 2022-11-09 by umlaeute