v4l2loopback/v4l2loopback #514: Release packages contain version 0.6 of v4l2loopback-ctl

copiltembel, 2022-12-15:
Following the steps described in [`DKMS`](https://github.com/umlaeute/v4l2loopback/tree/develop#dkms) section of the README, it looks like the releases contain the hard-coded(?)  version 0.6 of v4l2loopback-ctl.

To reproduce:

```
$ wget https://github.com/umlaeute/v4l2loopback/archive/v0.12.7.tar.gz
$ tar xvf v0.12.7.tar.gz
$ ./v4l2loopback-0.12.7/utils/v4l2loopback-ctl -v
./v4l2loopback-0.12.7/utils/v4l2loopback-ctl 0.6

Copyright (c) 2012 IOhannes m zmoelnig <zmoelnig@iem.at>
Copyright (c) 2012 Anton Novikov (random.plant@gmail.com)
```

Same for v0.12.6. 

Also, the procedure described in the mentioned section of the README does not install the v4l2loopback-ctl tool. Is this by design?

umlaeute, 2022-12-15:
>  releases contain the hard-coded(?) version 0.6 of v4l2loopback-ctl.

right.
the number is hardcoded, and was never updated.

however
- this has been fixed in `main`
- the released versions are released and thus immutable
- this issue seems so minor, that it doesn't warrant a bugfix release.

also: i don't think that `dkms` provides a mean to install user-space software (but i might be totally misguided; i haven't even checked!), so there's not much you can do here.

copiltembel, 2022-12-16:
Makes sense. 

I suggest adding a couple of notes in the README. Something like:

- The releases prior to and including [v0.12.7](https://github.com/umlaeute/v4l2loopback/releases/tag/v0.12.7) contain an old version of the v4l2loopback-ctl tool
- in DKMS section: "note that this will NOT install the v4l2loopback-ctl, you will have to do it yourself."

closed 2022-12-16 by copiltembel

umlaeute, 2022-12-16:
> The releases prior to and including [v0.12.7](https://github.com/umlaeute/v4l2loopback/releases/tag/v0.12.7) contain an old version of the v4l2loopback-ctl tool

they don't.
that is: they include the `v4l2loopback-ctl` tool that (spuposedly) works with the `v4l2loopback` module that they ship with.
it's just that nobody bothered to synchronize the version number between the cmdline tool and the kernel module.

> in DKMS section: "note that this will NOT install the v4l2loopback-ctl, you will have to do it yourself."

that's a good idea.
PRs welcome.

copiltembel, 2022-12-16:
> they don't. that is: they include the `v4l2loopback-ctl` tool that (spuposedly) works with the `v4l2loopback` module that they ship with. it's just that nobody bothered to synchronize the version number between the cmdline tool and the kernel module.
> 

I have to admit I am confused. Using release `v0.12.7`:

```
./v4l2loopback-ctl 
Usage:
    ./v4l2loopback-ctl [command] [args]

Commands:
     set-fps <FPS> /dev/videoX  Set device fps (if format is already
                                negotiated).

     set-timeout-image <IMAGE> /dev/videoX  Set timeout on device, and 
                                            optionally the placeholder picture,
                                            instead of default null frame.

     set-caps <CAPS> /dev/videoX  Fix format for the device (using GStreamer
                                  caps), so it will be maintained until reset,
                                  regardless of future readers/writers.

Options:
    FPS      frames per second, either as integer ('30') or fraction ('50/2')
    IMAGE    image file ("no-input.png"), or 'videotestsrc'
    CAPS     GStreamer caps ("video/x-raw, format=UYVY, width=640, height=480"),
             or 'any'
    /dev/videoX  the v4l2-loopback device file you want to act on
```

This looks way different than what the version in `main` outputs:

```
./v4l2loopback-ctl 
usage: ./v4l2loopback-ctl [general commands]

 general commands
 ================
	-v : print version and exit
	-h : print this help and exit
	 add {<flags>} [<device> [<outputdevice>]]
	 delete <device>
	 query <device>
	 set-fps <device> <fps>
	 get-fps <device>
	 set-caps <device> <caps>
	 get-caps <device>
	 set-timeout-image {<flags>} <device> <image>

 adding devices ('add')
 ======================
[...]
```

umlaeute, 2022-12-16:
> 
> I have to admit I am confused. Using release `v0.12.7`:
> 
[...]

> 
> This looks way different than what the version in `main` outputs:
> 

[...]

the version in `main` is a total rewrite and has much more powers (just like the kernel module now has more powers).

why do you think that the version in `main` ought to be the same as the version in the last release? (that assumption confuses *me*)

copiltembel, 2022-12-16:
I wasn't aware of the radical changes, I should have checked first.

archfz, 2022-12-26:
Thanks for this tool, but when is a proper release planned? The main should contain the last proper release, the documentation got me confused and wasting time.

raffaem, 2023-01-08:
> Thanks for this tool, but when is a proper release planned? The main should contain the last proper release, the documentation got me confused and wasting time.

So do I. I was trying to run `v4l2loopback-ctl add` on 0.12.7 (/ advertised as 0.6)

aramg, 2023-03-07:
Just to confirm, is the plan to get this up-to-date with the next release? i.e. either upgrade the prebuilt binary that's currently sitting in all the release branches (tags), or, replace this prebuilt binary with the source + make file and have the binary built during packaging downstream?