v4l2loopback/v4l2loopback #519: Gstreamer RTSP videostream freezes after first frame

Gerbaum, 2023-01-18:
My Spec:

Ubuntu 20.04 on imx6 quadcore board
Kernel 5.4.45
v4l2loopback version: 0.12.7

Hi,
my first gstreamer pipe is:

gst-launch-1.0 -v v4l2src device=/dev/video4 ! image/jpeg,width=1920,height=1080,framerate=30/1 ! videorate drop-only=true ! image/jpeg,framerate=10/1 ! jpegdec ! video/x-raw,width=1920,height=1080,format=I420 ! videoconvert !  v4l2sink device=/dev/video15 

Pipeline takes MJPEG Stream from USB Camera, limits framerate to 10 fps, decodes to Raw, and after a videoconvert it sends the stream to loopback device 15.

My Second Stream is:

gst-launch-1.0 -e -v  rtspclientsink name=s location=rtsp://localhost:8554/mystream v4l2src device=/dev/video15 ! video/x-raw,width=1920,height=1080,format=I420,framerate=30/1  ! queue !  videorate drop-only=true ! video/x-raw,width=1920,height=1080,format=I420,framerate=10/1 ! v4l2h264enc extra-controls="controls,video_bitrate=2000000;" ! video/x-h264, framerate=10/1 ! s.sink_1

Pipelines takes Raw Stream from loopback device 15, limits fps to 10 and encodes to h264. Last part is a sink to rtsp server from RTSP-Simple-Server running on the same device.

Running both streams together no error happens.

But if I want to watch the rtsp stream in vlc player, there is freeze after first frame. After a raise of vlc buffer time to 3 seconds, there is about one fps.

The pipe works without problems in vlc player, after skipping loopback part and connect the stream to one pipe. 

Could there be a problem with the timing of the frames, cause I send 10 fps to loopback15 with first pipe and grap 30 fps with second one?

Thanks

schplorg, 2023-03-20:
Try a different commit or stable release, fixed issues for me with streaming from OBS to virtual camera
https://obsproject.com/forum/threads/only-getting-a-still-image-first-frame-from-virtual-camera.164890/

pfactum, 2023-03-23:
The issue is somewhere around this:

* [Reset V4L2_BUF_FLAG_MAPPED if use_count <= 0](https://github.com/umlaeute/v4l2loopback/commit/f3f40c1be83abe1dd8f805519b293a85e17778ce)
* [Track active readers](https://github.com/umlaeute/v4l2loopback/commit/cc389462180f1d91401b51c415c54c9aa930d04e)
* [Support V4L2_EVENT_PRI_CLIENT_USAGE](https://github.com/umlaeute/v4l2loopback/commit/ceed405e453a6ab746d94334c03b4679dfb93ac6)
* [Fix signed integer overflows in increments.](https://github.com/umlaeute/v4l2loopback/commit/7c4ca0d207ab1338db2aae8e4a78a04a35fe25c8)
* [v4l2loopback: a new offset for V4L2_EVENT_PRI_CLIENT_USAGE](https://github.com/umlaeute/v4l2loopback/commit/c7a5cd4c06b41370790eec3a2a9884b764330074)
* [Allow a minimum of 0 fps](https://github.com/umlaeute/v4l2loopback/commit/2fe4e20fbb4be62998895b632e2c35d7c4b36102)

I've reverted these changes, and the playback doesn't freeze any more for me.

umlaeute, 2023-04-12:
that should be fixed with #533 (which has been merged)

closed 2023-04-12 by umlaeute