v4l2loopback/v4l2loopback #520: VIDIOC_S_FMT operation has no impact on the format of the output video device.

canberkgurel, 2023-02-10:
I'm using v4l2loopback to create a virtual camera device. The VIDIOC_S_FMT operation is unable to set the format of the output device. The operation doesn't return an error, which is expected as it was explained in the [documentation][1]. 

> Drivers should not return an error code unless the type field is
> invalid, this is a mechanism to fathom device capabilities and to
> approach parameters acceptable for both the application and driver.

The documentation also mentions that under normal circumstances the hardware should be programmed, however some very simple and inflexible devices may continue to retain the original format. I think this is exactly what's happening.

> On success, the driver may program the hardware, allocate resources and
> generally prepare for data exchange. ... Very simple, inflexible
> devices may even ignore all input and always return the default
> parameters.

The code.
```
// Step 1: open output video device here
#define VIDEO_OUT  "/dev/video6"
int output = open(VIDEO_OUT, O_RDWR);
```
```
// print specs of output device
v4l2 --   desc   Planar YUV 4:2:0 
v4l2 --   flags V4L2_FMT_FLAG_UNCOMPRESSED 
v4l2 --   fourcc 0x32315559  YUV 4:2:0 (V4L2_PIX_FMT_YUV420)
```
```
// Step 2: acquire video format from device
struct v4l2_format old_format;
memset(&vid_format, 0, sizeof(vid_format));
vid_format.type = V4L2_BUF_TYPE_VIDEO_OUTPUT;
if (ioctl(output, VIDIOC_G_FMT, &vid_format) < 0) {
    return -1;
}
```
```
// print old_format here
v4l2 --   width  640
v4l2 --   height 480
v4l2 --   pitch  640
v4l2 --   size   460800
v4l2 --   format 0x32315559  YUV 4:2:0 (V4L2_PIX_FMT_YUV420)
v4l2 --   color  0x8
v4l2 --   field  0x1
```
```
// Step 3: configure desired video format on device
struct v4l2_format new_format;
memset(&new_format, 0, sizeof(new_format));
new_format.type = V4L2_BUF_TYPE_VIDEO_OUTPUT;
new_format.fmt.pix.width = w;
new_format.fmt.pix.height = h;
new_format.fmt.pix.pixelformat = V4L2_PIX_FMT_YUYV;
new_format.fmt.pix.sizeimage = framesize;
new_format.fmt.pix.field = V4L2_FIELD_NONE;
```
```
// print new_format here
v4l2 --   width  640
v4l2 --   height 480
v4l2 --   pitch  1280
v4l2 --   size   614400
v4l2 --   format 0x56595559  YUV 4:2:2 (V4L2_PIX_FMT_YUYV)
v4l2 --   color  0x8
v4l2 --   field  0x1
```
```
// Step 4: attempt to set desired format
if (ioctl(output, VIDIOC_S_FMT, &new_format) < 0) {
    return -1;
}
// Step 5: validate 
memset(&old_format, 0, sizeof(old_format));
old_format.type = V4L2_BUF_TYPE_VIDEO_OUTPUT;
if (ioctl(output, VIDIOC_G_FMT, &old_format) < 0) {
    return -1;
}
```
```
// print old_format here
v4l2 --   width  640
v4l2 --   height 480
v4l2 --   pitch  640
v4l2 --   size   460800
v4l2 --   format 0x32315559  YUV 4:2:0 (V4L2_PIX_FMT_YUV420)
v4l2 --   color  0x8
v4l2 --   field  0x1
```

  [1]: https://www.kernel.org/doc/html/v4.9/media/uapi/v4l/vidioc-g-fmt.html#vidioc-g-fmt

  * `v4l2loopback` version: 0.12.5

  * kernel version: 5.15.0-60-generic

  * Distribution (+version): Ubuntu 20.04.5 LTS

#### Observed Results:

  * The print outs prove that `output` device doesn't allow the VIDIOC_S_FMT operation to change the default format.

#### Expected Results:

  * This doesn't seem to be expected as I see that VIDIOC_G_FMT and VIDIOC_S_FMT operations are used in [test.c](https://github.com/umlaeute/v4l2loopback/blob/main/examples/test.c).

umlaeute, 2023-02-13:
unable to reproduce.

please provide a fully working, compilable example source file.

also note, that as long as you have a consumer attached to the device, you will not be able to change the format (the `v4l2loopback` does *not* do any conversion at all; so we cannot change the format while somebody is consuming frames).

umlaeute, 2023-04-20:
Closing die to inactivity

closed 2023-04-20 by umlaeute