v4l2loopback/v4l2loopback #522: Ubuntu 20.04: v4l2loopback-dkms not found on latest Kernel version

anhduong2608, 2023-02-21:
### Step 2: Describe your environment

  * `v4l2loopback` version: "v4l2loopback-dkms is already the newest version (0.12.3-1ubuntu0.4)."

          sudo dmesg  | grep -i v4l2loopback

  * kernel version: "5.15.0-47-generic #51-Ubuntu SMP Thu Aug 11 07:51:15 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux"

          uname -a

  * Distribution (+version): Ubuntu 20.04.3 LTS/Docker container

          lsb_release -a

### Step 3: Describe the problem:
After installing v4l2loopback-module successfully follow the instructions:
`sudo apt install v4l2loopback-dkms`

When I try to load the module into the kernel: `sudo modprobe v4l2loopback`, the error occurs:
> modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/5.15.0-47-generic

I tried searching on Stackoverflow but only vaguely suggested is it because I use old update channel on Ubuntu?
What do I need to do in this case?

#### Steps to reproduce:

As above.

#### Observed Results:

As above.

#### Expected Results:

module loaded successfully and I can call it from command line.

anhduong2608, 2023-02-21:
> you cannot run `v4l2loopback` in a Docker-container: it is a _kernel_ module (and a container has no kernel)

can i install `v4l2loopback` module on the host and use it on the ubuntu docker?

jrx-sjg, 2023-09-04:
> you cannot run `v4l2loopback` in a Docker-container: it is a _kernel_ module (and a container has no kernel)

That's not true. Containers share kernel with their host, so you can use host's modules with containers without any issue more than installing corresponding kernel extras and headers on the container and run them with privileged mode.