v4l2loopback/v4l2loopback #525: ioctl(VIDIOC_G_FMT): Invalid argument Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument

randomkid1227, 2023-03-14:
#### Environment:
Version:
v4l2loopback driver version 0.12.7 (built from source on current kernel)
Linux universe 5.13.0-35-generic #40~20.04.1-Ubuntu SMP Mon Mar 7 09:18:32 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal

#### Steps to reproduce:
1. Write to device ffmpeg -re -i ~/path_to_vid/vid.mp4 -r 25 -f v4l2 -pix_fmt yuv420p /dev/video3
2. Wait for file to end (or disconnect if stream)
3. Write to device again  ffmpeg -re -i ~/Videos/day_center.mp4 -r 25 -f v4l2 -pix_fmt yuv420p /dev/video3

#### Observed Results:
Unable to write to the video device again until reloading module or rebooting (not an option for me):
modprobe -r v4l2loopback; depmod -a; modprobe v4l2loopback video_nr=2,3,4,5,6,7 card_label="first","second","third","fourth","fifth","sixth" exclusive_caps=1,1,1,1,1,1

Output:
Failure: [video4linux2,v4l2 @ 0x5637eb3d2b80] ioctl(VIDIOC_G_FMT): Invalid argumentte=N/A speed=N/A    
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Error initializing output stream 0:0 -- 

#### Expected Results:
v4l2loopback devices can be written to

**Note:**
Similar issue has posted before https://github.com/umlaeute/v4l2loopback/issues/247 however there was no solution and was closed due to inactivity.
I haven't encountered this on v4l2loopback v0.12.5, only after building v0.12.7

umlaeute, 2023-04-12:
unable to reproduce with current `main`  (2e4dcae2a832d9195d38bfd8745f0e5d1b941493) and linux-6.1.0

closed 2023-04-12 by umlaeute

jstiefel, 2023-10-11:
> unable to reproduce with current `main` ([2e4dcae](https://github.com/umlaeute/v4l2loopback/commit/2e4dcae2a832d9195d38bfd8745f0e5d1b941493)) and linux-6.1.0

I can confirm. Had the same issue with the 0.12.3 and 0.12.7 release on Ubuntu 20.04 (5.15.0-86-generic). But using the latest main (5bb9bed) resolved it.