v4l2loopback/v4l2loopback #544: Error when using nvv4l2camerasrc to get data from a virtual device created by v4l2loopback

tunghason, 2023-05-22:
Hi,
I get errors when using nvv4l2camerasrc plugin to get data from a virtual device created by v4l2loopback.
### My environment

  * `v4l2loopback` version: v0.12.7

  * kernel version: 5.10.104-tegra #5 SMP PREEMPT Fri Jan 6 19:17:42 +07 2023 aarch64 aarch64 aarch64 GNU/Linux

  * Distribution (+version): Ubuntu 20.04.5 LTS
  
  * Jetson: JetPack 5.0.2. 
  * Deepstream and Gstreamer version are 6.1.1 and 1.16.3 corresponding.
 
### Describe the problem:

#### Steps to reproduce:

  1. Create virtual devices:
  gst-launch-1.0 nvv4l2camerasrc device=/dev/video0 ! 'video/x-raw(memory:NVMM), format=UYVY' ! nvvidconv ! identity drop-allocation=true ! queue ! v4l2sink device=/dev/video2
  2. Get data from the virtual device /dev/video2 using nvv4l2camerasrc:
  gst-launch-1.0 nvv4l2camerasrc device=/dev/video2 ! 'video/x-raw(memory:NVMM), format=UYVY' ! nvvideoconvert ! fpsdisplaysink video-sink=fakesink --verbose

#### Observed Results:

  * I get "Internal data stream error" like this:
  
[image]

#### Expected Results:

  * When I use v4l2src plugin to get data from the virtual device by the command below, it works normally:
  gst-launch-1.0 v4l2src device=/dev/video2 ! 'video/x-raw' ! fpsdisplaysink video-sink=fakesink sync=false -v
    And logs like this:
[image]
  * For more information, nvv4l2camerasrc will work normally on the source device /dev/video0.

umlaeute, 2023-05-22:
does it still happen with current `main` (5eaff75d98e4695ce87db756e3fa4d81a657d785)?

tunghason, 2023-05-22:
Yeah, if I build at main branch, it still happen.
And one more issue at main branch is that I can not use nvv4l2camerasrc to clone data stream anymore, I must use v4l2src in this command below to clone data stream:
gst-launch-1.0 v4l2src device=/dev/video0 ! 'video/x-raw, format=UYVY' ! queue ! v4l2sink device=/dev/video2

tunghason, 2023-05-25:
Hello @umlaeute, is this still in progress ? Is there any chance to use nvv4l2camerasrc to get data from loopback devices ?

umlaeute, 2023-05-25:
i don't have a tegra-hardware, so I cannot test (nor solve) the issue.
so: no, there's no *active* pursuit of this issue (from my side).

in any case: try adding `tee` and/or `queue` elements (they often take care of sync-problems between gst-elements).

apart from that: why do you need to access the loopback device with `nvv4l2camerasrc` instead of the normal `v4l2src`? due to the nature of the loopback device, i'm not sure whether it will give you any performance improvements...

tunghason, 2023-05-25:
Ok, thank you.
About "why need nvv4l2camerasrc", I've tested nvv4l2camerasrc on the source device (/dev/video0), and I see that nvv4l2camerasrc is better than v4l2src in the aspect of speed and CPU usage, so I expect I will get the same effect on virtual devices, too.

umlaeute, 2023-05-26:
i did a quick google-check and found [this](https://docs.nvidia.com/jetson/archives/r35.3.1/DeveloperGuide/text/SD/Multimedia/AcceleratedGstreamer.html?highlight=nvv4l2camerasrc#progressive-capture-using-nvv4l2camerasrc):
> The <kbd>nvv4l2camerasrc</kbd> plugin default currently supports only DMABUF (importer role) streaming I/O mode with `V4L2_MEMORY_DMABUF`.

the `v4l2loopback` device currently only supports `V4L2_MEMORY_MMAP`, so that might explain why it's not working.

i'm therefore closing this as a duplicate of #501

closed 2023-05-26 by umlaeute