v4l2loopback/v4l2loopback #547: The module loads but not creating any device

happyme531, 2023-07-07:
* `v4l2loopback` version:  0.12.7 (0.12.7-397-g9ba7e29) 
  * kernel version: 5.10.160-rk35xx
  * Distribution (+version): Debian GNU/Linux 11 (bullseye)

#### Steps to reproduce:

  1. clone the v4l2loopback repo
  2. make, sudo make install, sudo depmod -a
  3. sudo modprobe v4l2loopback

#### Observed Results:

The module loaded: (`dmesg`)
```log
[57666.139808] v4l2loopback driver version 0.12.7 (0.12.7-397-g9ba7e29) loaded
```
but no new `/dev/videoX`  created and `/dev/v4l2loopback` is not exist
```bash
user@hinlink-h88k ~/v4l2loopback (main)> ls /dev/v4l2loopback
ls:  '/dev/v4l2loopback': no such file or directory
user@hinlink-h88k ~/v4l2loopback (main) [2]> ls /dev/video1
ls:  '/dev/video1':  no such file or directory
```

(/dev/video0 is the platform's video device and not related to v4l2loopback)
```bash
user@hinlink-h88k ~/v4l2loopback (main) [2]> ls /dev/
ashmem           gpiochip0    initctl@      mmcblk1boot0  rtc@     tty17  tty33  tty5   tty9      vcsa4
autofs           gpiochip1    input/        mmcblk1boot1  rtc0     tty18  tty34  tty50  ttyS2     vcsa5
block/           gpiochip2    kmsg          mmcblk1p1     shm/     tty19  tty35  tty51  ubi_ctrl  vcsa6
btrfs-control    gpiochip3    kvm           mmcblk1rpmb   snd/     tty2   tty36  tty52  uhid      vcsu
bus/             gpiochip4    log@          mpp_service   stderr@  tty20  tty37  tty53  uinput    vcsu1
char/            gpiochip5    loop0         mqueue/       stdin@   tty21  tty38  tty54  urandom   vcsu2
console          hdmirx_hdcp  loop1         net/          stdout@  tty22  tty39  tty55  v4l/      vcsu3
cpu_dma_latency  hugepages/   loop2         null          sw_sync  tty23  tty4   tty56  vcs       vcsu4
crypto           hwrng        loop3         port          tty      tty24  tty40  tty57  vcs1      vcsu5
cuse             i2c-0        loop4         ppp           tty0     tty25  tty41  tty58  vcs2      vcsu6
disk/            i2c-10       loop5         ptmx          tty1     tty26  tty42  tty59  vcs3      vendor_storage
dma_heap/        i2c-11       loop6         ptp0          tty10    tty27  tty43  tty6   vcs4      vhci
dri/             i2c-12       loop7         pts/          tty11    tty28  tty44  tty60  vcs5      video0
drm_dp_aux0      i2c-2        loop-control  ram0          tty12    tty29  tty45  tty61  vcs6      zero
drm_dp_aux1      i2c-6        mali0         random        tty13    tty3   tty46  tty62  vcsa      zram0
fd@              i2c-7        mapper/       rfkill        tty14    tty30  tty47  tty63  vcsa1
full             i2c-9        mem           rga           tty15    tty31  tty48  tty7   vcsa2
fuse             iio:device0  mmcblk1       rkspi-dev2    tty16    tty32  tty49  tty8   vcsa3
```

However there are something in the /sys/devices/virtual/video4linux/

```bash
user@hinlink-h88k ~/v4l2loopback (main)> ll /sys/devices/virtual/video4linux/
总用量 0
drwxr-xr-x 3 root root 0 Jul  7 17:45 video2/
user@hinlink-h88k ~/v4l2loopback (main)> ll /sys/devices/virtual/video4linux/video2/
总用量 0
-r--r--r-- 1 root root 4.0K Jul  7 17:52 buffers
-r--r--r-- 1 root root 4.0K Jul  7 17:45 dev
-rw-r--r-- 1 root root 4.0K Jul  7 17:52 dev_debug
-rw-r--r-- 1 root root 4.0K Jul  7 17:52 format
-r--r--r-- 1 root root 4.0K Jul  7 17:45 index
-rw-r--r-- 1 root root 4.0K Jul  7 17:52 max_openers
-r--r--r-- 1 root root 4.0K Jul  7 17:52 name
drwxr-xr-x 2 root root    0 Jul  7 17:52 power/
-r--r--r-- 1 root root 4.0K Jul  7 17:52 state
lrwxrwxrwx 1 root root    0 Jul  7 17:52 subsystem -> ../../../../class/video4linux/
-rw-r--r-- 1 root root 4.0K Jul  7 17:45 uevent
```

#### Expected Results:

The control device and video device are created for use.

#### Note

The kernel is Rockchip's vendor kernel. Vendor kernel is generally considered buggy so it could be actually a kernel-related issue. 
Kernel source is: https://github.com/armbian/linux-rockchip/tree/rk-5.10-rkr4
Used config is: https://github.com/happyme531/armbian-h88k-images/blob/master/config

umlaeute, 2023-07-10:
well: the mere existence of `/sys/devices/virtual/video4linux/video2/` shows that `v4l2loopback` (or some other kernel module that provides a similar/same "virtual" interface) is loaded and working (to some extent).

i cannot currently test on a linux-5 kernel.

in any case you could try loading the module with `debug=2` or some other high number  to see whether it gives some additional information.

hackermondev, 2023-10-23:
I'm experiencing this too but ``/sys/devices/virtual/video4linux/video2/`` doesn't even exist for me. 

I don't know if it's related but I have successfully loaded v4l2loopback already in the past and I removed it with ``modprobe -r v4l2loopback``.

umlaeute, 2023-10-23:
@hackermondev if it doesn't exist than either
- your test is too narrow (`/sys/devices/virtual/video4linux/video2/` is created only if `v4l2loopback` created `/dev/video2`; if you only created a single device, that might as well be `/sys/devices/virtual/video4linux/video0/`; or some other device, if the given video-device already exists)
- the module is not loading at all
- (some very old module is loaded)
- your kernel lacks sysfs support

a more robust test would be something like:
```sh
have_v4l2loopback=no
for f in  /sys/devices/virtual/video4linux/video*/name; do
  if [ -e "${f}" ]; then
    have_v4l2loopback=yes
    break
  fi
done
```
of course it always helps if you specify the
- kernel version
- v4l2loopback version
- module parameters you are passing when loading the module
- info about pre-existing video-devices

hackermondev, 2023-10-23:
@umlaeute thanks for the response.

it looks like there's a video10 in the ``/sys/devices/virtual/video4linux/`` directory and the script returns ``v4l2loopback support: yes``

but the v4l2loopback-ctl tool fails with ``unable to open control device '/dev/v4l2loopback': No such file or directory``. i've tried reinstalling it multiple times (make install, sudo modprobe v4l2loopback) and it doesn't work.

like i mentioned, i had it working perfectly well before (devices work and /dev/v4l2loopback existed) but I removed it with ``sudo modprobe -r v4l2loopback`` and it hasn't worked since.

- kernel version: ``5.4.0-164-generic``
- v4l2loopback version: commit 5bb9bed on the ``main`` branch (latest one right now)
- module parameters: when i first loaded it and it was working, i didn't send any parameters. current parameters  i'm trying to set are ``video_nr=10 exclusive_caps=1``, but it doesn't work even when i don't send any parameters now (only running ``sudo modprobe v4l2loopback``)

umlaeute, 2023-10-24:
@hackermondev it would be great if you could also say *how* you fixed the issue

hackermondev, 2023-10-24:
Yeah, sorry. I'm not sure 100% what fixed the issue but here's what I did:
- I uninstalled the ``v4l2loopback-dkms`` apt module I had installed (Not sure why I installed that when I was installing from the source)
- I used modprobe to removed some other modules that referenced v4l2loopback

``lsmod | grep v4l2loopback``
```
 v4l2loopback           49152  0
videodev              225280  1 v4l2loopback
```

Then, I ran ``sudo modprobe -r v4l2loopback`` and ``sudo modprobe -r videodev``. 
- I rebuilt everything from the source, reran ``sudo make install`` and ``sudo modprobe v4l2loopback exclusive_caps=1``

umlaeute, 2023-10-25:
> I uninstalled the v4l2loopback-dkms apt module I had installed (Not sure why I installed that when I was also using it from the source)

probably because it's mentioned somewhere, that DKMS will help you with signing modules.

umlaeute, 2024-03-19:
closing, as it has been resolved for the OP, and has become unreproducible