v4l2loopback/v4l2loopback #550: Not buildable on i686

mikhailnov, 2023-08-06:
Latest HEAD - https://github.com/umlaeute/v4l2loopback/commit/9ba7e294ea6f45410a15f192381d4918e8df4f02 - is buildable on x86_64 and aarch64, but is not on i686 (x86_32):

`ERROR: modpost: "__moddi3" [/root/v4l2loopback/v4l2loopback.ko] undefined!`

I do not see where `__moddi3` gets from. It is an x86_64-specific function...

Actually nobody is going to use it on i686, so it is not a big problem.

mikhailnov, 2023-08-06:
The latest commit on which a 32 bit kernel module for kernel 6.1 is buildable is b92e9ce7cccd30cd28743a3fe57eb7827909aeab o_O

mikhailnov, 2023-08-06:
So, commit 9f512816dfad3beb1d329bcb1d5955f6c5ddda99 broke 32 bit

pfactum, 2023-09-09:
Does this:

```diff
diff --git a/drivers/media/v4l2-core/v4l2loopback.c b/drivers/media/v4l2-core/v4l2loopback.c
index 4659bdf7ebea..012bbcb11173 100644
--- a/drivers/media/v4l2-core/v4l2loopback.c
+++ b/drivers/media/v4l2-core/v4l2loopback.c
@@ -1541,7 +1541,7 @@ static int vidioc_reqbufs(struct file *file, void *fh,
 {
 	struct v4l2_loopback_device *dev;
 	struct v4l2_loopback_opener *opener;
-	int i;
+	int i, t;
 	MARK();
 
 	dev = v4l2loopback_getdevice(file);
@@ -1594,7 +1594,8 @@ static int vidioc_reqbufs(struct file *file, void *fh,
 
 			/* after we update dev->used_buffers, buffers in outbufs_list will
 			 * correspond to dev->write_position + [0;b->count-1] range */
-			i = dev->write_position % b->count;
+			t = dev->write_position;
+			i = do_div(t, b->count);
 			list_for_each_entry(pos, &dev->outbufs_list,
 					    list_head) {
 				dev->bufpos2index[i % b->count] =
```

fix the issue?

mikhailnov, 2023-09-09:
No, it did not fix i686 builds

https://abf.io/build_lists/4681832
https://file-store.rosalinux.ru/api/v1/file_stores/7171b89d5a59138e3053993e5dc41928b58c60ce.log?show=true
DEBUG: ERROR: modpost: "__moddi3" [/builddir/build/BUILD/v4l2loopback-9ba7e294ea6f45410a15f192381d4918e8df4f02/v4l2loopback.ko] undefined!

pfactum, 2023-09-09:
Oh, sorry, I've just realised you are from Muscovy. Given you kill people in Ukraine, I'll not help you further.

Good luck.

mikhailnov, 2023-09-09:
Буду рад Вашему возвращению в здравоумие, если это когда-нибудь случится 🙂

umlaeute, 2023-09-09:
@pfactum please stop this.
you are free to help or not help whomever you like, but please keep your reasons for yourself when you are posting on *this* issue tracker.

closed 2024-03-21 by umlaeute

mikhailnov, 2024-03-22:
Thanks for the fix.