v4l2loopback/v4l2loopback #551: v4l2loopback-ctl add ?

aanno, 2023-09-06:
I've set up v4l2loopback 0.12.7 with dkms on fedora 38. Kernel module works, `v4l2loopback-ctl` is compiled.

However,

```bash
cd /usr/src/v4l2loopback-0.12.7/utils
sudo ./v4l2loopback-ctl add -n "Obs Virtual Camera" /dev/video5
Usage:
    ./v4l2loopback-ctl [command] [args]

Commands:
     set-fps <FPS> /dev/videoX  Set device fps (if format is already
                                negotiated).

     set-timeout-image <IMAGE> /dev/videoX  Set timeout on device, and 
                                            optionally the placeholder picture,
                                            instead of default null frame.

     set-caps <CAPS> /dev/videoX  Fix format for the device (using GStreamer
                                  caps), so it will be maintained until reset,
                                  regardless of future readers/writers.

Options:
    FPS      frames per second, either as integer ('30') or fraction ('50/2')
    IMAGE    image file ("no-input.png"), or 'videotestsrc'
    CAPS     GStreamer caps ("video/x-raw, format=UYVY, width=640, height=480"),
             or 'any'
    /dev/videoX  the v4l2-loopback device file you want to act on
```

Hence, it seems to me that `v4l2loopback-ctl add` is _not_ supported. Any idea on this?

umlaeute, 2023-09-06:
Sure: v4l2loopback 0.12.7 (or really 0.12.x) does not support dynamically adding or removing devices.

aanno, 2023-09-06:
Dear @umlaeute, 

could you _please_ elaborate on which versions support dynamic device management?

pallaswept, 2023-10-30:
> No released version does. I was hoping to get a `0.13.0` version (that would support it) out this summer, but apparently this hasn't happened (yet)

Just popped by to report this same issue, thought I was doing it wrong. Since I'm here I wanted to take the opportunity to say thanks for this project. I'll look forward to version 0.13.0 whenever you have the chance, but it's very helpful as it is :) Thanks!

dusanpol, 2024-10-13:
I encountered this issue when I tried to add loopback device as described in [readme](https://github.com/umlaeute/v4l2loopback#dynamic-device-management)
So... how am I supposed to add a loopback device?

pallaswept, 2024-10-13:
>  So... how am I supposed to add a loopback device?

You can add the device statically when you load the module https://github.com/umlaeute/v4l2loopback?tab=readme-ov-file#options