v4l2loopback/v4l2loopback #557: Failed to stream H264 to virtual camera: "does not support 2:4:7:1 colorimetry"

thilindaEdiri, 2023-10-02:
### My environment

  * `v4l2loopback` version: 0.12.5
  * GStreamer version: 1.16.3 

  * kernel version (`uname -a`):
5.15.0-84-generic #93~20.04.1-Ubuntu SMP Wed Sep 6 16:15:40 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
          
  * Distribution (+version) (`lsb_release -a`): 
  Distributor ID:	Ubuntu
  Description:	Ubuntu 20.04.6 LTS
  Release:	20.04
  Codename:	focal
          

### Problem Description:
I am trying to stream my USB cam video in H264 format to virtual camera using GStreamer. But i am getting error: `does not support 2:4:7:1 colorimetry`.
My USB cam (Arducam USB Camera: Arducam USB) support H264 video format.

**Trying to stream H264 video from USB camera (`/dev/video2`) to virtual camera (`/dev/video10`):**

`gst-launch-1.0 -v v4l2src device=/dev/video2 ! video/x-h264,width=640,height=480,framerate=30/1 ! v4l2sink device=/dev/video10 `

**Gets Following error:**
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Device '/dev/video2' does not support 2:4:7:1 colorimetry
Additional debug info:
gstv4l2object.c(3849): gst_v4l2_object_set_format_full (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Device wants 2:4:5:1 colorimetry
Execution ended after 0:00:00.019369981
Setting pipeline to NULL ...
Freeing pipeline ...
```

**Trying to get raw video from USB camera (`/dev/video0`) and convert to H264 and stream to virtual camera (`/dev/video10`):**
`gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoscale ! videoconvert ! x264enc ! h264parse ! video/x-h264,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! v4l2sink device=/dev/video10`

**Gets Following error:**
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstV4l2Src:v4l2src0.GstPad:src: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)30/1, format=(string)YUY2, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:5:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstVideoScale:videoscale0.GstPad:src: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)30/1, format=(string)YUY2, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:5:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)30/1, format=(string)Y42B, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
Redistribute latency...
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:sink: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)30/1, format=(string)Y42B, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:sink: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)30/1, format=(string)YUY2, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:5:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstVideoScale:videoscale0.GstPad:sink: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)30/1, format=(string)YUY2, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)2:4:5:1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:src: caps = video/x-h264, codec_data=(buffer)017a001effe1001d677a001ebcd940a03db016a0c040a8000003000800000301e478b16cb001000568ebecb22c, stream-format=(string)avc, alignment=(string)au, level=(string)3, profile=(string)high-4:2:2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, codec_data=(buffer)017a001effe1001d677a001ebcd940a03db016a0c040a8000003000800000301e478b16cb001000568ebecb22c, stream-format=(string)avc, alignment=(string)au, level=(string)3, profile=(string)high-4:2:2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, parsed=(boolean)true
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = video/x-h264, codec_data=(buffer)017a001effe1001d677a001ebcd940a03db016a0c040a8000003000800000301e478b16cb001000568ebecb22c, stream-format=(string)avc, alignment=(string)au, level=(string)3, profile=(string)high-4:2:2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:02.491361422
Setting pipeline to NULL ...
Freeing pipeline ...
```

**Successful when trying to stream raw video from `/dev/video0` to virtual camera `/dev/video10`**

`gst-launch-1.0 -v v4l2src device=/dev/video0 ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! videoconvert ! v4l2sink device=/dev/video10 `

[... 51 lines cut ...]
		sysfs.path = /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/video4linux/video2
		device.bus = usb
		device.subsystem = video4linux
		device.vendor.id = 0c45
		device.vendor.name = "Arducam\\x20Technology\\x20Co.\\x2c\\x20Ltd."
		device.product.id = 6366
		device.product.name = "Arducam\ USB\ Camera:\ Arducam\ USB"
		device.serial = Arducam_Technology_Co.__Ltd._Arducam_USB_Camera_UC684
		device.capabilities = :capture:
		device.api = v4l2
		device.path = /dev/video2
		v4l2.device.driver = uvcvideo
		v4l2.device.card = "Arducam\ USB\ Camera:\ Arducam\ USB"
		v4l2.device.bus_info = usb-0000:00:14.0-1
		v4l2.device.version = 331636 (0x00050f74)
		v4l2.device.capabilities = 2225078273 (0x84a00001)
		v4l2.device.device_caps = 69206017 (0x04200001)
	gst-launch-1.0 v4l2src device=/dev/video2 ! ...
```

I also tried with `v4l2loopback` versions `0.12.6, 0.12.7` and gstreamer version `1.18.6`. But same errors.

I would appreciate a prompt response.

Thank you.

umlaeute, 2023-10-03:
could you try with current `main` (9ba7e294ea6f45410a15f192381d4918e8df4f02)?

i cannot fully reproduce the problem (having no hardware that natively speaks h264), however:

the following pipeline appears to **work** for me with v0.12.7 and current main:
```sh
gst-launch-1.0 -v videotestsrc \
! x264enc \
! h264parse \
! "video/x-h264,width=640,height=480,pixel-aspect-ratio=1/1,framerate=30/1,interlace-mode=progressive,colorimetry=2:4:7:1" \
! v4l2sink device=/dev/video2
```

however, the following pipeline **fails** from me with v0.12.7 (note the "colorimetry=bt601")
```sh
gst-launch-1.0 -v videotestsrc \
! x264enc \
! h264parse \
! "video/x-h264,width=640,height=480,pixel-aspect-ratio=1/1,framerate=30/1,interlace-mode=progressive,colorimetry=bt601" \
! v4l2sink device=/dev/video2
```

*however*, the above pipeline appears to work with `main`

thilindaEdiri, 2023-10-05:
Hi @umlaeute,

Thank you for your response.

I tried with v4l2loopback v0.12.7 (https://github.com/umlaeute/v4l2loopback/commit/9ba7e294ea6f45410a15f192381d4918e8df4f02) with following pipeline (`/dev/video10` is the virtual camera).

```
gst-launch-1.0 -v videotestsrc \
! x264enc \
! h264parse \
! "video/x-h264,width=640,height=480,pixel-aspect-ratio=1/1,framerate=30/1,interlace-mode=progressive,colorimetry=2:4:7:1" \
! v4l2sink device=/dev/video10
```
I get the following error.

```
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, format=(string)Y444, multiview-mode=(string)mono, interlace-mode=(string)progressive
Redistribute latency...
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:sink: caps = video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, format=(string)Y444, multiview-mode=(string)mono, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:src: caps = video/x-h264, codec_data=(buffer)01f4001effe1001d67f4001e919b281407b602d41808150000030001000003003c8f162d9601000568ebec4480, stream-format=(string)avc, alignment=(string)au, level=(string)3, profile=(string)high-4:4:4, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, codec_data=(buffer)01f4001effe1001d67f4001e919b281407b602d41808150000030001000003003c8f162d9601000568ebec4480, stream-format=(string)avc, alignment=(string)au, level=(string)3, profile=(string)high-4:4:4, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, parsed=(boolean)true
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = video/x-h264, codec_data=(buffer)01f4001effe1001d67f4001e919b281407b602d41808150000030001000003003c8f162d9601000568ebec4480, stream-format=(string)avc, alignment=(string)au, level=(string)3, profile=(string)high-4:4:4, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.
Additional debug info:
../subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
```
For the following pipeline:
```
gst-launch-1.0 -v v4l2src device=/dev/video2 ! video/x-h264,width=640,height=480,framerate=30/1 \
 ! h264parse ! v4l2sink device=/dev/video10 
```

I get the following error:
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Device '/dev/video2' does not support 2:0:0:0 colorimetry
Additional debug info:
gstv4l2object.c(3849): gst_v4l2_object_set_format_full (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
Device wants 2:4:5:1 colorimetry
Execution ended after 0:00:00.175808594
Setting pipeline to NULL ...
Freeing pipeline ...
```

For the following pipeline:
```
gst-launch-1.0 -v v4l2src device=/dev/video2 \
 ! "video/x-h264,width=640,height=480,framerate=30/1,pixel-aspect-ratio=1/1,interlace-mode=progressive,colorimetry=2:4:5:1" \ 
 ! h264parse ! v4l2sink device=/dev/video10 
```
I get the following error:
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstV4l2Src:v4l2src0: Internal data stream error.
Additional debug info:
../subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.000449644
Setting pipeline to NULL ...
Freeing pipeline ...
```

Is it possible that the outcome can be changed based on the Linux distribution or the kernel version?

Because i want to run this on Jetson Orin with Ubuntu 20.04 installed. It also get the same errors as my laptop.

- Linux Distribution on Jetson: Ubuntu 20.04.6 LTS
- Kernel version on Jetson: 5.10.65-tegra

umlaeute, 2023-10-05:
rather than the Linux kernel version i would imagine that the GStreamer version matters.

I'm using gstreamer_1.22.6-1 (on my Debian/sid machine).

Newer Linux distributions will have newer versions of the packaged software, which in turn will have bugs fixed (and new bugs)

samhurst, 2024-11-12:
Apologies for reviving a year-old thread, but I'm having this exact same issue. I'm receiving video from a remote IP source, decoding it and then trying to present it as a webcam device within a web app. However, whenever I try to link the video decoder and the `v4l2sink` element it fails to link the pipeline because it can't match the `colorimetry` caps parameter between my decoder pipeline and the sink. In my case, the `colorimetry` caps parameter is set as "2:4:7:1".

I've done some digging and discovered that the 2:4:7:1 string comes from the [`gst_video_colorimetry_to_string`](https://gstreamer.freedesktop.org/documentation/video/video-color.html?gi-language=c#gst_video_colorimetry_to_string) function failing to understand the [`GstVideoColorimetry`](https://gstreamer.freedesktop.org/documentation/video/video-color.html?gi-language=c#GstVideoColorimetry) structure passed in. The reason for this is that in GStreamer's view, sRGB is always full range, whereas the v4l2loopback driver seems to use limited range (and thus the 2 in the colorimetry string).

Reading the v4l2 docs, [it seems to imply that sRGB should be limited range by default](https://www.kernel.org/doc/html/latest/userspace-api/media/v4l/colorspaces-details.html#col-srgb) - so am I misunderstanding something or is GStreamer wrong to assert that sRGB can only ever be full range?

I'm using Ubuntu 24.04, kernel 6.8.12, v4l2loopback version 0.12.7 (from Ubuntu repositories) and GStreamer 1.25.0 from git (as of a couple of days ago).

stephematician, 2024-12-13:
This works on main for me:

```
gst-launch-1.0 -v videotestsrc \
! x264enc \
! h264parse \
! "video/x-h264,width=640,height=480,pixel-aspect-ratio=1/1,framerate=30/1,interlace-mode=progressive,colorimetry=2:4:7:1" \
! v4l2sink device=/dev/video10
```

Although I don't know enough about this to confirm that the output is correct (messaed around with ffplay for a bit, but ran out of time).

sanbrother, 2025-01-02:
Seems commit 6e3d0a77736d7a035fa11639be49933f2a849a32 introduced this issue.
I'm writing an app, with one v4l2src, one videotestsrc, switched by an input-selector element. I see similar log about colorimetry.
But v0.12.6 seems to work.

sanbrother, 2025-01-02:
@stephematician Are you sure you are loading the custom-built kernel module?
modinfo v4l2loopback command may help
For me v0.12.6 works.

stephematician, 2025-01-02:
@sanbrother - yes, I am using the main branch of v4l2loopback. Can you provide an example of a command that produces the error?

sanbrother, 2025-01-03:
```
#include <gst/gst.h>

static GMainLoop *loop = NULL;

static gboolean
my_bus_callback(GstBus *bus, GstMessage *message, gpointer data)
{
    g_print("Got %s message\n", GST_MESSAGE_TYPE_NAME(message));

    switch (GST_MESSAGE_TYPE(message))
    {
    case GST_MESSAGE_ERROR:
    {
        GError *err;
        gchar *debug;

        gst_message_parse_error(message, &err, &debug);
        g_print("Error: %s\n", err->message);
        g_error_free(err);
        g_free(debug);

        g_main_loop_quit(loop);
        break;
    }
    case GST_MESSAGE_EOS:
        /* end-of-stream */
        g_main_loop_quit(loop);
        break;
    default:
        /* unhandled message */
        break;
    }

    /* we want to be notified again the next time there is a message
     * on the bus, so returning TRUE (FALSE means we want to stop watching
     * for messages on the bus and our callback should not be called again)
     */
    return TRUE;
}

/* Create the pipeline using gst_parse_launch */
static const gchar *pipeline_description =
  "input-selector name=selector ! "
  "videoconvert ! "
  "capsfilter caps=\"video/x-raw, width=640, height=480, format=YUY2\" ! "
  "tee name=t "
  "t. ! queue ! v4l2sink name=v4l2sink device=/dev/video51 "
  "t. ! queue ! v4l2sink device=/dev/video52 "
  "v4l2src name=src1 device=/dev/video0 ! selector. "
  "videotestsrc name=src2 ! selector.";

static GstElement *src1, *src2;
static GstElement *pipeline, *video_switch;
static GstElement *v4l2sink;

static gboolean
switch_timer(GstElement *video_switch)
{
    gint nb_sources;
    GstPad *active_pad, *new_pad;
    gchar *active_name;

    g_message("switching");
    g_object_get(G_OBJECT(video_switch), "n-pads", &nb_sources, NULL);
    g_object_get(G_OBJECT(video_switch), "active-pad", &active_pad, NULL);

    active_name = gst_pad_get_name(active_pad);
    printf("COCOA : active_name = %s\n", active_name);

    if (strcmp(active_name, "sink_0") == 0)
    {
        gst_element_set_state(GST_ELEMENT(src2), GST_STATE_PLAYING);
        new_pad = gst_element_get_static_pad(video_switch, "sink_1");
        gst_element_set_state(GST_ELEMENT(src1), GST_STATE_NULL);
    }
    else
    {
        gst_element_set_state(GST_ELEMENT(src1), GST_STATE_PLAYING);
        new_pad = gst_element_get_static_pad(video_switch, "sink_0");
        gst_element_set_state(GST_ELEMENT(src2), GST_STATE_NULL);
    }
    g_object_set(G_OBJECT(video_switch), "active-pad", new_pad, NULL);
    g_free(active_name);
    gst_object_unref(new_pad);

    g_message("current number of sources : %d, active source %s",
              nb_sources, gst_pad_get_name(active_pad));

    return (GST_STATE(GST_ELEMENT(video_switch)) == GST_STATE_PLAYING);
}

int main(int argc, char *argv[])
{
    GstBus *bus;

    /* Initialize GStreamer */
    gst_init(&argc, &argv);

    /* Create a main loop */
    loop = g_main_loop_new(NULL, FALSE);

    pipeline = gst_parse_launch(pipeline_description, NULL);
    if (!pipeline)
    {
        g_printerr("Failed to create pipeline from description.\n");
        return -1;
    }

    src1 = gst_bin_get_by_name(GST_BIN(pipeline), "src1");
    if (!src1) {
        g_printerr("Failed to get 'src1' element from pipeline.\n");
        return -1;
    }

    src2 = gst_bin_get_by_name(GST_BIN(pipeline), "src2");
    if (!src2) {
        g_printerr("Failed to get 'src2' element from pipeline.\n");
        return -1;
    }

    v4l2sink = gst_bin_get_by_name(GST_BIN(pipeline), "v4l2sink");
    if (!v4l2sink) {
        g_printerr("Failed to get 'v4l2sink' element from pipeline.\n");
        return -1;
    }

    video_switch = gst_bin_get_by_name(GST_BIN(pipeline), "selector");
    if (!video_switch) {
        g_printerr("Failed to get 'video_switch' element from pipeline.\n");
        return -1;
    }

    /* Add a bus watch to handle messages */
    bus = gst_element_get_bus(pipeline);
    gst_bus_add_watch(bus, my_bus_callback, NULL);
    gst_object_unref(bus);

    /* Start the pipeline */
    gst_element_set_state(pipeline, GST_STATE_PLAYING);
    g_usleep(G_USEC_PER_SEC);
    GST_DEBUG_BIN_TO_DOT_FILE_WITH_TS(GST_BIN(pipeline), GST_DEBUG_GRAPH_SHOW_ALL, "pipeline");

    g_timeout_add(4000, (GSourceFunc)switch_timer, video_switch);

    /* Run the main loop */
    g_main_loop_run(loop);

    /* Clean up */
    gst_element_set_state(pipeline, GST_STATE_NULL);
    gst_object_unref(pipeline);
    g_main_loop_unref(loop);

    return 0;
}
```

```
gcc gst_launch.c -o gst_launch `pkg-config --cflags --libs gstreamer-1.0`
./gst_launch

# open a new terminal window
vlc v4l2:///dev/video52

# you will see
# 0:00:21.038298627 133684 0x56431cfa8120 WARN                 default v4l2-utils.c:189:gst_v4l2_error:<v4l2sink1> error: Device '/dev/video52' does not support bt601 colorimetry
# 0:00:21.038362864 133684 0x56431cfa8120 WARN                 default v4l2-utils.c:192:gst_v4l2_error:<v4l2sink1> error: Device wants 2:4:7:1 colorimetry
```

sanbrother, 2025-01-03:
the example above runs on v0.13.0 with this patch
```diff
diff --git a/v4l2loopback.c b/v4l2loopback.c
index 61663d5..fe1f243 100644
--- a/v4l2loopback.c
+++ b/v4l2loopback.c
@@ -1562,9 +1562,11 @@ static int vidioc_reqbufs(struct file *file, void *fh,
                return 0;
        }
 
+#if 0
        if (V4L2_TYPE_IS_OUTPUT(b->type) && (!dev->ready_for_output)) {
                return -EBUSY;
        }
+#endif
 
        init_buffers(dev);
        switch (b->memory) {
```

stephematician, 2025-01-03:
> the example above runs on v0.13.0 with this patch
>
> ```diff
> diff --git a/v4l2loopback.c b/v4l2loopback.c
> index 61663d5..fe1f243 100644
> --- a/v4l2loopback.c
> +++ b/v4l2loopback.c
> @@ -1562,9 +1562,11 @@ [...]

That's related to ioctl REQBUFS - so this probably isn't an issue with colour space support; it's probably related to how v4l2loopback handles devices that are opened multiple times. I've been looking at this recently but I don't have a PR ready just yet (am on holidays) - as I encountered a slightly different problem at the same line of code.

sanbrother, 2025-01-03:
@umlaeute 

Recently, I've been trying the combination of gstreamer and v4l2loopback, and I want to make an attempt at sharing the camera data stream. It's not very urgent. Have a nice holiday!

stephematician, 2025-01-08:
@sanbrother - I tried your example in comment https://github.com/umlaeute/v4l2loopback/issues/557#issuecomment-2568664362. It SIGSEGV'd.

I used `modprobe v4l2loopback video_nr=51,52` assuming that you wanted `/dev/video51` and `/dev/video52` to be loopbacks. On my machine `/dev/video0` is a Logitech C270 - so a relatively sane webcam.

The main thread was at line 141:

```c
    g_usleep(G_USEC_PER_SEC);
```

Thread #5 "src1:src" received SIGSEGV somewhere in `libgstvideo4linux2.so` (no debug info available). Another time thread #6 "src2:src" received signal SIGSEGV.

It did run _once_ with output:

```console
user@home:~/doc/temp$ ./gst-test
** Message: 10:29:04.785: switching
COCOA : active_name = sink_0
** Message: 10:29:04.785: current number of sources : 2, active source sink_0
** Message: 10:29:08.789: switching
COCOA : active_name = sink_1
...
```

And then I tried another terminal:

```console
user@home:~$ vlc v4l2:///dev/video52
VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
[00007cd1bc001170] v4l2 demux error: cannot map buffer 0: Invalid argument
[00007cd1bc001170] v4l2 demux error: not a radio tuner device
[00007cd1bc003af0] v4l2 stream error: cannot map buffer 0: Invalid argument
```

Can you find a simpler, more stable example?

sanbrother, 2025-01-14:
Yeah, got it. I will keep trying to see if I can find out a more stable example.

stephematician, 2025-01-17:
> Yeah, got it. I will keep trying to see if I can find out a more stable example.

I'm still reworking some of the S_FMT and REQBUFS mechanics - but I believe this should be resolved in my experimental efforts here [d935c8e](https://github.com/umlaeute/v4l2loopback/pull/611/commits/d935c8ec93034f4ea4fdcd114e66fcacca27760c) and more generally PR #611. __This code is still experimental, I recommend you try first with a virtual machine (e.g. KVM)__.

stephematician, 2025-01-19:
@sanbrother - after thinking about this longer; I'm going back on the changes related to 'unlocking' colorspace that I made in PR #611.

I believe that it makes more sense for v4l2loopback to fail in your example.

The gstreamer pipeline in your example switches between sources with different colorspaces (one for `videotestsrc`, the other for `src1`), and hence so does (each) v4l2loopback sink. Once ffmpeg (or another program) opens the v4l2loopback device for _capture_, it _should_ (or must) lock the colorspace. On the next switch between sources, the gstreamer pipeline should fail as a result.

The solution would be to get gstreamer to convert the two sources into a common colorspace before sinking to the loopback device(s).

Of course, some changes are still needed to v4l2loopback - I am close to having a (non-draft) PR. Should push in the next day or two.

stephematician, 2025-01-19:
@samhurst Can you provide an example of the pipeline you're using and the output you get?

sanbrother, 2025-02-05:
Well, it is difficult to find a stable demo.
I have tried GST 1.20 and 1.24, v4l2loopback version 0.12.x and 0.13.0
The problem seems gone, I don't know why.
But I guess, it is related to format changing (or color changing), because I have two sources that provide different output

stephematician, 2025-03-07:
This is essentially the same problem as https://github.com/v4l2loopback/v4l2loopback/issues/271 -> still not sure how to stream H264 to a loopback node, and then retrieve it from the node.